r-lidar/lasR #3: drop_duplicates() filter crashes R Studio

harryseely, 2024-01-17:
Hi JR, 

When I try to implement the drop_duplicates() filter, R Studio crashes. 

Here is an example:

`las <- system.file("extdata", "MixedConifer.laz", package="lidR")`
`read = reader(las, filter = drop_duplicates())`
`write = write_las(paste0(tempdir(), "/*_rm_duplicates.laz"))`
`pipeline = read + write`
`ans = processor(pipeline)`

Reproducibility infomation:
print(sessionInfo())
R version 4.3.2 (2023-10-31 ucrt)
Platform: x86_64-w64-mingw32/x64 (64-bit)
Running under: Windows 10 x64 (build 19045)

Matrix products: default

locale:
[1] LC_COLLATE=English_Canada.utf8  LC_CTYPE=English_Canada.utf8   
[3] LC_MONETARY=English_Canada.utf8 LC_NUMERIC=C                   
[5] LC_TIME=English_Canada.utf8    

time zone: America/Vancouver
tzcode source: internal

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
[1] lidR_4.0.4 lasR_0.1.0

loaded via a namespace (and not attached):
 [1] dplyr_1.1.2        compiler_4.3.2     tidyselect_1.2.0   Rcpp_1.0.11       
 [5] yaml_2.3.7         fastmap_1.1.1      lattice_0.21-9     R6_2.5.1          
 [9] generics_0.1.3     classInt_0.4-9     sf_1.0-14          knitr_1.44        
[13] tibble_3.2.1       units_0.8-3        DBI_1.1.3          pillar_1.9.0      
[17] rlang_1.1.1        utf8_1.2.3         sp_2.0-0           terra_1.7-39      
[21] xfun_0.40          cli_3.6.1          magrittr_2.0.3     class_7.3-22      
[25] digest_0.6.33      grid_4.3.2         rstudioapi_0.15.0  lifecycle_1.0.3   
[29] vctrs_0.6.3        KernSmooth_2.23-22 proxy_0.4-27       evaluate_0.21     
[33] glue_1.6.2         data.table_1.14.8  raster_3.6-23      codetools_0.2-19  
[37] fansi_1.0.4        e1071_1.7-13       rmarkdown_2.24     pkgconfig_2.0.3   
[41] tools_4.3.2        htmltools_0.5.6   `

Jean-Romain, 2024-01-17:
Thank you for reporting, I may have missed testing this feature. Will fix it tomorrow.

harryseely, 2024-01-17:
Thanks! Excited to start using the new package!

Jean-Romain, 2024-01-18:
Fixed

commit 2574edf8 (2024-01-18): Fix #3 segfault drop_duplicates()
closed 2024-01-18 by Jean-Romain