r-lidar/lasR #4: [Feature request] Parallel processing of catalog + progress bar

JakubCha, 2024-02-05:
Hello JR!

It's a magnificent library!
I'd like to move some of my processing from `lidR` to `lasR`.
Is that currently possible or do you plan to support parallel processing of the catalog i.e `rasterize` process? I guess it won't be possible to do that in a manner similar to `lidR` with the usage of `future` because the whole processing is done in C++.
Another useful feature while processing catalog would be to see progress bar to estimate progress (obviously) and speed of current processing if it's scaling properly to huge amount of data. 

Kind Regards

Jean-Romain, 2024-02-05:
Parallel processing is a big topic. I don' t want to make the same mistakes as with lidR. I do have plans to bring parallelism to process multiple files at once, but it is much harder than you might think. I need to think about it A LOT and make a lot of tests.

You may think that making `rasterize` parallel is straightforward because you think that each file can be processed independently, and you are right in theory. 

In practice, the way a pipeline is built makes it much more difficult to write parallel code. There is no way to use `future` either, a pipeline cannot be divided in several independent R sessions. Everything must be done in c++. Moreover, in some algorithm such as tree segmentation, each file cannot easily be process independently because each process must communicate the tree ID they reached to other processes in order to label each tree safely. That was a big problem in lidR that I want to solve properly in `lasR`. Also some stages are simply *NOT* parallisable because they involve some user's R code like `rasterize()` or `callback()`.

Because a pipeline may include easy parallizable stages, hardly parallelizable stages and not parallelizable stages, you understand it is not that easy. So yes, I'll do it.  When I will have weeks to spend on it. Currently, the development in stand by because I already spent several months on it.

By the way the documentation of `processor` mentions:

> *ncores* integer. Not yet supported.

This is a mistake. Some parts of the codes are actually already parallelized.

```r
processor(pipeline, ncores = 4L, progress = TRUE)
```

Jean-Romain, 2024-02-15:
Just to let you know that I started working on parallelism. This is a hard problem, as hard as expected, but I'll cope with it. I worked on it two days straight, and I have the mind relatively clear. I already wrote a version that can parallel read LAS/LAZ file and apply some simple stages with very good benchmarks.

Jean-Romain, 2024-02-22:
@JakubCha @liamirwin @wiesehahn 

### Warning :warning:

I worked on it for a full week and pushed an experimental and unstable branch called `multithreading`. If you're interested in testing it, it's available, but please be aware that it's highly experimental. It passes all the unit tests, but there are still weird bugs hardly reproducible. If you'd rather not risk any issues, feel free to skip it. I'll take a break on it.

Also, every single piece of information you are going to read below is suggested to be invalidated in the future.

```r
remotes::install_github("r-lidar/lasR", ref = "multithreading")
```

### What's new :new:

There are now two strategies of parallelization: `concurrent-points` and `concurrent-files`.  

1. `concurrent-points`: This is what is already implemented in the released version. Some stages, such as `local_maximum` or `transform_with`, are already natively parallelized and can process multiple points simultaneously. The pipeline is not fully parallelized; only some stages are running in parallel.
2. `concurrent-files`: As explicitly named, it processes several files simultaneously. The entire pipeline is parallelized.
3. `nested` is not supported yet.

### Limitations :no_entry:

- `concurrent-points`: Only some stages are parallelized. Reading LAS/LAZ file is single-threaded, computing a DTM is single-threaded, segmenting the trees is single-threaded, and so on.
- `concurrent-files`: Everything is parallelized. Several LAS/LAZ files are read simultaneously, DTM, segmentation, and other stages are run in parallel on each file. But some pipelines do not support that kind of parallelization because they involve calling R itself, such as `callback()` or `rasterize()` with a custom user-defined R expression. Since R is single-threaded, calling R code in multiple threads is dangerous and usually crashes the session. My attempts to cope with that failed so far.
   ```r
   r <- reader(f)
   pipeline <- r + normalize() + rasterize(1, "max") + write_las() # supports concurrent files strategy
   pipeline <- r + reader(f) + rasterize(10, mean(Intensity)) # does NOT support concurrent files strategy
   ```

### How it works? :book: 

Currently

```r
processor(pipeline, ncores = 4, mode = concurrent_files())
processor(pipeline, ncores = 4, mode = concurrent_points())
```

Maybe later I'll implement it like that. It looks more user-friendly

```r
processor(pipeline, ncores = sequential())
processor(pipeline, ncores = concurrent_files(4))
processor(pipeline, ncores = concurrent_points(4))
processor(pipeline, ncores = nested(4,2))
```

### Known issues :boom: 

- The progress bar is supposed to be thread safe. In practice using the progress bar seems to fail. I'll investigate later. Do not use the progress bar. You can use `processor(... verbose = TRUE)`.
- #6 

### Benchmark :racehorse: 

Tested with 4 LAZ files and 4 cores (1 core per file). Overall, it provides a 2 to 3-fold speedup.

<details>
<summary>Sources and results</summary>

```r
library(lasR)

f = c("/home/jr/Documents/Ulaval/ALS data/BCTS/092L072244_BCTS_2.laz",
      "/home/jr/Documents/Ulaval/ALS data/BCTS/092L072422_BCTS_2.laz",
      "/home/jr/Documents/Ulaval/ALS data/BCTS/092L073133_BCTS_2.laz",
      "/home/jr/Documents/Ulaval/ALS data/BCTS/092L073311_BCTS_2.laz")

f = paste0(system.file(package="lasR"), "/extdata/bcts/")
f = list.files(f, pattern = "(?i)\\.la(s|z)$", full.names = TRUE)

normalize_pileline = reader(f) + normalize() + write_las()
dtm_dsm_pipeline = reader(f) + dtm() + chm()

dsm = chm()
chm = pit_fill(dsm)
tree = local_maximum(3)
segment = region_growing(chm, tree)
segmentation_pipeline = reader(f) + normalize() + dsm + chm + tree + segment

pre_read = reader(f) + lasR:::nothing(stream = T)

# ============
# PRE-READ
# ============

# ensure that the files are already in cache to ensure fair
# comparison of all pipelines
processor(pre_read)

#=============
# PIPELINE 1
# ============

t0 = Sys.time()
processor(normalize_pileline, ncores = 1)
Sys.time() - t0
# 1'12"

t0 = Sys.time()
processor(normalize_pileline, ncores = 8, mode = concurrent_points())
Sys.time() - t0
# 0'50"

t0 = Sys.time()
processor(normalize_pileline, ncores = 4, mode = concurrent_files())
Sys.time() - t0
# 0'32"

# =============
# PIPELINE 2
# =============

t0 = Sys.time()
processor(dtm_dsm_pipeline, ncores = 1)
Sys.time() - t0
# 0'41"

t0 = Sys.time()
processor(dtm_dsm_pipeline, ncores = 8, mode = concurrent_points())
Sys.time() - t0
# 0'40" (there is no parallelized stage is this pipeline)

t0 = Sys.time()
processor(dtm_dsm_pipeline, ncores = 4, mode = concurrent_files())
Sys.time() - t0
# 0'18"

# =============
# PIPELINE 3
# =============

t0 = Sys.time()
processor(segmentation_pipeline, ncores = 1)
Sys.time() - t0
# 2'21"

t0 = Sys.time()
processor(segmentation_pipeline, ncores = 8, mode = concurrent_points())
Sys.time() - t0
# 1'48"

t0 = Sys.time()
processor(segmentation_pipeline, ncores = 4, mode = concurrent_files())
Sys.time() - t0
# 1'0"
```
</details>

Jean-Romain, 2024-03-01:
In my latest commit

- Fixed multithreaded progress bar
- Fixed #6 by side effect
- Implemented the API with
  ```r
  processor(pipeline, strategy = sequential())
  processor(pipeline, strategy = concurrent_files(4))
  processor(pipeline, strategy = concurrent_points(4))
  ```

Todo:

- [ ] support nested parallelism
- [ ] support stages that are calling R C API. Currently, parallelism is disabled when these stages are in the pipeline.

Jean-Romain, 2024-03-13:
The implementation of multi-file parallelism is now completed. The branch `multithreading` is stable and passes all unit tests on Linux Windows and MacOS. We will run tests on larger datasets and call it good. I'm closing this issue.

closed 2024-03-13 by Jean-Romain

wiesehahn, 2024-04-05:
Would the implementation of nested parallelism also mean that it is applied on the level of single files? 
For COPC files this might be doable and seems to be done in https://github.com/PDAL/wrench ("PDAL wrench can split the input into multiple tiles that can be processed independently in parallel").

Different topic, but I am wondering is the concept of Virtual Point Cloud (VPC) different from lidr catalogs?

Jean-Romain, 2024-04-05:
I'm not sure to understand your question. Or, at least, I'm not sure my interpretation of the question match yours.

First, what do you mean by copc files? Regular tiled dataset but with copc spatial indexing? Or a massive copc file covering the entire dataset in a single file? In the first case everything is identical to what you are used to. In the later case nested parallelism may apply to a single file but probably not the way you think. Actually, `concurent-files` applies to chunks not to files. Like in lidR the default is to process by files (chunk == file) and you can process multiple files simultaneously. I recommend working this way but you can split the dataset into chunks (like in `lidR`) and this is the only way to process masive copc files that can't fit in memory. In this case multiple chunks can be processed in parallel and nested parallelism applies.

Hope this answer your question.