r-lidar/lasR #12: Processing las tiles that overlap or are adjacent

georgewoolsey, 2024-03-21:
I have hundreds or even thousands of `las` files in tiles and I have not been able to successfully process las tiles that overlap (with potential duplicate points) or are adjacent (i.e. share a border) using lasR_0.3.2

In this example the error `Error: in write_lax while initalizing:` occurs when processing a `LAScatalog` with tiles that are adjacent, line:
`lasR::exec(pipeline = pipeline, on = ctg_small, buffer = 10, chunk = 100)`

Here is a toy example with adjacent tiles in the `LAScatalog` but the same error occurs if there are overlapping tiles in the catalog:
```
library(lasR)
library(lidR)
f = system.file("extdata", "Megaplot.las", package = "lasR")
ctg = lidR::readLAScatalog(f)

# simple pipeline
tri = lasR::triangulate(filter = lasR::keep_ground())
dtm = lasR::rasterize(res = 1, tri)
pipeline = tri + dtm

# process in chunks (e.g. processing collection with large files or processing a massive copc files)
ans_notile = lasR::exec(pipeline = pipeline, on = ctg, buffer = 10, chunk = 100)
plot(ans_notile[[1]])

# create tiles that are adjacent for testing
lidR::opt_output_files(ctg) = paste0(tempdir(), "{XLEFT}_{YBOTTOM}") # label outputs based on coordinates
lidR::opt_chunk_buffer(ctg) = 0
lidR::opt_chunk_size(ctg) = 100 # retile
ctg_small = lidR::catalog_retile(ctg) # apply retile
plot(ctg_small)

# reset chunk size in lidR to use lasR tiles
lidR::opt_chunk_buffer(ctg_small) = 10
lidR::opt_chunk_size(ctg_small) = 100

# process adjacent tiles in chunks
ans_yestile = lasR::exec(pipeline = pipeline, on = ctg_small, buffer = 10, chunk = 100)
plot(ans_yestile[[1]])
```

Jean-Romain, 2024-03-21:
I do not have error with your example. At the end I have

```r
ans_yestile = lasR::exec(pipeline = pipeline, on = ctg_small, buffer = 10, chunk = 100)
#> 12 files do not have a spatial index. Spatial indexing speeds up tile buffering and spatial queries drastically.
#> Files will be indexed on-the-fly. This will take some extra time now but will speed up everything later.
```

Which is the expected behavior.  Also :

`plot(ans_yestile[[1]])` looks incorrect, like if the buffer was not properly loaded for each chunk. I'll take a look at this specific issue.

If you found another issue, please share data with a reproducible example. If your example is supposed to be reproducible please tell me exactly where it failed and give me more details (OS, R version, loaded package and so on).

Jean-Romain, 2024-03-21:
I confirm the bug. I was using the development version of the package in which the bug seems to be fixed

Jean-Romain, 2024-03-21:
Fixed

closed 2024-03-21 by Jean-Romain

georgewoolsey, 2024-03-22:
I confirm this is fixed using the original example code when processing with lasR_0.3.3. Thanks for the quick response.

commit c0fb2beb (2024-03-21): Fix #12