r-lidar/lasR #13: Processing by chunk creates gaps - incorrect buffering or chunking

Jean-Romain, 2024-03-21:
```r
library(lasR)
library(lidR)
f = system.file("extdata", "Topography.las", package = "lasR")
ctg = lidR::readLAScatalog(f)

# simple pipeline
tri = lasR::triangulate(filter = lasR::keep_ground())
dtm = lasR::rasterize(res = 1, tri)
pipeline = tri + dtm

# process in chunks (e.g. processing collection with large files or processing a massive copc files)
ans_notile = lasR::exec(pipeline = pipeline, on = ctg, buffer = 10, chunk = 100)
plot(ans_notile[[1]])

# create tiles that are adjacent for testing
lidR::opt_output_files(ctg) = paste0(tempdir(), "{XLEFT}_{YBOTTOM}") # label outputs based on coordinates
lidR::opt_chunk_buffer(ctg) = 0
lidR::opt_chunk_size(ctg) = 100 # retile
ctg_small = lidR::catalog_retile(ctg) # apply retile
plot(ctg_small)

# reset chunk size in lidR to use lasR tiles
lidR::opt_chunk_buffer(ctg_small) = 10
lidR::opt_chunk_size(ctg_small) = 100

# process adjacent tiles in 
chunks
ans_yestile = lasR::exec(pipeline = pipeline, on = ctg_small, buffer = 10, chunk = 100)
plot(ans_yestile[[1]])
```

[image]

closed 2024-03-21 by Jean-Romain

georgewoolsey, 2024-03-22:
I replicated a similar issue using `lasR_0.3.3` when writing an intermediate raster file to disk instead of a tempfile:

```
library(lasR)
library(lidR)
f = system.file("extdata", "Topography.las", package = "lasR")
ctg = lidR::readLAScatalog(f)

# create tiles that are overlapping for testing
lidR::opt_output_files(ctg) = paste0(tempdir(), "{XLEFT}_{YBOTTOM}") # label outputs based on coordinates
lidR::opt_chunk_buffer(ctg) = 10
lidR::opt_chunk_size(ctg) = 100 # retile
ctg_small = lidR::catalog_retile(ctg) # apply retile
plot(ctg_small)
# reset chunk size in lidR to use lasR tiles
lidR::opt_chunk_buffer(ctg_small) = 30
lidR::opt_chunk_size(ctg_small) = 0

# simple pipeline
tri = lasR::triangulate(filter = lasR::keep_ground())
# intermediate raster to temptif
dtm_temptif = lasR::rasterize(res = 1, tri, ofile = lasR::temptif())
# intermediate raster to disk
dtm_disktif = lasR::rasterize(res = 1, tri, ofile = "*dtm.tif")
# pipeline
pipeline_temptif = tri + dtm_temptif + lasR::pit_fill(raster = dtm_temptif)
pipeline_disktif = tri + dtm_disktif + lasR::pit_fill(raster = dtm_disktif)

# process adjacent tiles in chunks
ans_temptif = lasR::exec(pipeline = pipeline_temptif, on = ctg_small, buffer = 10, chunk = 100)
ans_disktif = lasR::exec(pipeline = pipeline_disktif, on = ctg_small, buffer = 10, chunk = 100)
# plot
par(mfrow = c(1, 2))
plot(ans_temptif[["pit_fill"]], main = "ofile = temptif")
plot(ans_disktif[["pit_fill"]], main = "ofile = disk")
par(mfrow = c(1,1))
```
[image]

Jean-Romain, 2024-03-22:
I just found the same bug myself 1 minute ago. I don't think it is relating to chunking. I did not find a way to reproduce on demand. It might be related to pit_fill itself. Anyway I'll take your example in a new issue

commit a26e10a8 (2024-03-21): Fix #13