r-lidar/lasR #16: local_maximum() set to a higher value than 3 causes a bugged pit_fill plot

kalinowskapatrycja, 2024-03-24:
Increasing local_maximum() to 7 from 3 creates a broken pit_fill diagram. Code is taken from: https://r-lidar.github.io/lasR/articles/lasR2.html#tree-segmentation 

```
library(lasR)
library(tools)

args = commandArgs(trailingOnly = TRUE)
f = args[1]

del = triangulate(filter = keep_first())
chm = rasterize(0.5, del)
chm2 = pit_fill(chm)
[Rplots (2).pdf](https://github.com/r-lidar/lasR/files/14737060/Rplots.2.pdf)

seed = local_maximum(7)
tree = region_growing(chm2, seed)
pipeline = del + chm + chm2 +  seed + tree
ans = exec(pipeline, on = f)

col = grDevices::colorRampPalette(c("blue", "cyan2", "yellow", "red"))(25)
terra::plot(ans$rasterize, col = col, mar = c(1, 1, 1, 3))
terra::plot(ans$pit_fill, col = col, mar = c(1, 1, 1, 3))
terra::plot(ans$region_growing, col = col[sample.int(25, 297, TRUE)], mar = c(1, 1, 1, 3))
plot(ans$local_maximum$geom, add = T, pch = 19, cex = 0.5)

```
[Rplots (2).pdf](https://github.com/r-lidar/lasR/files/14737063/Rplots.2.pdf)

Jean-Romain, 2024-03-24:
Didn't try your example. It's late tonight here but considering the image I'm pretty sure this has been fixed few days again in #15 and #13. Which version are you using.

kalinowskapatrycja, 2024-03-28:
I have just updated to ‘0.3.4’ and the issue is still present.

Jean-Romain, 2024-03-28:
Ok, so please share a minimal reproducible example including data. If it is reproducible with data included in the package please tell me which one.

kalinowskapatrycja, 2024-03-29:
The code in my original comment is what I used. I ran it with this file as the argument
[LAZ Example.zip](https://github.com/r-lidar/lasR/files/14808485/LAZ.Example.zip)

Jean-Romain, 2024-03-29:
Thanks. Tested with 0.3.5 and reproduced. Tested with 0.4.0 and worked. So the bug seems to be fixed in the devel branch. I will investigate.

If you want, you could start using the version 0.4.0 that is in stable state. The main new feature is the full parallelism of the pipelines. https://github.com/r-lidar/lasR/blob/devel/NEWS.md

```r
remotes::install_github("r-lidar/lasR", ref = "devel")
```

Jean-Romain, 2024-03-30:
I merge 0.4.0 into the master branch. I consider this bug fixed

commit 347e2ee1 (2024-03-22): Fix #16
closed 2024-03-30 by Jean-Romain