r-lidar/lasR #17: transform_with using pit_fill stage raster not currently possible

georgewoolsey, 2024-03-29:
Using lasR_0.3.4 it is not currently possible to use `lasR::transform_with` with the option `stage` set to the raster resulting from a `lasR::pit_fill` step. Given that `lasR::pit_fill` returns a raster, is it possible to get this functionality?

In the example below, the line `norm = lasR::transform_with(stage = pitfill, operator = "-")` returns the error message:

`Error in lasR::transform_with(stage = pitfill, operator = "-") : the stage must be a triangulation or a raster stage`

```
library(lasR)
f = system.file("extdata", "Topography.las", package="lasR")
# lasR steps
tri = lasR::triangulate(filter = lasR::keep_ground())
dtm = lasR::rasterize(res = 1, tri)
pitfill = lasR::pit_fill(raster = dtm)
norm = lasR::transform_with(stage = pitfill, operator = "-") # stage = pitfill not possible
write = lasR::write_las()
chm = lasR::rasterize(res = 1, "max")
# pipeline
pipeline = tri + dtm + pitfill + norm + write
# execute
ans = lasR::exec(pipeline, on = f)
```

Jean-Romain, 2024-03-29:
This is an unexpected use case but I do agree that it should work. I will check out why it does not work. It is probably as simple as a bad tag in the R API. The internal engine handles the case for sure.

closed 2024-03-29 by Jean-Romain

Jean-Romain, 2024-03-29:
Fixed. However, I cannot refrain myself to mention that `pit_fill` is an algorithm to enhance CHM computed with a high resolution. It is not design to smooth DTM or I don't know what are your intentions. Anyway, you are free to use it the way you see fit.

I appreciate that you tested something unconventional and reported a bug. If you want, you could start using the version 0.4.0 that is in stable state. The main new feature is the full parallelism of the pipeline. https://github.com/r-lidar/lasR/blob/devel/NEWS.md

```r
remotes::install_github("r-lidar/lasR", ref = "devel")
```

georgewoolsey, 2024-03-29:
Thanks for the update, @Jean-Romain . I appreciate your insight on the `pit_fill` algorithm as well.

In my experience, normalizing high density point clouds (i.e. >500 points per m2) using `lasR::transform_with()` is very slow to process when the `stage` option is set to the result of `lasR::triangulate` compared to the result of `lasR::rasterize`. 

Further, `lasR::rasterize` often returns digital terrain models with empty tiles which can impact the normalization results obtained via `lasR::transform_with()`. To fill these empty DTM raster tiles `lasR::pit_fill()` works reasonably well and this was the motivation for this submission.

Here is a (relatively) simple pipeline to test `lasR::transform_with()` using the result of `lasR::triangulate` versus the result of `lasR::rasterize`:

```
library(lasR)
library(tidyverse)
# file list
f = list.files(paste0(system.file(package="lasR"), "/extdata/bcts/"), pattern = "(?i)\\.la(s|z)$", full.names = TRUE)
# lasR stages
tri = lasR::triangulate(filter = lasR::keep_ground() + lasR::keep_class(c(9)))
dtm = lasR::rasterize(res = 1, tri)
norm_tri = lasR::transform_with(stage = tri, operator = "-") # ! key difference here
norm_dtm = lasR::transform_with(stage = dtm, operator = "-") # ! key difference here
write_tri = lasR::write_las(paste0(tempdir(), "/*_norm_tri.laz"))
write_dtm = lasR::write_las(paste0(tempdir(), "/*_norm_dtm.laz"))
# set up pipelines
pipeline_norm_tri = lasR::reader_las() + tri + dtm + norm_tri + write_tri
pipeline_norm_dtm = lasR::reader_las() + tri + dtm + norm_dtm + write_dtm
```

I executed each pipeline 50 times on a small data set to compare run times:

```
# function to execute pipelines and store runtime data
exec_time <- function(sample=1) {
  # remove .la(s|z) and .tif files from tempdir
  file.remove(
    list.files(tempdir(), pattern = "(?i)\\.la(s|z)$|(?i)\\.tif$",full.names = T)
  )
  # execute pipelines
  t1 = Sys.time()
  ans_norm_tri = lasR::exec(pipeline = pipeline_norm_tri, on = f)
  t2 = Sys.time()
  ans_norm_dtm = lasR::exec(pipeline = pipeline_norm_dtm, on = f)
  t3 = Sys.time()
  # timers
  pipeline = c("normalize with triangulation", "normalize with dtm")
  seconds = c(
    as.numeric(difftime(t2, t1, units = c("secs")))
    , as.numeric(difftime(t3, t2, units = c("secs")))
  )
  data.frame(pipeline, seconds, sample = sample)
}
# run each pipeline x times and store runtime result in data frame
nsamp = 50
exec_time_data = purrr::map(1:nsamp,exec_time) %>% dplyr::bind_rows()
# results
lm1 = lm(seconds ~ 0 + factor(pipeline), data = exec_time_data)
summary(lm1)
anova(lm1)
# plot
ggplot(exec_time_data, aes(x = pipeline, y = seconds)) + geom_boxplot() + 
  labs(subtitle = paste("number of samples for each pipeline:",nsamp))
```

On this small data, the run time is reduced by about 10% when normalizing with a raster:

[image]

On a bigger data set ([lasR_normalize_test.pdf](https://github.com/r-lidar/lasR/files/14807871/lasR_normalize_test.pdf)) with 81.15 million points and 775 points per m2 the difference is even more pronounced:

[image]

Jean-Romain, 2024-03-29:
```r
norm_tri = lasR::transform_with(stage = tri, operator = "-") # ! key difference here
norm_dtm = lasR::transform_with(stage = dtm, operator = "-") # ! key difference here
```
Of course you have a massive difference of runtime. You are not doing the same thing at all. In the first case you are computing the exact elevation, for each point, by linear interpolation of the triangles. In the second case you are computing the approximate elevation of each point per pixel. The difference is huge.

> Further, lasR::rasterize often returns digital terrain models with empty tiles which can impact the normalization results 

Not sure to see what do you mean.

Anyway your benchmark are not comparable. You are not performing the same task. Normalized with a raster is indeed faster but also less accurate. See also https://r-lidar.github.io/lidRbook/norm.html

georgewoolsey, 2024-03-29:
Thanks for taking the time to review and for clarifying the methodological difference. As you note in the `lidR` book, the choice is up to the use to balance computation time versus accuracy.

>Further, lasR::rasterize often returns digital terrain models with empty tiles which can impact the normalization results

I was just noting that empty cells after applying the `lasR::rasterize` function can occur but these are a function the input point cloud data and the rasterization method used, nothing to do with the `lasR::rasterize` itself. Filling these empty cells was the initial motivation to use the results of `lasR::pit_fill` in the `lasR::transform_with` step.

Jean-Romain, 2024-03-29:
Also I'd like to recall that `lasR` is still in a maturing state. It is no longer an experimental package. I'm strongly confident it is powerful and well-designed. Yet it requires broader testing. So do not hesitate to communicate with me if you have any concern. Seeing more use cases helps.

georgewoolsey, 2024-03-29:
Strongly agree that `lasR` is powerful and well-designed! Most things that I am noting are driven by my inexperience and less bugs. Keep up the great work :)

Jean-Romain, 2024-03-29:
> I was just noting that empty cells after applying the lasR::rasterize function can occur but these are a function the input point cloud data and the rasterization method used, nothing to do with the [...]

You rasterized a triangulation. You should not see empty pixels inside the convex hull. Please report with an example.

georgewoolsey, 2024-03-29:
> You rasterized a triangulation. You should not see empty pixels inside the convex hull. Please report with an example. This may be related to floating points inaccuracies

example:

```
library(lasR)
library(terra)
# pipeline steps
  classify = function(
      # csf options
      smooth = FALSE, threshold = 0.5
      , resolution = 0.5, rigidness = 1L
      , iterations = 500L, step = 0.65
    ){
      csf = function(data, smooth, threshold, resolution, rigidness, iterations, step)
      {
        id = RCSF::CSF(data, smooth, threshold, resolution, rigidness, iterations, step)
        class = integer(nrow(data))
        class[id] = 2L
        data$Classification = class
        return(data)
      }
      
      classify = lasR::callback(
        csf
        , expose = "xyz"
        # csf options
        , smooth = smooth, threshold = threshold
        , resolution = resolution, rigidness = rigidness
        , iterations = iterations, step = step
      )
      return(classify)
  }
tri = lasR::triangulate(filter = lasR::keep_ground())
dtm = lasR::rasterize(res = 0.5)
pipeline = lasR::reader_las() + classify() + tri + dtm
# file
f = system.file("extdata/bcts", "bcts_2.laz", package="lasR")
# execute pipeline
ans = lasR::exec(pipeline = pipeline, on = f)
# plot
terra::plot(ans[[1]], colNA = "black", mar = c(1,1,3,1), main = paste(basename(f), "\n(na's colored black)"))
```

[image]

georgewoolsey, 2024-03-29:
> > I was just noting that empty cells after applying the lasR::rasterize function can occur but these are a function the input point cloud data and the rasterization method used, nothing to do with the lasR::rasterize itself. Filling these empty cells was the initial motivation to use the results of lasR::pit_fill in the lasR::transform_with step.
> 
> You rasterized a triangulation. You should not see empty pixels inside the convex hull. Please report with an example. This may be related to floating points inaccuracies

I could not recreate empty cells in a raster created with a triangulation using data from the package. Here is an example that uses 
a high density point cloud from [Sherwood et al. (2018)](https://doi.org/10.5066/P9XQYCD0) accessible [here](https://www.sciencebase.gov/catalog/item/5b1ee160e4b092d965254a60)

```
library(lasR)
library(terra)
# pipeline steps
  classify = function(
      # csf options
      smooth = FALSE, threshold = 0.5
      , resolution = 0.5, rigidness = 1L
      , iterations = 500L, step = 0.65
    ){
      csf = function(data, smooth, threshold, resolution, rigidness, iterations, step)
      {
        id = RCSF::CSF(data, smooth, threshold, resolution, rigidness, iterations, step)
        class = integer(nrow(data))
        class[id] = 2L
        data$Classification = class
        return(data)
      }
      
      classify = lasR::callback(
        csf
        , expose = "xyz"
        # csf options
        , smooth = smooth, threshold = threshold
        , resolution = resolution, rigidness = rigidness
        , iterations = iterations, step = step
      )
      return(classify)
  }
tri = lasR::triangulate(filter = lasR::keep_ground() + lasR::keep_class(9))
dtm = lasR::rasterize(res = 1, tri, ofile = "dtm.tif")
pipeline = lasR::reader_las() + classify() + tri + dtm
# file
url = "https://www.sciencebase.gov/catalog/file/get/5b1ee160e4b092d965254a60?f=__disk__3f%2F23%2F3a%2F3f233a0f5ba806838b48deb36e0324324c8bf6c7"
f = "2017042FA_LakeBluffs_PointCloud.laz"
# change this to TRUE to download data
if(FALSE){
  options(timeout = 60 * 20) # timeout for download
  # download file
  if(file.exists(f) == FALSE){
    # download data
    download.file(url = url, destfile = f)
  }else{print("file already exists")}
}
# execute pipeline
ans = lasR::exec(pipeline = pipeline, on = f)
# plot
terra::plot(ans[[1]], colNA = "red", mar = c(1,2,3,4), main = paste(basename(f), "\n(na's colored red)"))
```
[image]

Note, the empty cells (red) around the land are expected but the empty cells within the land are not.

commit b9620fd1 (2024-03-29): Fix #17