r-lidar/lasR #18: Empty pixels in the rasterization of a triangulation

Jean-Romain, 2024-03-29:
I could not recreate empty cells in a raster created with a triangulation using data from the package. Here is an example that uses 
a high density point cloud from [Sherwood et al. (2018)](https://doi.org/10.5066/P9XQYCD0) accessible [here](https://www.sciencebase.gov/catalog/item/5b1ee160e4b092d965254a60)

```r
library(lasR)
library(terra)
# pipeline steps
  classify = function(
      # csf options
      smooth = FALSE, threshold = 0.5
      , resolution = 0.5, rigidness = 1L
      , iterations = 500L, step = 0.65
    ){
      csf = function(data, smooth, threshold, resolution, rigidness, iterations, step)
      {
        id = RCSF::CSF(data, smooth, threshold, resolution, rigidness, iterations, step)
        class = integer(nrow(data))
        class[id] = 2L
        data$Classification = class
        return(data)
      }
      
      classify = lasR::callback(
        csf
        , expose = "xyz"
        # csf options
        , smooth = smooth, threshold = threshold
        , resolution = resolution, rigidness = rigidness
        , iterations = iterations, step = step
      )
      return(classify)
  }
tri = lasR::triangulate(filter = lasR::keep_ground() + lasR::keep_class(9))
dtm = lasR::rasterize(res = 1, tri, ofile = "dtm.tif")
pipeline = lasR::reader_las() + classify() + tri + dtm
# file
url = "https://www.sciencebase.gov/catalog/file/get/5b1ee160e4b092d965254a60?f=__disk__3f%2F23%2F3a%2F3f233a0f5ba806838b48deb36e0324324c8bf6c7"
f = "2017042FA_LakeBluffs_PointCloud.laz"
# change this to TRUE to download data
if(FALSE){
  options(timeout = 60 * 20) # timeout for download
  # download file
  if(file.exists(f) == FALSE){
    # download data
    download.file(url = url, destfile = f)
  }else{print("file already exists")}
}
# execute pipeline
ans = lasR::exec(pipeline = pipeline, on = f)
# plot
terra::plot(ans[[1]], colNA = "red", mar = c(1,2,3,4), main = paste(basename(f), "\n(na's colored red)"))
```
[image]

Note, the empty cells (red) around the land are expected but the empty cells within the land are not.

_Originally posted by @georgewoolsey in https://github.com/r-lidar/lasR/issues/17#issuecomment-2027799320_

Jean-Romain, 2024-03-29:
One points not related to the issue itself: the ground classification is very bad. The CSF parameters are not adequate and I don't think it will perform well in such noisy photometric point cloud

We are clearly in a limit case with floating point accuracy issue. The ground points are so dense, the triangles so tiny, that some pixels are inside no triangle because of accuracy issues.

Fast and reproducible example: (1) create spatial index (2) classifiy and save the file

```r
f = "issue 18/2017042FA_LakeBluffs_PointCloud_subset_classified.laz"

tri = lasR::triangulate(filter = lasR::keep_ground() + lasR::keep_class(9))
dtm = lasR::rasterize(res = 1, tri, ofile = "*_dtm.tif")
pipeline = tri + dtm

ans = lasR::exec(pipeline = pipeline, on = f)
terra::plot(ans[[1]], colNA = "red", mar = c(1,2,3,4), main = paste(basename(f), "\n(na's colored red)"))
```

A simple way to demonstrate and solve it is to get 1/100 of the ground points (still 5 ground points/m²). It works.

```r
tri = lasR::triangulate(filter = "-keep_class 2 -keep_class 9 -keep_random_fraction 0.01")
```

While this issue need to be addressed, at this density of point, my advice is anyway to decimate your ground points. With 1/100 of the ground points you already have 5 ground pts/m² to compute a DTM with a 1 m resolution! You could even decimate to 1/250. This will solve your computation time issue in the same time.

Jean-Romain, 2024-03-30:
Fixed. I strongly improved arithmetic accuracy of the internal function `point_in_triangle`

closed 2024-03-30 by Jean-Romain

georgewoolsey, 2024-04-04:
Thanks for the fix! Since implemented, I've seen no empty cells when rasterizing with a triangulation and no longer using pit_fill on  the DTM raster to fill holes ;)

commit 3d440770 (2024-03-30): Fix #18