r-lidar/lasR #20: malloc() Arrow and Parquet Error

kalinowskapatrycja, 2024-03-31:
I get the following error:
```
$ Rscript process.r ST3_A01_2018_P04_LAS_2.laz
ERROR 1: /usr/lib/gdalplugins/ogr_Arrow.so: undefined symbol: _ZNK5arrow16KeyValueMetadata3GetERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE
ERROR 1: /usr/lib/gdalplugins/ogr_Arrow.so: undefined symbol: _ZNK5arrow16KeyValueMetadata3GetERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE
ERROR 1: /usr/lib/gdalplugins/ogr_Parquet.so: undefined symbol: _ZNK5arrow16KeyValueMetadata8ContainsERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE
ERROR 1: /usr/lib/gdalplugins/ogr_Parquet.so: undefined symbol: _ZNK5arrow16KeyValueMetadata8ContainsERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE
malloc(): invalid next size (unsorted)
zsh: IOT instruction (core dumped)  Rscript process.r ST3_A01_2018_P04_LAS_2.laz

```

When running this code on the tutorial page:
```
ibrary(lasR)

args = commandArgs(trailingOnly = TRUE)
f = args[1]

del = triangulate(filter = keep_first())
chm = rasterize(0.5, del)
chm2 = pit_fill(chm)
seed = local_maximum_raster(chm2, 3)
tree = region_growing(chm2, seed)
pipeline = del + chm + chm2 +  seed + tree
ans = exec(pipeline, on = f)

col = grDevices::colorRampPalette(c("blue", "cyan2", "yellow", "red"))(25)
col2 = grDevices::colorRampPalette(c("purple", "blue", "cyan2", "yellow", "red", "green"))(50)
terra::plot(ans$rasterize, col = col, mar = c(1, 1, 1, 3))
terra::plot(ans$pit_fill, col = col, mar = c(1, 1, 1, 3))
terra::plot(ans$region_growing, col = col2[sample.int(50, 277, TRUE)], mar = c(1, 1, 1, 3))
plot(ans$local_maximum$geom, add = T, pch = 19, cex = 0.5)
```

I am using this file:
[LAZ.Example.zip](https://github.com/r-lidar/lasR/files/14817905/LAZ.Example.zip)

The previous example on the tutorial page works fine. gdal and arrow are installed.

Jean-Romain, 2024-04-01:
I confirm the issue. Thank you for reporting.

Jean-Romain, 2024-04-01:
Very stupid bug. I should have tested the code on a bigger example. Fixed now. Thanks

closed 2024-04-01 by Jean-Romain

kalinowskapatrycja, 2024-04-01:
Thank you!

May I ask whether the error attached is an issue with lasR or my system? I have reinstalled lasR, gdal and arrow with no avail.

[errors.txt](https://github.com/r-lidar/lasR/files/14828507/errors.txt)

Jean-Romain, 2024-04-01:
Not related to me. At least not directly. I don't know what is the problem. The package is supposed to compile on windows according to the continuous integration with github action https://github.com/r-lidar/lasR/actions/runs/8511576203

What is weird here is that it compiled for you previously. Nothing changed since lasR 0.1.0 with respect to gdal.

Jean-Romain, 2024-04-01:
Well, according to your paths you are on linux or mac. I can try to help but have no idea of the problem.

kalinowskapatrycja, 2024-04-01:
> Well, according to your paths you are on linux. I can try to help but have no idea of the problem. You were able to install 0.4.0 and nothing change except 2 lines of code to fix two simple bugs

Well it looks like everything is functioning correctly even with the error. So I may just ignore this for now. It does look like an issue with my system.

Jean-Romain, 2024-04-01:
Mmm it was not an issue at the end of the compilation while linking?

kalinowskapatrycja, 2024-04-01:
> Mmm it was not an issue at the end of the compilation while linking?

I am not too sure what that means unfortunately. How do I find out 

I only had this warning when installing lasR
[image]

PS, does lasR support the feature in lidR - 7.1.2 Local Maximum Filter with variable windows size?
I am struggling to get good tree detection without the use of this function (Smaller trees are given huge canopy sizes so my entire plot is covered in canopy even if the .laz has visible ground)
Also, is my y axis supposed to start from 0, or is 170 fine in the pit_fill diagram?
[image]
[image]

Jean-Romain, 2024-04-02:
> PS, does lasR support the feature in lidR - 7.1.2 Local Maximum Filter with variable windows size?

No it does not. One of the main reasons is because it is much slower to compute. The second reason is that I must register a set of pre-defined functions in c++. Out of question to allow a user-defined function like in lidR it is too complex. Overall I'm keen to do it but it is not my priority.

> Also this link no longer seems to be working in the Parallel Processing section https://r-lidar.github.io/lasR/benchmarks.html Though I don't strictly need it, just wanted to let you know.

This link in incorrect. Where did you find it? Thanks

chrisJuresh, 2024-04-02:
> No it does not. One of the main reasons is because it is much slower to compute. The second reason is that I must register a set of pre-defined functions in c++. Out of question to allow a user-defined function like in lidR it is too complex. Overall I'm keen to do it but it is not my priority.

I see, is it best to use this function for data with similar size trees then, and not amazon rain-forest data with extreme variance in canopy and heights?
I should be able to use lasR for the normalisation and chm for the speed, and just have lidR handle the rest right?
Also, I am yet to use the Derived metrics feature in lidR, but I assume I can get a .csv of all of the trees and their canopy size and height with it. Can I do the same with lasR?

> This link in incorrect. Where did you find it? Thanks
Here is a screenshot
[image]

Jean-Romain, 2024-04-02:
If you have any question, please open a dedicated thread in the discussion section

commit 12677342 (2024-04-01): Fix #20