r-lidar/lasR #22: segfault when processing a catalog with "processed" tiles

jmmonnet, 2024-04-03:
Hi, I want to run a normalization (and then add dtm computation) pipeline on selected tiles of a `LAScatalog`. When the pipeline is run on all tiles (no `processed` field), the pipeline runs smoothly, but when the `processed` fields indicates only a subset of tiles to process, the output files are created but then R crashes with a `segfault` error. I have `lasR 0.4.2` installed today.

Here are some files for testing:
https://filesender.renater.fr/?s=download&token=6a1fdcd3-50a2-462a-861c-79a93f85afee

The code I ran:
```
cata <- lidR::catalog("./las")
library(lasR)
# read catalog
read = reader(cata)
# build del
tri = triangulate(filter = "-keep_class 2 9")
# normalize
trans = transform_with(tri)
# write
write2 = write_las("./norm/*.laz")
#
# run pipeline
ans1 = exec(read + tri + trans + write2)
outputfiles <- list.files("./norm/", full.names = TRUE)
print(outputfiles)
#
# remove previous output files
unlink(outputfiles)
#
# set files to process in catalog
cata$processed <- FALSE
cata$processed[2:3] <- TRUE
# read catalog
read = reader(cata)
# run pipeline
ans1 = exec(read + tri + trans + write2)
```

The error I get with R ran from a console:

```
 *** caught segfault ***
address (nil), cause 'memory not mapped'

Traceback:
 1: processor(pipeline, ncores = with$ncores, progress = with$progress,     ...)
 2: exec(read + tri + trans + write2)
```

Jean-Romain, 2024-04-03:
Fixed.

FYI `reader(cata)` is deprecated. It works with `exec` because I added a convenient workaround. The correct syntax is:

```r
read = reader_las()
tri = triangulate(filter = "-keep_class 2 9")
trans = transform_with(tri)
write2 = write_las()
ans1 = exec(read + tri + trans + write2,  on = cata)
```

When `reader_las()` is called without argument, it can be omitted.

```r
tri = triangulate(filter = "-keep_class 2 9")
trans = transform_with(tri)
write2 = write_las()
ans1 = exec(tri + trans + write2,  on = cata)
```

closed 2024-04-03 by Jean-Romain

jmmonnet, 2024-04-03:
Thank you for the tips and for the fix. Fast and efficient as usual !

Jean-Romain, 2024-04-04:
Please reinstall in an hour or two (or now with `remotes::`) my fix was not correct. It fixed the segfault but breaks the order of the output with parallel processing. This was not caught by unit test but it was an happy accident.

commit 2ab41c53 (2024-04-03): Fix #22
commit b7f7fd9e (2024-04-04): Fix #22