r-lidar/lasR #24: possible mistake in VPC/STAC creation

wiesehahn, 2024-04-09:
Hi, I am just starting to play around with lasR. 

I recognised that in the file that `write_vpc()` creates there is an attribute "datatime", but I think in STAC it should be "datetime" (see e.g. https://github.com/stac-extensions/pointcloud/blob/main/examples/example-autzen.json#L86).

wiesehahn, 2024-04-09:
Two other things I just recognised, 
- "pc:count" is 0 for all features (files) in my VPC, is there any possible reason this might have been popluated like this?
- "pc:encoding" is "?" for all features. It seems to have been removed from the specification (https://github.com/stac-extensions/pointcloud/issues/8).

Jean-Romain, 2024-04-09:
Support of vpc is partial and requires a review. Thank you for your contribution. I already know that I did not implement the full VPC specification. The specifications are complex and spread into multiple sub-specifications... For example, I think it is mandatory to export the bounding box in long/lat and it is not mandatory to export the bbox in the CRS of the point cloud. It should be the opposite, in my opinion...

>  there is an attribute "datatime"

That looks like a typo I'll check.

> "pc:count" is 0 for all features (files) in my VPC

It should not. `write_vpc()`  is tested in the unit tests and produces the correct values. Maybe a bug with a LAS format (LAS 1.4 ?). Please report with a reproducible example.

> "pc:encoding" is "?" 
 
I created a VPC file with PDAL and then copied the stucture of the file. PDAL probably produces this one. I will verify what is the purpose of this attribute.

Jean-Romain, 2024-04-09:
This file included with the package has been created with pdal wrench https://github.com/r-lidar/lasR/blob/main/inst/extdata/bcts/bcts.vpc. It has `pc:encoding = "?"`

Jean-Romain, 2024-04-09:
`datatime` was a typo. Fixed.

Jean-Romain, 2024-04-10:
Reproduced `pc:count = 0` with

```r
library(lasR)
f <- system.file("extdata", "las14_prf6.laz", package="rlas")
pipeline <- write_vpc(tempfile(fileext = ".vpc"))
exec(pipeline, on = f)
```

Jean-Romain, 2024-04-10:
Fixed `pc:count` for LAS 1.4

commit caa16da3 (2024-04-10): Fix #24
closed 2024-04-10 by Jean-Romain