r-lidar/lasR #26: posibility to write compressed raster output

wiesehahn, 2024-04-10:
`rasterize()` and its derivates write an uncompressed raster output to disk, however in my opinion some kind of (lossless) compression (e.g. DEFLATE) almost always makes sense. In a random example tile I tested the DTM size was 2300kb uncompressed (as written by lasR) and 600kb compressed with DEFLATE in gdal. 

Not sure if this would be easy to implement or makes sense to you as well though.

Jean-Romain, 2024-04-10:
Your request makes a lot of sense and was in my todo list. First I must figure out how to write compressed rasters with GDAL in the general case (i.e. not only `GeoTiff`).

Jean-Romain, 2024-04-10:
raster in `GeoTiff` format are now created with `COMPRESS=DEFLATE`, `PREDICTOR=2`,`TILED=YES` effectively reducing the size of the rasters. 217 KB to 158 KB in my test. Is it the best option? I don't know. Is there a way to control it for users ? Not yet. This why it is in the `devel` branch.

Jean-Romain, 2024-04-10:
Currently lasR only writes single precision floating point data `GDT_Float32`. Thank you for sharing resources. lasR writes the raster by chunks. The raster is created full size and populated for each pieces of the point cloud processed. Does it change something for compression strategy? I don't know. I will read your resource later.

Jean-Romain, 2024-04-10:
Thank you for sharing the links. The second link especially was really what I needed to understand GDAL compression options. I read other ressources and I may opt for `Zstandard` compression. It seems to be more efficient in I/O than deflate for a similar compression ratio.

Jean-Romain, 2024-04-11:
I thought about it and I won't implement any control over raster options from user side. I'll do it on feature request. This is too much work for too little needs.  If somebody wants compression on another format, they will ask. Thus I'm closing.

closed 2024-04-11 by Jean-Romain