r-lidar/lasR #29: Problem when using filter in rasterize()

wiesehahn, 2024-04-12:
When I try to calculate to calculate some rasterized metric with a filter I get strange results, where some areas (small tiles), get no value or wrong values. Its less pronounced with the testdata but it looks weird to me as well.

```r
f <- system.file("extdata", "Topography.las", package="lasR")

imean <- rasterize(2, "imean")
pipeline = imean
ans = exec(pipeline, on = f)

imean <- rasterize(2, "imean", filter = drop_class(2))
pipeline = imean
ans = exec(pipeline, on = f)
```

without filter:
[image]

with filter:
[image]

E.g. with the filter I would not expext that the area in the lower left changes, since we are dropping ground points but this area is water.

wiesehahn, 2024-04-12:
> By the way in this specific case you should prefer my version of the pipeline. It reads and load less points, implies less computation and should be faster and more memory efficient. Yet, your code is correct and should work.

But if I want to calculate multiple metrics, e.g. DTM, CHM and imean filering in initial read is not an option I guess. Would you suggest to do this separate then or is it more effective to do it in a pipeline like `pipeline = dtm + norm + chm + imean` ?

Jean-Romain, 2024-04-12:
Absolutely not. In the more complex pipeline you are describing, your use case is the correct way to do it. I was just signaling the information for this specific simple pipeline just in case it may help.

Jean-Romain, 2024-04-12:
Fixed. 

Fun fact: there was a comment on the problematic line of code with something like: `//TODO: check if this line is correct`.

commit 830c7bc7 (2024-04-12): Fix #29 rasterize with filter
closed 2024-04-12 by Jean-Romain