r-lidar/lasR #30: fatal error when processing files

wiesehahn, 2024-04-15:
When I try to run my pipeline R crashes after some time. Before that happens lasR tries to spatially index laz files and successfully processes some files (in my case 2 files of each DTM, CHM and imean are created).

My pipeline looks as follows and worked on files in a different folder. Files here are slightly larger (250 - 550 MB).
```r
library(lasR)
library(here)

f = here("data")

dtm = dtm(res = 0.5, ofile = here(f, "*_dtm.tif"))
norm = normalize()
chm = chm(0.5, ofile = here(f, "*_chm.tif"))
filter = "-keep_first -drop_z_below 5"
imean <- rasterize(c(0.5, 1), "imean", filter = filter, ofile= here(f, "*_imean.tif"))
pipeline = dtm + norm + chm + imean
ans = exec(pipeline, on = f)
```
R then prints a message inficating that files are not indexed.

`6 files do not have a spatial index. Spatial indexing speeds up tile buffering and spatial queries drastically.
Files will be indexed on-the-fly. This will take some extra time now but will speed up everything later.`

I thought that the indexing caused trouble, so I tried to index files with lastools prior to executing the pipeline. But this seems not to be the problem as some error persists. 

`ERROR: memory reallocation failed`

```r
> sessionInfo()
R version 4.3.3 (2024-02-29 ucrt)
Platform: x86_64-w64-mingw32/x64 (64-bit)
Running under: Windows 10 x64 (build 17763)

Matrix products: default

locale:
[1] LC_COLLATE=German_Germany.1252  LC_CTYPE=German_Germany.1252    LC_MONETARY=German_Germany.1252 LC_NUMERIC=C                   
[5] LC_TIME=German_Germany.1252    

time zone: Europe/Berlin
tzcode source: internal

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
[1] here_1.0.1 lasR_0.4.5

loaded via a namespace (and not attached):
[1] compiler_4.3.3  rprojroot_2.0.4 tools_4.3.3    
```

Jean-Romain, 2024-04-15:
It is specific to your dataset or is it reproducible with any dataset?

wiesehahn, 2024-04-15:
Not sure yet, I have to try with different files tomorrow. Is there any
possibility to artificially blow up the size of included test files
otherwise?

Jean-Romain, 2024-04-15:
Not sure to understand your question. 

I tested your pipeline successfully on my toy data.

wiesehahn, 2024-04-16:
Ok in first first setup I put a copy of one file (255 MB .laz) with different naming 10 times a folder and started the process:
Memory usage increased a lot at the beginning of processing, up to the point when memory in use by processes and standby memory occupied the entire memory (32 GB in my case). Then R crashes and memory is freed. 

Next I tried to simplify the pipeline, for less processing, but this didnt change anything.
```r
dtm = dtm(res = 2, ofile = here(f, "*_dtm.tif"))
norm = normalize()
chm = chm(2, ofile = here(f, "*_chm.tif"))
pipeline = dtm + norm + chm
```

Then I deleted all but one file in the folder and started the pipeline, this time the process is successful.

Then I copied the file one time, so I have two files and started the process:
Memory usage increased slightly, but not up to the available limit, and R crashes quite quickly.

I also tried the same with a different file, which also works for one file in folder but not two.

Changing from default to `set_parallel_strategy(concurrent_files(4))` didnt change anything.

(If necessary I might provide a file for testing)

wiesehahn, 2024-04-16:
I just recognized that it works with two different files. So my test cases above didnt work because it was two times the same file?!

Jean-Romain, 2024-04-16:
I guess it is possible. Yet I cannot reproduce

```r
library(lasR)

dtm = dtm(res = 2, ofile = paste0(tempdir(), "*_dtm.tif"))
norm = normalize()
chm = chm(2, ofile = paste0(tempdir(), "/*_chm.tif"))
pipeline = dtm + norm + chm

f <- system.file("extdata", "Topography.las", package="lasR")
f = c(f, f)

exec(pipeline, on = f)
```

wiesehahn, 2024-04-16:
Ah ok, makes sense for my stupid test scenario of having multiple times the same file. However this does not really explain why it crashes for two times the same file, or?

And my initial problem is a different one, as I came across the error without duplicates. I guess it has to do with my data as I just came a cross a file where the error occurs also for a single file. 

checking it with lidR the following is returned (at first glance it looks quite similar to another file without problems though):
```r
> summary(t)
class        : LAS (v1.4 format 8)
memory       : 3.7 Gb 
extent       : 547000, 548000, 5725000, 5726000 (xmin, xmax, ymin, ymax)
coord. ref.  : ETRS89 / ETRS89 / UTM 32N + DHHN2016 
area         : 0.99 km²
points       : 38.58 million points
density      : 38.85 points/m²
density      : 26.41 pulses/m²
File signature:           LASF 
File source ID:           0 
Global encoding:
 - GPS Time Type: Standard GPS Time 
 - Synthetic Return Numbers: no 
 - Well Know Text: CRS is WKT 
 - Aggregate Model: false 
Project ID - GUID:        00000000-0000-0000-0000-000000000000 
Version:                  1.4
System identifier:        LAStools (c) by rapidlasso GmbH 
Generating software:      lasnoise (2 
File creation d/y:        292/2023
header size:              375 
Offset to point data:     3115 
Num. var. length record:  5 
Point data format:        8 
Point data record length: 59 
Num. of point records:    38576879 
Num. of points by return: 26219368 7867930 3318024 958421 187588 24949 575 23 1 0 0 0 0 0 0 
Scale factor X Y Z:       0.001 0.001 0.001 
Offset X Y Z:             0 5e+06 0 
min X Y Z:                547000 5725000 203.47 
max X Y Z:                548000 5726000 355.73 
Variable Length Records (VLR):
   Variable Length Record 1 of 5 
       Description: by LAStools of rapidlasso GmbH 
       Tags:
          Key 1024 value 1 
          Key 3072 value 25832 
          Key 3076 value 9001 
          Key 4099 value 9001 
          Key 4096 value 7837 
   Variable Length Record 2 of 5 
       Description: RIEGL Extra Bytes 
       Extra Bytes Description:
          Distance: Distance
          Group: Group
          Normal: Normal vector 2+15+15 bits
   Variable Length Record 3 of 5 
       Description:  
       Extra Bytes Description:
          Distance: Distance
          Group: Group
          Normal: Normal vector 2+15+15 bits
   Variable Length Record 4 of 5 
       Description: TerraScan Extra Bytes 
       Extra Bytes Description:
          Distance: Distance
          Group: Group
          Normal: Normal vector 2+15+15 bits
   Variable Length Record 5 of 5 
       Description: by LAStools of rapidlasso GmbH 
       WKT OGC COORDINATE SYSTEM: COMPD_CS["ETRS89 / ETRS89 / UTM 32N + DHHN2016",PROJCS["ETRS89 / ETRS8 [...] (truncated)
Extended Variable Length Records (EVLR):  void
> plot(t)
Fehler in rgl::rgl.setMouseCallbacks(button, begin, update, dev = dev,  : 
  rgl device is not open
> las_check(t)

 Checking the data
  - Checking coordinates... ✓
  - Checking coordinates type... ✓
  - Checking coordinates range... ✓
  - Checking coordinates quantization... ✓
  - Checking attributes type... ✓
  - Checking ReturnNumber validity... ✓
  - Checking NumberOfReturns validity... ✓
  - Checking ReturnNumber vs. NumberOfReturns... ✓
  - Checking RGB validity... ✓
  - Checking absence of NAs... ✓
  - Checking duplicated points... ✓
  - Checking degenerated ground points...
    ⚠ There were 40 degenerated ground points. Some X Y coordinates were repeated but with different Z coordinates
  - Checking attribute population...
    🛈 'EdgeOfFlightline' attribute is not populated
  - Checking gpstime incoherances ✓
  - Checking flag attributes... ✓
  - Checking user data attribute... ✓
 Checking the header
  - Checking header completeness... ✓
  - Checking scale factor validity... ✓
  - Checking point data format ID validity... ✓
  - Checking extra bytes attributes validity... ✓
  - Checking the bounding box validity... ✓
  - Checking coordinate reference system... ✓
 Checking header vs data adequacy
  - Checking attributes vs. point format... ✓
  - Checking header bbox vs. actual content... ✓
  - Checking header number of points vs. actual content... ✓
  - Checking header return number vs. actual content... ✓
 Checking coordinate reference system...
  - Checking if the CRS was understood by R... ✓
 Checking preprocessing already done 
  - Checking ground classification... yes
  - Checking normalization... no
  - Checking negative outliers... ✓
  - Checking flightline classification... yes
 Checking compression
  - Checking attribute compression...
   -  ScanDirectionFlag is compressed
   -  EdgeOfFlightline is compressed
   -  Synthetic_flag is compressed
   -  Keypoint_flag is compressed
   -  Withheld_flag is compressed
   -  Overlap_flag is compressed
   -  UserData is compressed
   - ```

wiesehahn, 2024-04-16:
> Really, I need the files to reproduce. Degenerated points may or may not be the cause. I need to test and profile running code.

try the file under this link:
 https://transfer.pcloud.com/download.html?code=5Z1mGQ0Zr75m3x87U3kZyQYUZYYPMxTJ7QQbEQkR5sUWr6802GXOV

Jean-Romain, 2024-04-16:
Can you confirm it crashed with `pipeline = lasR:::nothing(T)`

wiesehahn, 2024-04-16:
> Can you confirm it crashes with `pipeline = lasR:::nothing(T)`

yes it does crash

Jean-Romain, 2024-04-16:
The error should be handled without crash, that is an easy fix. Yet the error exist. The memory cannot be allocated. The error is: `Insufficient memory`. I do not understand why. The amount of memory to allocate is 2.2 GB which is a lot but reasonable. Looks like a limit at 2147483647 bytes. I'm investigating.

Jean-Romain, 2024-04-16:
Fixed. lasR can read files bigger than 2.147 GB

closed 2024-04-16 by Jean-Romain

wiesehahn, 2024-04-16:
nope, unfortunately it still crashes 
(also with `pipeline = lasR:::nothing(T)`)

Jean-Romain, 2024-04-16:
Did you install from github and compile yourself ?

wiesehahn, 2024-04-16:
second option

Jean-Romain, 2024-04-16:
This repos is a CRAN-like repos. Like the CRAN the package takes time to be available. It builds the package every 1h hour. Thus, you must wait 30-40 minutes on average and a solid hour in the worst case to get the latest binaries. You installed the package too soon. If my message dates from less than an hour your are better to install from GitHub.

wiesehahn, 2024-04-16:
ah ok, its a little confusing that it seems package version is updated, but now I know
```r
other attached packages:
[1] lasR_0.4.6
```

Jean-Romain, 2024-04-16:
The best way to do

```r
dtm = dtm(res = 0.5, ofile = here(f, "*_dtm.tif"))
norm = normalize()
```

is 

```r
del = triangulate(filter = keep_ground())
dtm = rasterize(0.5, del)
norm = transform_with(del)
```

This way you compute the triangulation only once.

Alternative syntax

```r
dtm = dtm(0.5)
norm = transform_with(dtm[[1]])
```

wiesehahn, 2024-04-17:
> ```r
> del = triangulate(filter = keep_ground())
> dtm = rasterize(0.5, del)
> norm = transform_with(del)
> ```

I guess this is taking the triangulated points as input for terrain calculation, what I dont get is that this is passing `del` to `add_class`, right?
`dtm(res = 1, add_class = NULL, ofile = tempfile(fileext = ".tif"))`

Jean-Romain, 2024-04-17:
No, the function `dtm()` was not used here. Btw `dtm()` is just a convenient wrapper. 

```r
dtm <- function(res = 1, add_class = NULL, ofile = tempfile(fileext = ".tif"))
{
  filter = keep_ground()
  if (!is.null(add_class)) filter <- filter + keep_class(add_class)
  tin <- triangulate(filter = filter)
  chm <- rasterize(res, tin, ofile = ofile)
  return(tin+chm)
}
```

commit 5504ccf3 (2024-04-16): Fix #30 alloc memory above 2.14 GB