r-lidar/lasR #33: local_maximum_raster returns unpopultated attributes

wiesehahn, 2024-04-25:
`local_maximum()` returns Intensity, gpstime, ReturnNumber, Classification and ScanAngle. 
I guess this is just the attributes of that point (highest). I expected just XYZ values, would it make sense to make this optional or add some information in the description about the outcome?

Also `local_maximum_raster()` creates the same values and populates them with 0 except for ReturnNumber which is 1. Does it make sense at all to return these values here as they are obviously unusable?

Jean-Romain, 2024-04-25:
It definitively makes sense.

I've been asked in lidR to add the attributes of the points. It was not possible in lidR so I did it in lasR. And now I am asked to do not add the attributes in lasR :confused: . I'll make it optional this is the best options.

For the raster version it totally makes sense to remove the attributes that are meaningless.

Jean-Romain, 2024-04-25:
For raster it is an accident yes. I just recycled the code for point clouds and treated the raster as a regularly spaced point cloud. With optional attributes I will address both cases in one shot.

Jean-Romain, 2024-04-25:
Fixed

closed 2024-04-25 by Jean-Romain

wiesehahn, 2024-04-27:
More a question than a bug slightly related to this issue. I am wondering if there is an efficient way to get attribute values for each local maximum based on points/pixels in the neighborhood?
E.g. if I want to get the mean of some attribute for each local maximum, based on all points above min height and within window size diameter. I think about how to get mean intensity for each tree as an indicator for species/vitality.

Jean-Romain, 2024-04-27:
What you want is a stage like `lidR::point_metrics()` that takes as input a set of 3d points (e.g. the output of `local_maximum()`) and that compute some metrics using the points in the neighborhood of the user-defined locations.

```r
lm = local_maximum()
nm = neighborhood_metrics(locations = lm, metrics = "imean")
pipeline = lm + nm
```

This is definitively something doable. But no tool is available for that yet.

commit a7d391ff (2024-04-25): Fix #33