r-lidar/lasR #34: problems with VPC

wiesehahn, 2024-04-29:
I created VPC with lasR
```r

create_vpc <- function(input_files, output_file){
  
  if (!file.exists(output_file)){
    
    pipeline = lasR::write_vpc(output_file)
    lasR::exec(pipeline, on = input_files)
    
  } else {
    print("VPC already exists!")
  }
  
}
```

Jean-Romain, 2024-04-29:
VPC specification support is not comprehensive yet. I implemented the very minimal parts for it to work (I assumed).

> Apart from some attributes which are additionally created via QGIS, I recognised that datetime is not populated correctly in lasR.

This is correct. I generated VPC files with PDAL and this was not populated either. So I guessed it was not critical.

> I guess there are more problems as a virtual point cloud I created for own data from lasR could not be read by QGIS and crashed R when executing a pipeline on it.

For R please report a reproducible example. The feature is tested internally with success. For QGIS do you have an error or something. I don't even know how to open that in QGIS.

Regarding CRS I will check out. The string is supposed to be working with escaped quote. I did it few days ago.

Overall there is some work on VPC support and I knew it.

wiesehahn, 2024-04-29:
> This is correct. I generated VPC files with PDAL and this was not populated either. So I guessed it was not critical.

I guess ist not critical, but could be benefitial later to e.g. filter on date or similar.

> For R please report a reproducible example. The feature is tested internally with success. For QGIS do you have an error or something. I don't even know how to open that in QGIS.

You can simply drag it in the viewport and it renders the extent geometry. I guess this is the reason for error, as lasR does not create the geometry attribute.

This is what I get for the above mentioned testfile:
`Invalid Data Source: .../vpc_lasr.vpc is not a valid or recognized data source, error: ""`

And this for my own files:
`Invalid Data Source: .../2023_lasr.vpc is not a valid or recognized data source, error: "JSON parsing error: [json.exception.parse_error.101] parse error at line 17, column 20: syntax error while parsing value - invalid literal; last read: '"proj:wtk2": C'"`

wiesehahn, 2024-04-29:
just recognised that I get the same error in R when trying to run a pipeline on it:
`Error: In the parser while reading the file collection: JSON parsing error: [json.exception.parse_error.101] parse error at line 17, column 20: syntax error while parsing value - invalid literal; last read: '"proj:wtk2": C'`

Jean-Romain, 2024-04-29:
- I confirm that WKT CRS is badly written in the  `v0.4.y` but it should be correct in `v0.5.0`. 
- I confirm that QGIS cannot read the file because of the missing `geometry`.

The good new is that in `v0.5.0` since I added `set_crs()`, the CRS are better handled with GDAL and I can now add the `geometry` field in long/lat which was not possible in previous versions.

I will fix that in `0.5.0` and merge in `master`. I will disable the capacity to abort the computation for now because it is unstable and I can't find a fix easily. This is the only thing that prevent me to release `0.5.0`.

closed 2024-05-02 by Jean-Romain

wiesehahn, 2024-05-02:
With lasR 0.5.0 R crashes when trying to write the VPC for certain files with e.g.:

```r
f = here::here("mylas.laz")
if (!file.exists(here::here("vpc_lasr.vpc"))){  
  pipeline = write_vpc(here::here("vpc_lasr.vpc"))
  exec(pipeline, on = f) 
} else {
  print("VPC already exists!")
}
```
This is the case for files which have no CRS assigned. 
Output for `pipeline <- reader_las()+ summarise()`:
```
$crs
[1] ""

$epsg
[1] 0
```

So I tried to set the CRS and write the VPC with:
```r
f = here::here("mylas.laz")
if (!file.exists(here::here("vpc_lasr.vpc"))){  
  pipeline = set_crs(25832) + write_vpc(here::here("vpc_lasr.vpc"))
  exec(pipeline, on = f) 
} else {
  print("VPC already exists!")
}
```
But R still crashes.

Output for `pipeline <- reader_las() + set_crs(25832) + summarise()`:
```
$crs
[1] "PROJCRS[\"ETRS89 / UTM zone 32N\",BASEGEOGCRS[\"ETRS89\",ENSEMBLE[\"European Terrestrial Reference System 1989 ensemble\",MEMBER[\"European Terrestrial Reference Frame 1989\"],MEMBER[\"European Terrestrial Reference Frame 1990\"],MEMBER[\"European Terrestrial Reference Frame 1991\"],MEMBER[\"European Terrestrial Reference Frame 1992\"],MEMBER[\"European Terrestrial Reference Frame 1993\"],MEMBER[\"European Terrestrial Reference Frame 1994\"],MEMBER[\"European Terrestrial Reference Frame 1996\"],MEMBER[\"European Terrestrial Reference Frame 1997\"],MEMBER[\"European Terrestrial Reference Frame 2000\"],MEMBER[\"European Terrestrial Reference Frame 2005\"],MEMBER[\"European Terrestrial Reference Frame 2014\"],ELLIPSOID[\"GRS 1980\",6378137,298.257222101,LENGTHUNIT[\"metre\",1]],ENSEMBLEACCURACY[0.1]],PRIMEM[\"Greenwich\",0,ANGLEUNIT[\"degree\",0.0174532925199433]],ID[\"EPSG\",4258]],CONVERSION[\"UTM zone 32N\",METHOD[\"Transverse Mercator\",ID[\"EPSG\",9807]],PARAMETER[\"Latitude of natural origin\",0,ANGLEUNIT[\"degree\",0.0174532925199433],ID[\"EPSG\",8801]],PARAMETER[\"Longitude of natural origin\",9,ANGLEUNIT[\"degree\",0.0174532925199433],ID[\"EPSG\",8802]],PARAMETER[\"Scale factor at natural origin\",0.9996,SCALEUNIT[\"unity\",1],ID[\"EPSG\",8805]],PARAMETER[\"False easting\",500000,LENGTHUNIT[\"metre\",1],ID[\"EPSG\",8806]],PARAMETER[\"False northing\",0,LENGTHUNIT[\"metre\",1],ID[\"EPSG\",8807]]],CS[Cartesian,2],AXIS[\"(E)\",east,ORDER[1],LENGTHUNIT[\"metre\",1]],AXIS[\"(N)\",north,ORDER[2],LENGTHUNIT[\"metre\",1]],USAGE[SCOPE[\"Engineering survey, topographic mapping.\"],AREA[\"Europe between 6Â°E and 12Â°E: Austria; Belgium; Denmark - onshore and offshore; Germany - onshore and offshore; Norway including - onshore and offshore; Spain - offshore.\"],BBOX[38.76,6,84.33,12.01]],ID[\"EPSG\",25832]]"

$epsg
[1] 25832
```