r-lidar/lasR #39: With lasR 0.5.0 R crashes when trying to write the VPC for certain files with e.g.:

Jean-Romain, 2024-05-02:
With lasR 0.5.0 R crashes when trying to write the VPC for certain files with e.g.:

```r
f = here::here("mylas.laz")
if (!file.exists(here::here("vpc_lasr.vpc"))){  
  pipeline = write_vpc(here::here("vpc_lasr.vpc"))
  exec(pipeline, on = f) 
} else {
  print("VPC already exists!")
}
```
This is the case for files which have no CRS assigned. 
Output for `pipeline <- reader_las()+ summarise()`:
```
$crs
[1] ""

$epsg
[1] 0
```

So I tried to set the CRS and write the VPC with:

```r
f = here::here("mylas.laz")
if (!file.exists(here::here("vpc_lasr.vpc"))){  
  pipeline = set_crs(25832) + write_vpc(here::here("vpc_lasr.vpc"))
  exec(pipeline, on = f) 
} else {
  print("VPC already exists!")
}
```
But R still crashes.

Output for `pipeline <- reader_las() + set_crs(25832) + summarise()`:
```
$crs
[1] "PROJCRS[\"ETRS89 / UTM zone 32N\",BASEGEOGCRS[\"ETRS89\",ENSEMBLE[\"European Terrestrial Reference System 1989 ensemble\",MEMBER[\"European Terrestrial Reference Frame 1989\"],MEMBER[\"European Terrestrial Reference Frame 1990\"],MEMBER[\"European Terrestrial Reference Frame 1991\"],MEMBER[\"European Terrestrial Reference Frame 1992\"],MEMBER[\"European Terrestrial Reference Frame 1993\"],MEMBER[\"European Terrestrial Reference Frame 1994\"],MEMBER[\"European Terrestrial Reference Frame 1996\"],MEMBER[\"European Terrestrial Reference Frame 1997\"],MEMBER[\"European Terrestrial Reference Frame 2000\"],MEMBER[\"European Terrestrial Reference Frame 2005\"],MEMBER[\"European Terrestrial Reference Frame 2014\"],ELLIPSOID[\"GRS 1980\",6378137,298.257222101,LENGTHUNIT[\"metre\",1]],ENSEMBLEACCURACY[0.1]],PRIMEM[\"Greenwich\",0,ANGLEUNIT[\"degree\",0.0174532925199433]],ID[\"EPSG\",4258]],CONVERSION[\"UTM zone 32N\",METHOD[\"Transverse Mercator\",ID[\"EPSG\",9807]],PARAMETER[\"Latitude of natural origin\",0,ANGLEUNIT[\"degree\",0.0174532925199433],ID[\"EPSG\",8801]],PARAMETER[\"Longitude of natural origin\",9,ANGLEUNIT[\"degree\",0.0174532925199433],ID[\"EPSG\",8802]],PARAMETER[\"Scale factor at natural origin\",0.9996,SCALEUNIT[\"unity\",1],ID[\"EPSG\",8805]],PARAMETER[\"False easting\",500000,LENGTHUNIT[\"metre\",1],ID[\"EPSG\",8806]],PARAMETER[\"False northing\",0,LENGTHUNIT[\"metre\",1],ID[\"EPSG\",8807]]],CS[Cartesian,2],AXIS[\"(E)\",east,ORDER[1],LENGTHUNIT[\"metre\",1]],AXIS[\"(N)\",north,ORDER[2],LENGTHUNIT[\"metre\",1]],USAGE[SCOPE[\"Engineering survey, topographic mapping.\"],AREA[\"Europe between 6Â°E and 12Â°E: Austria; Belgium; Denmark - onshore and offshore; Germany - onshore and offshore; Norway including - onshore and offshore; Spain - offshore.\"],BBOX[38.76,6,84.33,12.01]],ID[\"EPSG\",25832]]"

$epsg
[1] 25832
```

_Originally posted by @wiesehahn in https://github.com/r-lidar/lasR/issues/34#issuecomment-2090104565_

Jean-Romain, 2024-05-02:
Ok I guess it is only with `write_vpc()`?

In theory this should work

```r
pipeline = set_crs(25832) + write_vpc(here::here("vpc_lasr.vpc"))
```

In pratice it does not because `write_vpc` is actually executed before anything else. It made sense and was much simpler to handle internally  and invisible for users but I do agree that now with the new `set_crs()` stage + upcoming new stages `write_vpc` should behave as a regular stage.

wiesehahn, 2024-05-02:
Ok, that explains why it didnt work. 

> Ok I guess it is only with `write_vpc()`?

 Yes so far I only experienced this with write_vpc()

In case input is an ungeoreferenced point cloud it might make sense to abort with an error before R crashes.

Jean-Romain, 2024-05-02:
Absolutely I will fix it. In the previous versions the long/lat bounding box was not written in the vpc file but this prevented QGIS to open the file meaning that it was not an optional field. Now we MUST have a valid CRS to be able to transform to WGS84 and `write_vpc` can no longer work without a CRS. 

I will update today with an error to avoid crashing and within a week I will rewrite `write_vpc` such as it behaves as a regular stages and you can use `set_crs()`

wiesehahn, 2024-05-02:
Thanks, as said earlier I am just playing around with lasR and try to report things I am struggling with, but I have no need that there is a quick fix or enhancement. :)

Jean-Romain, 2024-05-02:
And I appreciate you help A LOT. I just cannot find all bugs, test everything alone in addition to my regular work.

Jean-Romain, 2024-05-02:
Fixed the crash with an error. Did not increment the version. It is still 0.5.0

Jean-Romain, 2024-05-02:
This now works

```r
pipeline = set_crs(25832) + write_vpc(here::here("vpc_lasr.vpc"))
```

closed 2024-05-02 by Jean-Romain