r-lidar/lasR #47: pipeline as named list

wiesehahn, 2024-05-13:
in lasR a pipiline is a unnamed list, would it possibly make sende to name the list?

This would make it more convenient to access certain stages and change attributes. 

As implemented currently we can do e.g.
```r
pipeline[[1]]$filter
```
With a named list 
it would be e.g.
```r
pipeline$reader_las$filter
```
Not sure about the naming though when there are multiple CHM stages for example.

Jean-Romain, 2024-05-13:
This is a reasonable request but you figured out yourself why I did not do it yet :wink: . We have the same issue with the named `list` containing the results. Added in the todo list with a low priority.

closed 2024-05-13 by Jean-Romain

Jean-Romain, 2024-05-13:
Done. I'm using `make.names()` from R so duplicated names are handled appending `.1`, `.2` and so on.

wiesehahn, 2024-05-14:
Thanks.

>  However, you are not supposed to manipulate these objects. They can change without notice.

What exactly does this mean?
Is it a bad idea to something like `pipeline$svd$features <- "l"` to change stage attributes before rerunning?

Jean-Romain, 2024-05-14:
Yes. To create a `svd` stage there is a documented function `geometry_feature()`. The return of this function is not documented. You found the structure of the object because R is permissive and offers you functions like `str` and Rstudio offers you auto-completion.

So, you are using non documented stuff, and non documented stuff may change without notice, for example if I start using a structure that is no longer a `list`. In your case it is unlikely to change but I don't know what kind of operations you may try in the future. You could easily break something.

wiesehahn, 2024-05-14:
alright, got it, you can easily see the structure a pipeline is stored as and for experimenting it was just convinient to update individual attributes. But of course its also possible to write the pipeline with altered attribute again. Although we could also iterate through attributes with this ;)

Jean-Romain, 2024-05-14:
As long as you are aware that you are using undocumented tricks, you can do whatever you want :wink:

commit 1e92b1af (2024-05-13): Fix #47