r-lidar/lasR #49: new feature neighborhood_metrics()

wiesehahn, 2024-05-14:
> PS: please open an issue for feature request about what we talked about few days ago.

Original question in https://github.com/r-lidar/lasR/issues/33 

> I am wondering if there is an efficient way to get attribute values for each local maximum based on points/pixels in the neighborhood?
E.g. if I want to get the mean of some attribute for each local maximum, based on all points above min height and within window size diameter. I think about how to get mean intensity for each tree as an indicator for species/vitality.

Response
> What you want is a stage like lidR::point_metrics() that takes as input a set of 3d points (e.g. the output of local_maximum()) and that computes some metrics using the points in the neighborhood of the user-defined locations.

```r
lm = local_maximum()
nm = neighborhood_metrics(locations = lm, metrics = "imean")
pipeline = lm + nm
```

I have not entirely thought through this, but the function could use similar inputs as `geometry_features()`, namely `k` and `r` to define k-nearest neighbours and radius of the neighborhood. For my initial example something like the following would then work to add an attribute `imean` to ttops. 

```r
lm = local_maximum(ws = 5, min_height = 2, filter = "-drop_class 2", use_attribute = "Z")
nm = neighborhood_metrics(locations = lm, metrics = "imean", r = 2.5, filter = "-drop_class 2 -drop_z_below 2", ofile = tempgpkg())
pipeline = lm + nm
```

I think a filter option here is less problematic than in `geometry_features()` as we are writing the attribute to 3D points and not the point cloud?!

As stated earlier, this is in idea which I think could be useful, but there is no need to feel pressured to implement it for an individual like me.

Jean-Romain, 2024-05-14:
> As stated earlier, this is in idea which I think could be useful, but there is no need to feel pressured to implement it for an individual like me.

It is also interesting to me to see if I can construct a pipeline of that complexity. But I see no strong limitation, the engine is already designed to handle this kind of use case and most of the code to make this feature already exist. I have no internal structure to add custom attribute to geographic points but it is probably not a big issue. Your example looks good and I can't see any obvious feature to add. `locations` could handle anything that is made of point including the point cloud itself. Thus such stage could allow to reproduce `lidR::point_metrics()`.

Jean-Romain, 2024-05-17:
So far I wrote 80% of the code. The remaining 20% are not the funniest and will likely take 80% of the time :smile:

wiesehahn, 2024-05-28:
Just out of curiosity I was wondering if the neighborhood metrics feature would allow to reclassify some of the points based on neighbouring classifications. Lets say we have a lasfile with many points classified as overlap points (class 12), but I would like to use these data points in areas of higher density (due to overlapping flight paths). One way would be to reclassify the entire point cloud, but I was wondering if we could simply assign the classification based on neighbouring points (with all the problems this may have)? If I am correct we could provide a classification value and additionally flag points as overlap with LAS 1.4.

I think so far there is nothing like a metric for the mode of classification, so I hyphotetically called it "cmode", which is also redundant in the case of k=1.

```r
overlap <- delete_points(keep_class(12))
nm <- neighborhood_metrics(locations = overlap, metrics = "cmode", k = 1, r = 0.5, filter = "-drop_class 12") 

pipeline <- reader_las() + nm + write_las()
exec(pipeline, on = f)
```

Jean-Romain, 2024-06-07:
I implemented `neighborhood_metrics()`. It is currently NOT exported because I'm not sure of the implementation and the details. I'd like the stage to be more generic and do more than only some metrics at the location of the local maximum. Moreover I did not test it extensively. Yet it works and you can test it. The documentation is below. I'm closing this issue. If you have any trouble please report in a new issue.

```
#' Compute metrics for a neighborhood
#'
#' This stage calculates specified metrics for a given neighborhood. Currently the neighborhood to be
#' "local_maximum" stage. For each local maximum found by the local maximum stage, it searches for the
#' points in neighborhood and computes metrics using these points.
#'
#' @section Operators:
#' Each string is composed of two parts separated by an underscore. The first part is the attribute
#'  on which the metric must be computed (e.g., z, intensity, classification).
#' The second part is the name of the metric (e.g., mean, sd, cv). A string thus typically looks like
#' `"z_max"`, `"intensity_min"`, `"z_mean"`, `"classification_mode"`.\cr\cr
#' The available attributes are accessible via a single letter or via their lowercase name: t - gpstime,
#' a - angle, i - intensity, n - numberofreturns, r - returnnumber, c - classification,
#' s - synthetic, k - keypoint, w - withheld, o - overlap (format 6+), u - userdata, p - pointsourceid,
#' e - edgeofflightline, d - scandirectionflag, R - red, G - green, B - blue, N - nir.\cr\cr
#' The available metric names are: count, max, min, mean, median, sum, sd, cv, pX (percentile), aboveX, and mode.
#' Some metrics have an attribute + name + a parameter X, such as `pX` where `X` can be substituted by a number.
#' Here, `z_pX` represents the Xth percentile; for instance, `z_p95` signifies the 95th
#' percentile of z. `z_aboveX` corresponds to the percentage of points above X (sometimes called canopy cover).\cr\cr
#' It is possible to call a metric without the name of the attribute. In this case, z is the default.
#'
#' @param metrics Character vector. "min", "max" and "count" are accepted as well as many others
#' (see section 'Operators'). If `NULL` nothing is computed.
#' @param neighborhood Currently support only a "local_maximum" stage.
#' @param k,r integer and numeric respectively for k-nearest neighbours and radius of the neighborhood
#' sphere. If k is given and r is missing, computes with the knn, if r is given and k is missing
#' computes with a sphere neighborhood, if k and r are given computes with the knn and a limit on the
#' search distance.
#' @param ofile A file path where the output will be stored. Default is a temporary GeoPackage file.
#'
#' @template return-vector
#'
#' @examples
#' f <- system.file("extdata", "MixedConifer.las", package = "lasR")
#' read <- reader_las()
#' lmf <- local_maximum(5, ofile = "")
#' nnm = lasR:::neighborhood_metrics(lmf, k = 10, metrics = c("i_mean", "count"))
#' ans <- exec(read + lmf + nnm, on = f)
#' ans
#' @noRd
```

wiesehahn, 2026-01-19:
You said that you would like to have it more generic, at least for `local_maximum` in the meantime we have an option to store this info as attribute (https://github.com/r-lidar/lasR/pull/199). 

So might it be an option to add two filters to the function? One to define the starting points (centers) of the neighborhood and one (optional) to define the points which are considered for the metrics if in range. This looks more generic to me. 

So e.g. to calculate the metrics above based on all non-ground points in 10m radius around local maxima would look like this:

```
pipeline <- 
  add_extrabytes("uchar", "lm", "Local maxima") +
  local_maximum(ws = 5, store_in_attribute = "lm") +
  rasterize(res = 10,
            operators = "HAG_max",
            filter = "lm == 1") +
  nm = lasR:::neighborhood_metrics(centers = "lm == 1", r = 10, metrics = c("i_mean", "count"), filter = "Classification != 2")
```