r-lidar/lasR #50: VPC has slightly different extent than LASheader

wiesehahn, 2024-05-15:
When creating a VPC I experienced that the max_x and max_y values are slightly different than they are in the LASheader. 
Is this due to some reprojection? I thought that `proj::bbox` would be read directly from the header and only the item geometry has to be reprojected?!

``` r
library(lasR)

f <- "L:/luftbilder/ni/digitalbefliegungen/befliegungen_2023/solling_nlf/03_laz_ground_noise/2023_06_solling_32_533_5727.laz"

t <- lidR::readLASheader(f)
print(t)
#> File signature:           LASF 
#> File source ID:           0 
#> Global encoding:
#>  - GPS Time Type: Standard GPS Time 
#>  - Synthetic Return Numbers: no 
#>  - Well Know Text: CRS is WKT 
#>  - Aggregate Model: false 
#> Project ID - GUID:        00000000-0000-0000-0000-000000000000 
#> Version:                  1.4
#> System identifier:        LAStools (c) by rapidlasso GmbH 
#> Generating software:      lasnoise (230330) academic 
#> File creation d/y:        289/2023
#> header size:              375 
#> Offset to point data:     3115 
#> Num. var. length record:  5 
#> Point data format:        8 
#> Point data record length: 59 
#> Num. of point records:    54426088 
#> Num. of points by return: 26936068 16408565 7822735 2572271 588175 94986 3078 202 8 0 0 0 0 0 0 
#> Scale factor X Y Z:       0.001 0.001 0.001 
#> Offset X Y Z:             0 5e+06 0 
#> min X Y Z:                533000 5727000 273.292 
#> max X Y Z:                534000 5728000 423.078 
#> Variable Length Records (VLR):
#>    Variable Length Record 1 of 5 
#>        Description: by LAStools of rapidlasso GmbH 
#>        Tags:
#>           Key 1024 value 1 
#>           Key 3072 value 25832 
#>           Key 3076 value 9001 
#>           Key 4099 value 9001 
#>           Key 4096 value 7837 
#>    Variable Length Record 2 of 5 
#>        Description: RIEGL Extra Bytes 
#>        Extra Bytes Description:
#>           Distance: Distance
#>           Group: Group
#>           Normal: Normal vector 2+15+15 bits
#>    Variable Length Record 3 of 5 
#>        Description:  
#>        Extra Bytes Description:
#>           Distance: Distance
#>           Group: Group
#>           Normal: Normal vector 2+15+15 bits
#>    Variable Length Record 4 of 5 
#>        Description: TerraScan Extra Bytes 
#>        Extra Bytes Description:
#>           Distance: Distance
#>           Group: Group
#>           Normal: Normal vector 2+15+15 bits
#>    Variable Length Record 5 of 5 
#>        Description: by LAStools of rapidlasso GmbH 
#>        WKT OGC COORDINATE SYSTEM: COMPD_CS["ETRS89 / ETRS89 / UTM 32N + DHHN2016",PROJCS["ETRS89 / ETRS8 [...] (truncated)
#> Extended Variable Length Records (EVLR):  void
t$`Min X`
#> [1] 533000
t$`Min Y`
#> [1] 5727000
t$`Max X`
#> [1] 534000
t$`Max Y`
#> [1] 5728000

ans <-exec(write_vpc(tempfile(fileext = ".vpc")), on = f)
j <- yyjsonr::read_geojson_file(ans)
j$`proj:bbox`
#> [1] "[533000.0,5727000.0,533999.999,5727999.999]"
```

<sup>Created on 2024-05-15 with [reprex v2.1.0](https://reprex.tidyverse.org)</sup>

Jean-Romain, 2024-05-15:
R never displays the exact values but rather rounded values with 8 digits. Retry with `options(digits = 10)`. The bbox in the vpc file is printed with 3 digits after the decimal.

``` r
x = 533999.999
x
#> [1] 534000
options(digits = 10)
x
#> [1] 533999.999
```

wiesehahn, 2024-05-15:
ah yes, and because I printed it as a string in one case there was the difference, didnt think of this. thanks for the explanation.

closed 2024-05-15 by Jean-Romain

wiesehahn, 2024-05-15:
I have to revisit this again. 

Actually the geometry of VPC (above I was talking about the attribute `proj:bbox`) is slighly different than the extent defined in the LASheader. And it seems to be more than just a rounding issue.

As it is reprojected I am not sure if it has to do with some projection inaccuracies. 

In the example below in my understanding `lasextent` and `vpcextent` should be the same, but they are not.

``` r
library(lasR)

f <- "/2024_01.1_solling_32_548_5725.laz"

fileheader <- lidR::readLASheader(f)
print(fileheader)
#> File signature:           LASF 
#> File source ID:           0 
#> Global encoding:
#>  - GPS Time Type: Standard GPS Time 
#>  - Synthetic Return Numbers: no 
#>  - Well Know Text: CRS is WKT 
#>  - Aggregate Model: false 
#> Project ID - GUID:        00000000-0000-0000-0000-000000000000 
#> Version:                  1.4
#> System identifier:         
#> Generating software:      TerraScan 
#> File creation d/y:        123/2024
#> header size:              375 
#> Offset to point data:     2822 
#> Num. var. length record:  3 
#> Point data format:        7 
#> Point data record length: 42 
#> Num. of point records:    72117685 
#> Num. of points by return: 33631860 18217797 12388393 6186347 1639714 48032 5177 357 8 0 0 0 0 0 0 
#> Scale factor X Y Z:       0.001 0.001 0.001 
#> Offset X Y Z:             2e+06 6500000 0 
#> min X Y Z:                548000 5725000 228.937 
#> max X Y Z:                549000 5726000 902.383 
#> Variable Length Records (VLR):
#>    Variable Length Record 1 of 3 
#>        Description:  
#>        Tags:
#>           Key 3072 value 25832 
#>    Variable Length Record 2 of 3 
#>        Description:  
#>        WKT OGC COORDINATE SYSTEM: PROJCRS["ETRS89 / UTM zone 32N",BASEGEOGCRS["ETRS89",ENSEMBLE["Europea [...] (truncated)
#>    Variable Length Record 3 of 3 
#>        Description: RIEGL Extra Bytes 
#>        Extra Bytes Description:
#>           Amplitude: Echo signal amplitude [dB]
#>           Reflectance: Echo signal reflectance [dB]
#>           Deviation: Pulse shape deviation
#> Extended Variable Length Records (EVLR):  void

lasextent <- sf::st_bbox(c(xmin = fileheader$`Min X`, xmax = fileheader$`Max X`, ymax = fileheader$`Max Y`, ymin = fileheader$`Min Y`), crs = lidR::st_crs(fileheader))
lasextent
#>    xmin    ymin    xmax    ymax 
#>  548000 5725000  549000 5726000

ans <- exec(write_vpc(tempfile(fileext = ".vpc")), on = f)

j <- yyjsonr::read_geojson_file(ans)
j$`proj:bbox`
#> [1] "[548000.0,5725000.0,548999.999,5725999.999]"

geom <- sf::st_transform(j$geometry, crs = lidR::st_crs(fileheader))

vpcextent <- sf::st_bbox(geom)
vpcextent
#>      xmin      ymin      xmax      ymax 
#>  547990.6 5725000.0  549009.6 5726000.0
```

Jean-Romain, 2024-05-15:
In summary:

1. bbox recorded in the header of the LAS files is written **as is** in `proj:bbox`
2. bbox recorded in the header of the LAS files is written **reprojected** in WGS 84 in `geometry:coordinates`
3. You took the long/lat coordinates and reprojected back to the CRS of the LAS file
4. You have ~10 cm of error on `x`

I'm not a specialist of projections but I'm not surprised more than that. 

1. *WGS84 provided positional accuracy in the order of one to two metres.* ([ref](https://skybrary.aero/articles/world-geodetic-system-1984-wgs84)). By doing the round trip UTM 32N -> WGS 84 -> UTM 32N you probably accumulated the errors

2. In the VPC file created by QGIS you provided to me in #34 we can see that we have slightly different coordinates
   - lasR   [ [-111.20402**8964**,34.4576**59149**], [-111.203051179,34.457659149], [-111.203051179,34.458471515], [-111.204028964,34.458471515], [-111.204028964,34.457659149] ]
   - QGIS [[-111.20402**756076804**, 34.4576**61674533846**], [-111.2040295335688, 34.45847241226331],[-111.203049776868, 34.458474039458004],[-111.20304781353978, 34.457663301687525],[-111.20402756076804, 34.457661674533846]]
 
Which one is correct? None! Let's have a look at the raw values vs QGIS VPC file vs lasR vpc file in QGIS. In Yellow the UTM bbox, in red the QGIS WGS 84 bbox, in green the lasR WGS 84 bbox. The offset is ~10-12 cm which is exactly what you are observing

[image]

Why QGIS and lasR have different output? I have no idea. Both are using GDAL. Is it an operating system dependent issue? (your vpc built on windows, mine on linux). Is it a minor difference in the code? I don't know but the accuracy is similar.

Jean-Romain, 2024-05-15:
My bad you have 10 meters of error. I can't explain easily 10 meters of error. But first I would check where the error comes from.

Jean-Romain, 2024-05-15:
I may have found the issue. I was confused between https://epsg.io/4979 and https://epsg.io/4326. I don't know the difference. On my tests it changes nothing. Does it change something for you now? I'm using `EPSG:4326`

Jean-Romain, 2024-05-15:
> EPSG:4979 is also based on the WGS84 ellipsoid, and differs from EPSG:4326 only in that it uses a 3D coordinate system, whereas 4326 uses a 2D coordinate system

So no, it is not the problem.

Jean-Romain, 2024-05-15:
I don't know what to say.

``` r
library(sf)
#> Linking to GEOS 3.11.1, GDAL 3.6.4, PROJ 9.1.1; sf_use_s2() is TRUE

bbox = c(xmin = 548000, ymin = 5725000, xmax = 549000, ymax = 5726000)
bbox = st_bbox(bbox)
bbox = st_as_sfc(bbox)
utm = st_crs(25832)
wgs = st_crs(4326)

st_crs(bbox) = utm

st_bbox(st_transform(bbox, wgs))
#>      xmin      ymin      xmax      ymax 
#>  9.694165 51.673836  9.708766 51.682913
st_bbox(st_transform(st_transform(bbox, wgs), utm))
#>    xmin    ymin    xmax    ymax 
#>  548000 5725000  549000 5726000
```

wiesehahn, 2024-05-15:
``` r
library(lasR)

f <- "/2024_01.1_solling_32_548_5725.laz"

fileheader <- lidR::readLASheader(f)
print(fileheader)
#> File signature:           LASF 
#> File source ID:           0 
#> Global encoding:
#>  - GPS Time Type: Standard GPS Time 
#>  - Synthetic Return Numbers: no 
#>  - Well Know Text: CRS is WKT 
#>  - Aggregate Model: false 
#> Project ID - GUID:        00000000-0000-0000-0000-000000000000 
#> Version:                  1.4
#> System identifier:         
#> Generating software:      TerraScan 
#> File creation d/y:        123/2024
#> header size:              375 
#> Offset to point data:     2822 
#> Num. var. length record:  3 
#> Point data format:        7 
#> Point data record length: 42 
#> Num. of point records:    72117685 
#> Num. of points by return: 33631860 18217797 12388393 6186347 1639714 48032 5177 357 8 0 0 0 0 0 0 
#> Scale factor X Y Z:       0.001 0.001 0.001 
#> Offset X Y Z:             2e+06 6500000 0 
#> min X Y Z:                548000 5725000 228.937 
#> max X Y Z:                549000 5726000 902.383 
#> Variable Length Records (VLR):
#>    Variable Length Record 1 of 3 
#>        Description:  
#>        Tags:
#>           Key 3072 value 25832 
#>    Variable Length Record 2 of 3 
#>        Description:  
#>        WKT OGC COORDINATE SYSTEM: PROJCRS["ETRS89 / UTM zone 32N",BASEGEOGCRS["ETRS89",ENSEMBLE["Europea [...] (truncated)
#>    Variable Length Record 3 of 3 
#>        Description: RIEGL Extra Bytes 
#>        Extra Bytes Description:
#>           Amplitude: Echo signal amplitude [dB]
#>           Reflectance: Echo signal reflectance [dB]
#>           Deviation: Pulse shape deviation
#> Extended Variable Length Records (EVLR):  void

lasextent <- sf::st_bbox(c(xmin = fileheader$`Min X`, xmax = fileheader$`Max X`, ymax = fileheader$`Max Y`, ymin = fileheader$`Min Y`), crs = lidR::st_crs(fileheader))
lasextent
#>    xmin    ymin    xmax    ymax 
#>  548000 5725000  549000 5726000

lasextentwgs84 <- sf::st_transform(sf::st_as_sfc(lasextent), crs = lidR::st_crs(4979))
sf::st_bbox(lasextentwgs84)
#>      xmin      ymin      xmax      ymax 
#>  9.694165 51.673836  9.708766 51.682913

lasextentreprojected <- sf::st_transform(lasextentwgs84, crs = lidR::st_crs(fileheader))
sf::st_bbox(lasextentreprojected)
#>    xmin    ymin    xmax    ymax 
#>  548000 5725000  549000 5726000
```

just checked exactly the same :)

So reprojection is not the issue

Jean-Romain, 2024-05-16:
What are the coordinates the VPC file? It should be `9.694165 51.673836  9.708766 51.682913`

wiesehahn, 2024-05-16:
The geometry in VPC is written not just as an extent with `xmin`, `ymin`, `xmax`, `ymax` but it is written as a bounding box geometry with coordinates.  for `lower left`, `lower right`,  `upper right`, `upper left`, `lower left`.

And there are slight differences between the VPC created in 
lasR 
```
"geometry": {"coordinates": [[[9.694164960, 51.673921883, 0], [9.708765883, 51.673921883, 0], [9.708765883, 51.682826291, 0], [9.694164960, 51.682826291, 0], [9.694164960, 51.673921883, 0] ]], "type": "Polygon"}
```
and PDAL
```
"geometry":  {"coordinates": [[[9.694164960268289, 51.673921882842684, 228.937], [9.694302443808501, 51.682912666123364, 228.937], [9.708765883038515, 51.682826291169825, 902.383], [9.70862553607275, 51.673835535568685, 902.383], [9.694164960268289, 51.673921882842684, 228.937]]], "type": "Polygon"}
```

while the order is `lower left`, `lower right`,  `upper right`, `upper left`, `lower left` in lasR,
 it is `lower left`,  `upper left`,  `upper right`, `lower right`, `lower left` in PDAL.

But this alone makes not the difference for the extent. Also we see that lower left and upper right coordiantes are the same, while lower right and upper left differ. 

I think this is because in lasR lower right and upper left are populated with values from the extent as well 

This is the geometries next to each other on a map in EPSG:25832 (red: lasR, blue: PDAL)
[image]

[image]

Jean-Romain, 2024-05-16:
> I think this is because in lasR lower right and upper left are populated with values from the extent in source EPSG as well, but instead should be calculated in WGS84 as well.
I am not entirely sure but the following images provide some insights (see orthogonal lines).

I think you got it right. This was my immediate through when seeing your image. We can't reproject a rectangle using only the two corners. That make sense once said but I missed it.

Jean-Romain, 2024-05-16:
I pushed a commit in which the 4 corners of the rectangle are re-projected individually. Now, when displayed in QGIS the VPC produced by `lasR` matches exactly the UTM bbox for the `MixedConifer.las` file. The VPC generated by PDAL (the one you shared with me in an other issue) is still a little bit off (~15-30 cm).

wiesehahn, 2024-05-16:
BTW: you might want to populate min and max Z as well

Jean-Romain, 2024-05-16:
Sure I will. In the current state I have no access to this value in the code of write_vpc. It need some minor refactoring first.

wiesehahn, 2024-05-16:
I can confirm the coordinates look correct now. 
However, the json is not valid any more (under windows). `href` is e.g. `C:\path\file.laz` which leads to 

```
Error in yyjsonr::read_geojson_file(ans) : 
  Error parsing JSON file 'C:\Users\jwiesehahn\AppData\Local\Temp\RtmpOCjoa3\file156041f952644.vpc': invalid escaped character in string code: 10 at position: 2815
```

This was not the case in the example in https://github.com/r-lidar/lasR/issues/34. And I think might be introduced with the commit to no longer forces the pipeline to read the files?!

Jean-Romain, 2024-05-16:
Weird. I did not touch anything related to the file in the VPC code. Maybe this commit https://github.com/r-lidar/lasR/commit/4350584a48d27e3e94b1d55ebbab8c65f879ed86. In this commit I normalize the path using `normalizePath` to ensure the path has a valid format. Maybe the behavior on Windows lead to unexpected results. But units tests are ok. That is my best guess. I will checkout tomorrow. I pushed a commit that reverts this modification. It is works for you then we have the problem. `normalizePath()` being important, I must find a fix at the c++ level.

wiesehahn, 2024-05-17:
> I pushed a commit that reverts this modification. If it works for you then we have the problem.

With this revert I can confirm it works again

Jean-Romain, 2024-05-17:
Ok I pushed a commit that restores `normalizePath` and change `\` to `/` in the c++ code. It shoud work on windows.