r-lidar/lasR #52: invalid VPC due to path under windows

wiesehahn, 2024-05-17:
As mentioned in https://github.com/r-lidar/lasR/issues/50#issuecomment-2116003542

> the json is not valid any more (under windows). href is e.g. C:\path\file.laz which leads to

revert the normalize_path commit fixes this on general. 

E.g. this works
```r
exec(write_vpc(vpcfile), on = "C:/folder/")
```
producing `"href": "C:/folder/lasfile.laz`

however this is still invalid
```r
exec(write_vpc(vpcfile), on = "C:/folder")
```
producing `"href": "C:/folder\lasfile.laz`

Jean-Romain, 2024-05-17:
Fixed few seconds ago

Jean-Romain, 2024-05-17:
Well no. It no longer compiles...

Jean-Romain, 2024-05-17:
Also `zmin` ans `zmax` are written in the file. Which of the 4 corners of the rectangle are assigned `zmin` and `zmax`? I copied the structure of the vpc file produced by QGIS but I think it does not matter.

wiesehahn, 2024-05-17:
can confirm its fixed. 

> Which of the 4 corners of the rectangle are assigned `zmin` and `zmax`?

Actually I have no clue neither.

closed 2024-05-17 by wiesehahn