r-lidar/lasR #53: writing spatial index to disk

wiesehahn, 2024-05-17:
When running a pipeline on data without spatial index the files will be indexed on the fly. 

> X files do not have a spatial index. Spatial indexing speeds up tile buffering and spatial queries drastically.
Files will be indexed on-the-fly. This will take some extra time now but will speed up everything later.

To make the last sentence even more true I think it would make sense (maybe optionally) to not just index files temporarily but instead write the index to disk. Otherwise spatial indexing (which takes quite long) is performed again each time we process the data. And apart from minimally increased storage size I see no disadvantage. 

So would it be possible to store the index as `.lax` file or even better incorporated to the `.laz` file?

Jean-Romain, 2024-05-17:
This is exactly what is happening. If you do not have spatial indexing files,  they are written on disk along with your las/laz files. Did you check your directory? A bug is always possible.

> or even better incorporated to the .laz file?

This was the original behavior. Then I reverted back to standalone `.lax` because I wanted to avoid questions like this one :wink: . Many people are barely aware or not aware at all of all these spatial indexing questions. The process being enforced I preferred to create visible new files rather than invisible embedded files to highlight that *something important happened*. What I can propose for seasoned point cloud users like you is the ability to call the stage `write_lax()` explicitly.

```r
pipeline = write_lax(embedded = TRUE) + reader_las() + dtm()
```

wiesehahn, 2024-05-17:
I guess it's a bug then. I can check later but I am pretty sure no separate lax files. Indexing was done multiple times when running a new pipeline on the same data in the same R Session. 
A dedicated stage might make sense as part of a workflow.

Jean-Romain, 2024-05-17:
> Indexing was done multiple times when running a new pipeline on the same data in the same R Session.

A bug definitively

wiesehahn, 2024-05-17:
Is there a way to find out via R if a file is indexed ?

I thought `lidR::las_check()` might do or it might even be stated in the header but seems its not.

Jean-Romain, 2024-05-17:
No, `las_check()` tests the validity of the LAS file with respect to the specification. The header contains no information about spatial indexation, this is not part of the specification. However you have `lidR::is_indexed()` but this function is outdated. Checks for the presence of `lax` files but cannot test for embedded index in `laz` nor for the newer `copc.laz`.

`lidR` needs an update on this aspect. And `lasR` internally has up to date tools but nothing exposed to the used. However when writing a VPC file you can see a non standard attribute `indexed` added by lasR to be able to know if the file is indexed without reading the file.

Jean-Romain, 2024-05-17:
Actually I have an internal function `lasR:::is_indexed(f)`. You can use it.

Jean-Romain, 2024-05-17:
Can you please run that

```r
tmpdir = paste0(tempdir(), "/testlax")
if (!dir.exists(tmpdir)) dir.create(tmpdir)

# create a datset without lax
f = paste0(system.file(package="lasR"), "/extdata/bcts/")
f = list.files(f, pattern = "(?i)\\.la(s|z)$", full.names = TRUE)
z = file.copy(f, tmpdir)
f = list.files(tmpdir, pattern = "(?i)\\.la(s|z)$", full.names = TRUE)

indexed = lasR:::is_indexed(f)
indexed

# between two tiles with a buffer
pipeline = reader_las_circles(885150, 629400, 10, buffer = 5L) + lasR:::nothing()
ans = exec(pipeline, on = f)

indexed = lasR:::is_indexed(f)
indexed

pipeline = local_maximum(3)
exec(pipeline, on = f)

lax = list.files(tmpdir, pattern = "(?i)\\.lax$", full.names = TRUE)
lax

pipeline = local_maximum(3)
exec(pipeline, on = f)
```

wiesehahn, 2024-05-17:
This seems to work

``` r
library(lasR)

tmpdir = paste0(tempdir(), "/testlax")
if (!dir.exists(tmpdir)) dir.create(tmpdir)

# create a datset without lax
f = paste0(system.file(package="lasR"), "/extdata/bcts/")
f = list.files(f, pattern = "(?i)\\.la(s|z)$", full.names = TRUE)
z = file.copy(f, tmpdir)
f = list.files(tmpdir, pattern = "(?i)\\.la(s|z)$", full.names = TRUE)

indexed = lasR:::is_indexed(f)
indexed
#> [1] FALSE FALSE FALSE FALSE

# between two tiles with a buffer
pipeline = reader_las_circles(885150, 629400, 10, buffer = 5L) + lasR:::nothing()
ans = exec(pipeline, on = f)
#> 4 files do not have a spatial index. Spatial indexing speeds up tile buffering and spatial queries drastically.
#> Files will be indexed on-the-fly. This will take some extra time now but will speed up everything later.

indexed = lasR:::is_indexed(f)
indexed
#> [1]  TRUE  TRUE FALSE FALSE

pipeline = local_maximum(3)
exec(pipeline, on = f)
#> 2 files do not have a spatial index. Spatial indexing speeds up tile buffering and spatial queries drastically.
#> Files will be indexed on-the-fly. This will take some extra time now but will speed up everything later.
#> Simple feature collection with 9849 features and 0 fields
#> Geometry type: POINT
#> Dimension:     XYZ
#> Bounding box:  xmin: 885022.8 ymin: 629157.8 xmax: 885228.9 ymax: 630219.4
#> z_range:       zmin: 326.4 zmax: 418.46
#> Projected CRS: NAD83 / BC Albers
#> # A tibble: 9,849 x 1
#>                            geom
#>                     <POINT [m]>
#>  1 Z (885069.5 629361.8 333.72)
#>  2   Z (885026 629297.9 338.16)
#>  3  Z (885035.8 629299.1 332.1)
#>  4 Z (885070.3 629359.8 331.66)
#>  5   Z (885059 629259.3 345.86)
#>  6 Z (885051.4 629259.1 340.77)
#>  7 Z (885023.4 629299.5 337.56)
#>  8 Z (885036.4 629301.6 331.65)
#>  9   Z (885032 629300.8 337.98)
#> 10 Z (885041.5 629257.8 347.98)
#> # i 9,839 more rows

lax = list.files(tmpdir, pattern = "(?i)\\.lax$", full.names = TRUE)
lax
#> [1] "C:\\Users\\JWIESE~1\\AppData\\Local\\Temp\\RtmpWmEPQ0/testlax/bcts_1.lax"
#> [2] "C:\\Users\\JWIESE~1\\AppData\\Local\\Temp\\RtmpWmEPQ0/testlax/bcts_2.lax"
#> [3] "C:\\Users\\JWIESE~1\\AppData\\Local\\Temp\\RtmpWmEPQ0/testlax/bcts_3.lax"
#> [4] "C:\\Users\\JWIESE~1\\AppData\\Local\\Temp\\RtmpWmEPQ0/testlax/bcts_4.lax"

pipeline = local_maximum(3)
exec(pipeline, on = f)
#> Simple feature collection with 9849 features and 0 fields
#> Geometry type: POINT
#> Dimension:     XYZ
#> Bounding box:  xmin: 885022.8 ymin: 629157.8 xmax: 885228.9 ymax: 630219.4
#> z_range:       zmin: 326.4 zmax: 418.46
#> Projected CRS: NAD83 / BC Albers
#> # A tibble: 9,849 x 1
#>                            geom
#>                     <POINT [m]>
#>  1 Z (885069.5 629361.8 333.72)
#>  2   Z (885026 629297.9 338.16)
#>  3  Z (885035.8 629299.1 332.1)
#>  4 Z (885070.3 629359.8 331.66)
#>  5   Z (885059 629259.3 345.86)
#>  6 Z (885051.4 629259.1 340.77)
#>  7 Z (885023.4 629299.5 337.56)
#>  8 Z (885036.4 629301.6 331.65)
#>  9   Z (885032 629300.8 337.98)
#> 10 Z (885041.5 629257.8 347.98)
#> # i 9,839 more rows
```

<sup>Created on 2024-05-17 with [reprex v2.1.0](https://reprex.tidyverse.org)</sup>

Jean-Romain, 2024-05-17:
The code I shared is part of my unit tests and indeed it works. So there is something else we must identify on your side.

wiesehahn, 2024-05-17:
ok, I might got it. 
I performed the pipeline with a VPC as input. But in the VPC it is hardcoded that `"index:indexed": false`. 
So no matter if there are indices in the meantime or not it will calculate them. Does this make sense?

Jean-Romain, 2024-05-17:
Yes ! nice catch ! Ok that one was not obvious and I will need to think hard to find how to solve it. The goal of using a VPC is to NOT read the files. But to know if it is indexed I do need to read the file.

Jean-Romain, 2024-05-17:
So, I thought about it and there is not real big problem. Yes the VPC explicitly  state that the file are not indexed. This has for effect to automatically add a emit a warning and add a `write_lax()` stage but the `write_lax()` stage does not re-index the files. So in practice you have a false positive warning and it looks like the files are re-indexed because `write_lax()` is called but they are not. The overhead is minimal.

Yet, we have a false positive warning that I can't fix easily and there is another known issue: if the VPC file is not written by lasR there is no way to know if the files are indexed without actually reading the files.

Jean-Romain, 2024-05-18:
In v0.6.0 (devel branch) there is now a `write_lax` stage that can be explicitly used. This does not solve the issue with the warning and the VPC files but offers a new features.

wiesehahn, 2024-05-18:
> So, I thought about it and there is no real big problem. Yes the VPC explicitly states that the files are not indexed. This has for effect to automatically emit a warning and add a `write_lax()` stage but the `write_lax()` stage does not re-index the files. So, in practice you have a false positive warning and it looks like the files are re-indexed because `write_lax()` is called but they are not. The overhead is minimal.

Not sure if I understood this. So if it would be stated that files are not indexed then write_lax is called, which does not run if it is evaluated that files are actually indexed and is run if they are actually unindexed. But as you state below files have to be read to evaluate this, correct? So reading files would be the overhead.
If the file can be edited it would be good to update this in either this case or the case that unindexed are indexed.

I really just came into this situation because I built the VPC in a first step, which I would say is the intended way. So I think it would make sense to evaluate if it is indexed or not and then index them at this stage (`write_vpc`), then in the normal case this situation would not happen if VPC is created from lasR.

> 
> Yet, we have a false positive warning that I can't fix easily and there is another known issue: if the VPC file is not written by lasR there is no way to know if the files are indexed without actually reading the files.

Ah ok, `"index:indexed"` was not in the VPC spec but you added it. Might make sense to have this in the point cloud extension (`"pc:indexed"`)?!
Do you know if there is a reason that this is not defined e.g. in the LASheader? Sounds like this would make things easier not just here. 

> 
> The best I can think about today is to either write `true` in the VPC or nothing. If there is no `index` attribute in the VPC then `lasR` could emit a more accurate warning such as:
> 
> > Impossible to determine if the point cloud is spatially indexed using the VPC file. Spatial indexing speeds up tile buffering and spatial queries drastically. Files will be indexed on-the-fly if they are not. This may take some extra time now but will speed up everything later.

Jean-Romain, 2024-05-18:
>  So reading files would be the overhead.

We need to read the header, not the entire file. So, yes there is an overhead but minor. It matters more for very large collection of files. Try `readLAScatalog()` on 10.000 files. This is the basically the overhead. It is virtually null compared to the actual processing time.

> Ah ok, "index:indexed" was not in the VPC spec but you added it. Might make sense to have this in the point cloud extension ("pc:indexed")?!

Yes why not. Ill check the spec to see what I can do.

> Do you know if there is a reason that this is not defined e.g. in the LASheader? Sounds like this would make things easier not just here.

It is. Reading the headers you can see that the file is a copc file or you can see that there is an embedded lax. For standalone lax it consists only in checking if there is a second files with the same name but the lax extension. The problem is the following: reading the header of a large collection (e.g. 10.000) files takes times. Easily 1 or 2 minutes on a HDD, less on a SSD. Having a VPC files allows to build the collection internally in a 10th of a second. But if we need to read the headers of all the files anyway there is not reason to have a VPC file. You can simply provide the path to the folder with your files. In this case all the headers are read. This is the problem I'm trying to solve.

wiesehahn, 2024-05-21:
> I really just came into this situation because I built the VPC in a first step, which I would say is the intended way. So I think it would make sense to evaluate if it is indexed or not and then [...]

``` r
library(lasR)
f <- list.files(path = "L:/path_to_files", pattern = "*.laz$", full.names = TRUE)
lasR:::is_indexed(f)
#>  [1] FALSE FALSE FALSE FALSE FALSE FALSE FALSE FALSE FALSE FALSE FALSE FALSE
#> [13] FALSE FALSE FALSE FALSE FALSE FALSE FALSE FALSE FALSE FALSE FALSE FALSE
#> [25] FALSE FALSE FALSE FALSE FALSE FALSE FALSE FALSE

pipeline = write_vpc( tempfile(fileext = ".vpc"))
exec(pipeline, on = f)
#> [1] "C:\\Users\\JWIESE~1\\AppData\\Local\\Temp\\RtmpCqsHbX\\file80ac2c326925.vpc"

lasR:::is_indexed(f)
#>  [1] FALSE FALSE FALSE FALSE FALSE FALSE FALSE FALSE FALSE FALSE FALSE FALSE
#> [13] FALSE FALSE FALSE FALSE FALSE FALSE FALSE FALSE FALSE FALSE FALSE FALSE
#> [25] FALSE FALSE FALSE FALSE FALSE FALSE FALSE FALSE
```

just wanted to add an example for the case mentioned above which I guess would be the simplest entrypoint to reduce this problem.

Also I am wondering is there a limit for minimal size when to start indexing? E.g. I expected this to invoke indexing as well. 
``` r
library(lasR)
f <- c("Path/without_index_1.laz",
       "Path/without_index_2.laz",
       "Path/without_index_3.laz",
       "Path/without_index_4.laz")

lasR:::is_indexed(f)
#> [1] FALSE FALSE FALSE FALSE

pipeline <- reader_las() + chm(50, ofile = tempfile(fileext = ".tif"))
exec(pipeline, on = f)
#> class       : SpatRaster 
#> dimensions  : 80, 9, 1  (nrow, ncol, nlyr)
#> resolution  : 50, 50  (x, y)
#> extent      : 547550, 548000, 5724000, 5728000  (xmin, xmax, ymin, ymax)
#> coord. ref. : ETRS89 / UTM zone 32N (EPSG:25832) 
#> source      : file16e40567d63f5.tif 
#> name        : max

lasR:::is_indexed(f)
#> [1] FALSE FALSE FALSE FALSE
```

<sup>Created on 2024-05-21 with [reprex v2.1.0](https://reprex.tidyverse.org)</sup>

Jean-Romain, 2024-05-21:
> just wanted to add an example for the case mentioned above which I guess would be the simplest entrypoint to reduce this problem.

What do you mean? Do you suggest `write_vpc` to be associated with `write_lax` by default?  My point is that a VPC should be created in a 10th of second withtout changing anything. Adding an on-the-fly indexation stage would really break this.

> Also I am wondering is there a limit for minimal size when to start indexing? E.g. I expected this to invoke indexing as well.

It stars indexing at two files **if necessary**. Here you are in a special case where CHM is the only stage and the pipeline can be streamed. Some stage are streamable, this mean that one point is loaded at a time and passes through the pipeline. Not the entire point cloud. There is no need to buffer the tiles because we are loading only a single point at a time. It uses virtually 0 memory and is faster. Thus `write_lax` is not called. No need of spatial index to build a CHM.

wiesehahn, 2024-05-21:
Ah yes, if it is about speed of this stage and there are streamable stages which dont make use of indexes this makes sense. 
I thought about the case where indexes are created on-the-fly in a later stage anyway and VPC might serve as input for multiple pipelines in the future. In this case it would have been more intuitive for me if the indexing would be appllied at the very beginning when the VPC is created (although this means files have to be read here and in a later stage). But with a dedicated `write_lax` stage this would be possible anyway if preferred by the user, so forget about it. :)

Jean-Romain, 2024-05-21:
I indeed think that 

```r
pipeline = `write_lax() + write_vpc()`
```

Is the best option. Yet, in the current state it won't work because the code is a actually a little like

```r
ctg = readLAScatalog()
for (file in ctg$filenames)
   write_lax(file)
write_vpc(ctg)
```

As you can see the `ctg` is loaded first but not updated to change the indexed/non indexed state of the file. Thus `false` will be written anyway because it was the state of the files when reading them. This is something I will improve in 0.6.0.

Jean-Romain, 2024-05-22:
v0.6.0 supports

```r
pipeline = write_lax(embedded = TRUE) + write_vpc()
```

The last issue is the false positive warning with VPC files.

Jean-Romain, 2024-05-22:
Done. If something is not working properly please open a new issue.

closed 2024-05-22 by Jean-Romain

wiesehahn, 2024-05-22:
So providing a VPC with no information about the index (created outside lasR) or with wrong information `"index:indexed": false` (because it was created in lasR prior to indexation) invokes the indexation if necessary, but files are not actually indexed because `write_lax()` evaluates that there is already an index, correct?

I would be favour of your suggestion earlier, as this might be better for interoperability

> The best I can think about today is to either write true in the VPC or nothing. If there is no index attribute in the VPC then lasR could emit a more accurate warning

Might also be good to either add/update information about the index in the VPC automatically or provide some information how this would be done and what are the benfits.

In the case of `"index:indexed": false` the attribute prefix `index:`  stands for the STAC extension (e.g. similar to `"pc:count":` for the point cloud extension). So it might make sense to call it `lasr:` as there is no index extension and you could possible add further attributes later which are related to lasR ?!

wiesehahn, 2024-05-22:
> Maybe you are right I can still change it. But it is not specific to lasR actually. One option more generic would be `"pc:index": "lax"`. This offers more freedom.

The problem is that item properties of `pc:` are defined in the [Point Cloud Extension Specification](https://github.com/stac-extensions/pointcloud?tab=readme-ov-file#item-properties) which is referenced under 
```json
"stac_extensions": [
      "https://stac-extensions.github.io/pointcloud/v1.0.0/schema.json"
]
```
While this works I guess to be valid this has to be in the specification otherwise it points to it but its not defined.

Jean-Romain, 2024-05-22:
I do agree but what are we allowed to change freely? `index:indexed` was my original attempt for this reason.

wiesehahn, 2024-05-22:
I dont know if this is validated somewhere where it could cause trouble. 
I think `index:indexed` is better than `pc:` (as there is a clear spec) and equal to `lasr:` (as both dont exist). I just mentioned `lasr:` as I thought this might be more clear as where this comes from and gives the possibility to add more attributes later if necessary without using new prefixes (which might cause trouble if this is already defined as e.g. `label:`) 
(existing extensions see https://stac-extensions.github.io/)