r-lidar/lasR #59: Variable window size in `local_maximum()`

wiesehahn, 2024-05-22:
For many applications in forestry one important processing step is the individual tree detection. And while there are some methods available, in my opinion a universal good working solution is still an unsolved problem. 

Local maximum method is one of the simplest and most widely used method which is currently also the only implemented method in lasR.
At current state a fixed window size is the only option. In lidR `ws` can also be a function to adapt the search window by e.g. stand height to achieve better results in hereogeneous forest stands. 

As far as I understand it is not easy at all to implement this (custom functions defined in R) in C++. Do you think it would be possible or make sense to define a relative simple linear interpolation function to optionally consider minimum and maximum window size depending on height? Or is there some restraints in how this is processed in internally?

With something like the following basic function we would not have the same versatility as we have in lidR, yet we have some option to consider the tendency of larger crown sizes with increasing height.

``` r
f <- function(h, h_min, h_max, ws_min, ws_max) {
  ws <- (ws_min - h_min * ((ws_max -ws_min) / (h_max - h_min))) + h * ((ws_max -ws_min) / (h_max - h_min))
  ws[h < h_min] <- ws_min
  ws[h > h_max] <- ws_max
  return(ws)
}

heights <- seq(0,30,0.5)
ws <- f(h = heights, h_min = 5, h_max = 25, ws_min = 3, ws_max = 12)
plot(heights, ws, type = "l",  ylim = c(0,15))
```

[image]

Jean-Romain, 2024-05-22:
> As far as I understand it is not easy at all to implement this (custom functions defined in R) in C++. 

This is correct but not the actual reason why I removed variable windows size. There are several reasons why there is no variable windows size actually. One reason is that lasR is intended to be decoupled from R. The R version being only an API to something that can also be a standalone software or a python package. If I inject R code, this will no longer be possible. I want `lasR` to be usable without R (except stages `rasterize()` and `callback()` that are specialized R additions)

> Do you think it would be possible or make sense to define a relative simple linear interpolation function to optionally consider minimum and maximum window size depending on height?

Absolutely. This would be possible and the very best options. So why I did not do it? Because variable windows size (vws) is much slower (x20 slower in order of magnitude). Try to benchmark both fixed and variable sizes with lidR on a point cloud (with 4.1.1). With the constant increasing of dataset sizes, vws is impracticable in my opinion on a point cloud. It was initially designed for raster.

Is is possible to make vws faster? Maybe but this requires some additional development. Currently this is very low in my priority list but as I told you in my email this could be changed against funding.

wiesehahn, 2024-05-22:
Thanks for the explanation. 

> I want lasR to be usable without R

Thats why I thought more about defining the function on the C++ side with arguments (e.g. min and max ws) passed from R or whichever API. 

Btw: There is a need for better (computationally efficient, more accurate, able to detect trees in multiple canopy layers) tree detection algorithms anyway. I would be happy to contribute if anyone is about to do this.  :)

agrigoriev, 2026-10-13:
@Jean-Romain this was implemented in https://github.com/r-lidar/lasR/pull/325. 
I guess we can close this one.

closed 2026-10-13 by Jean-Romain