r-lidar/lasR #62: `neighborhood_metrics()` does not return metrics when using wildcards for ofile

wiesehahn, 2024-06-10:
I implemented `neighborhood_metrics()` in the `devel`  branch. It is currently NOT exported because I'm not sure of the implementation and the details. I'd like the stage to be more generic and do more than only some metrics at the location of the local maximum. Moreover I did not test it extensively. Yet it works and you can test it. The documentation is below. I'm closing this issue. If you have any trouble please report in a new issue.

```
#' Compute metrics for a neighborhood
#'
#' This stage calculates specified metrics for a given neighborhood. Currently the neighborhood to be
#' "local_maximum" stage. For each local maximum found by the local maximum stage, it searches for the
#' points in neighborhood and computes metrics using these points.
#'
#' @section Operators:
#' Each string is composed of two parts separated by an underscore. The first part is the attribute
#'  on which the metric must be computed (e.g., z, intensity, classification).
#' The second part is the name of the metric (e.g., mean, sd, cv). A string thus typically looks like
#' `"z_max"`, `"intensity_min"`, `"z_mean"`, `"classification_mode"`.\cr\cr
#' The available attributes are accessible via a single letter or via their lowercase name: t - gpstime,
#' a - angle, i - intensity, n - numberofreturns, r - returnnumber, c - classification,
#' s - synthetic, k - keypoint, w - withheld, o - overlap (format 6+), u - userdata, p - pointsourceid,
#' e - edgeofflightline, d - scandirectionflag, R - red, G - green, B - blue, N - nir.\cr\cr
#' The available metric names are: count, max, min, mean, median, sum, sd, cv, pX (percentile), aboveX, and mode.
#' Some metrics have an attribute + name + a parameter X, such as `pX` where `X` can be substituted by a number.
#' Here, `z_pX` represents the Xth percentile; for instance, `z_p95` signifies the 95th
#' percentile of z. `z_aboveX` corresponds to the percentage of points above X (sometimes called canopy cover).\cr\cr
#' It is possible to call a metric without the name of the attribute. In this case, z is the default.
#'
#' @param metrics Character vector. "min", "max" and "count" are accepted as well as many others
#' (see section 'Operators'). If `NULL` nothing is computed.
#' @param neighborhood Currently support only a "local_maximum" stage.
#' @param k,r integer and numeric respectively for k-nearest neighbours and radius of the neighborhood
#' sphere. If k is given and r is missing, computes with the knn, if r is given and k is missing
#' computes with a sphere neighborhood, if k and r are given computes with the knn and a limit on the
#' search distance.
#' @param ofile A file path where the output will be stored. Default is a temporary GeoPackage file.
#'
#' @template return-vector
#'
#' @examples
#' f <- system.file("extdata", "MixedConifer.las", package = "lasR")
#' read <- reader_las()
#' lmf <- local_maximum(5, ofile = "")
#' nnm = lasR:::neighborhood_metrics(lmf, k = 10, metrics = c("i_mean", "count"))
#' ans <- exec(read + lmf + nnm, on = f)
#' ans
#' @noRd
```

_Originally posted by @Jean-Romain in https://github.com/r-lidar/lasR/issues/49#issuecomment-2155077149_

When using wildcards in `ofile` argument the calculated metrics are not exported.

``` r
library(lasR)
f <- system.file("extdata", "MixedConifer.las", package = "lasR")
read <- reader_las()
lmf <- local_maximum(5, ofile = "")
nnm = lasR:::neighborhood_metrics(lmf, k = 10, metrics = c("i_mean", "count"), ofile = paste0(tempdir(), "/*_ttops.gpkg"))
ans <- exec(read + lmf + nnm, on = f)
ans
#> Simple feature collection with 177 features and 0 fields
#> Geometry type: POINT
#> Dimension:     XYZ
#> Bounding box:  xmin: 481260 ymin: 3812921 xmax: 481349.8 ymax: 3813011
#> z_range:       zmin: 2.42 zmax: 32.07
#> Projected CRS: NAD83 / UTM zone 12N
#> # A tibble: 177 x 1
#>                          geom
#>                   <POINT [m]>
#>  1 Z (481304.9 3812944 18.28)
#>  2 Z (481309.9 3812944 22.55)
#>  3 Z (481307.5 3812954 19.76)
#>  4 Z (481346.9 3812973 23.29)
#>  5 Z (481264.9 3813009 24.14)
#>  6 Z (481302.8 3812955 20.93)
#>  7 Z (481295.9 3812976 24.95)
#>  8 Z (481297.8 3812992 24.37)
#>  9 Z (481313.6 3812940 23.93)
#> 10 Z (481263.5 3812958 23.92)
#> # i 167 more rows
```

``` r

f <- system.file("extdata", "MixedConifer.las", package = "lasR")
read <- reader_las()
lmf <- local_maximum(5, ofile = "")
nnm = lasR:::neighborhood_metrics(lmf, k = 10, metrics = c("i_mean", "count"), ofile = tempfile(fileext = "ttops.gpkg"))
ans <- exec(read + lmf + nnm, on = f)
ans
#> Simple feature collection with 177 features and 2 fields
#> Geometry type: POINT
#> Dimension:     XYZ
#> Bounding box:  xmin: 481260 ymin: 3812921 xmax: 481349.8 ymax: 3813011
#> z_range:       zmin: 2.42 zmax: 32.07
#> Projected CRS: NAD83 / UTM zone 12N
#> # A tibble: 177 x 3
#>    i_mean count                       geom
#>     <dbl> <dbl>                <POINT [m]>
#>  1   91      10 Z (481309.9 3812944 22.55)
#>  2   83.5    10 Z (481304.9 3812944 18.28)
#>  3   89      10 Z (481307.5 3812954 19.76)
#>  4  101.     10 Z (481264.9 3813009 24.14)
#>  5  104.     10 Z (481346.9 3812973 23.29)
#>  6   59.1    10 Z (481302.8 3812955 20.93)
#>  7   61.1    10 Z (481297.8 3812992 24.37)
#>  8  109.     10 Z (481313.6 3812940 23.93)
#>  9   83.8    10 Z (481295.9 3812976 24.95)
#> 10   72.3    10 Z (481263.5 3812958 23.92)
#> # i 167 more rows
```

<sup>Created on 2024-06-10 with [reprex v2.1.0](https://reprex.tidyverse.org)</sup>

Jean-Romain, 2024-06-10:
I don't see any problem here. The first case produced `MixedConifer_ttops.gpkg` with 177 trees, the second case produced `ttops.gpkg` with 177 trees.

Jean-Romain, 2024-06-10:
Ho ok I got it !

Jean-Romain, 2024-06-10:
This is not an issue with `neighborhood_metrics` but rather with how attributes are handle. We can reproduce with `local_maximum` only

``` r
library(lasR)
f <- system.file("extdata", "MixedConifer.las", package = "lasR")
read <- reader_las()
lmf <- local_maximum(5, ofile = paste0(tempdir(), "/*_ttops.gpkg"), record_attributes = T)
ans <- exec(read + lmf, on = f)
ans
#> Simple feature collection with 177 features and 0 fields
#> Geometry type: POINT
#> Dimension:     XYZ
#> Bounding box:  xmin: 481260 ymin: 3812921 xmax: 481349.8 ymax: 3813011
#> z_range:       zmin: 2.42 zmax: 32.07
#> Projected CRS: NAD83 / UTM zone 12N
#> # A tibble: 177 × 1
#>                          geom
#>                   <POINT [m]>
#>  1 Z (481304.9 3812944 18.28)
#>  2 Z (481346.9 3812973 23.29)
#>  3 Z (481309.9 3812944 22.55)
#>  4 Z (481307.5 3812954 19.76)
#>  5 Z (481302.8 3812955 20.93)
#>  6 Z (481295.9 3812976 24.95)
#>  7   Z (481301 3812964 22.47)
#>  8 Z (481264.9 3813009 24.14)
#>  9 Z (481297.8 3812992 24.37)
#> 10 Z (481263.5 3812958 23.92)
#> # ℹ 167 more rows
```

``` r

f <- system.file("extdata", "MixedConifer.las", package = "lasR")
read <- reader_las()
lmf <- local_maximum(5, ofile = paste0(tempdir(), "/ttops.gpkg"), record_attributes = T)
ans <- exec(read + lmf, on = f)
ans
#> Simple feature collection with 177 features and 5 fields
#> Geometry type: POINT
#> Dimension:     XYZ
#> Bounding box:  xmin: 481260 ymin: 3812921 xmax: 481349.8 ymax: 3813011
#> z_range:       zmin: 2.42 zmax: 32.07
#> Projected CRS: NAD83 / UTM zone 12N
#> # A tibble: 177 × 6
#>    Intensity gpstime ReturnNumber Classification ScanAngle
#>        <int>   <dbl>        <int>          <int>     <dbl>
#>  1       100 150748.            1              1        -8
#>  2       141 151388.            1              1        -4
#>  3       125 152207.            1              1        10
#>  4        94 150749.            1              1        -4
#>  5       103 150748.            1              1        -7
#>  6       103 151389.            1              1        -8
#>  7       108 150748.            1              1        -5
#>  8       117 150748.            1              1        -6
#>  9        81 152207.            1              1        13
#> 10       124 151389.            1              1        -5
#> # ℹ 167 more rows
#> # ℹ 1 more variable: geom <POINT [m]>
```

<sup>Created on 2024-06-10 with [reprex v2.1.0](https://reprex.tidyverse.org)</sup>

commit 59d486b1 (2024-06-10): fix #62
closed 2024-06-10 by Jean-Romain