r-lidar/lasR #64: rasterize operators fail on RGB

robmusktppl, 2024-06-20:
Hi Jean-Romain

I'm a new user of your excellent lasR package. 

I'm trying to rasterize the RGB attributes of a las file but the operators appear only to process the first channel, i.e. the outputs are all equal.

This is the rasterize call: ```chmRGB <- rasterize(0.05, operators = c('R_max', 'G_max', 'B_max'))```

Am I missing something? I have attached a sample las file and this is my script:

```
# c:\r\r-4.4.0\bin\r.exe
version
library(lasR)
library(lidR)
packageVersion('lasR')
packageVersion('lidR')
f <- 'D:/Taswood/Towerhill UAV Trial/LAS/TPPL_Day_1_Area_2_5ms_clip_qha.las'
lasf <- readLAS(f)
summary(lasf@data[, .(R,G,B)])
plot(lasf, color = 'RGB')
read <- reader_las()
chmRGB <- rasterize(0.05, operators = c('R_max', 'G_max', 'B_max'))
pipeline <- read + chmRGB
anslasf <- exec(pipeline, on = f)
plot(anslasf)
summary(anslasf[[1]][])
summary(anslasf[[2]][])
summary(anslasf[[3]][])
```

It returns the following:
```
> version
               _
platform       x86_64-w64-mingw32
arch           x86_64
os             mingw32
crt            ucrt
system         x86_64, mingw32
status
major          4
minor          4.0
year           2024
month          04
day            24
svn rev        86474
language       R
version.string R version 4.4.0 (2024-04-24 ucrt)
nickname       Puppy Cup
> library(lasR)
Warning message:
package 'lasR' was built under R version 4.5.0 
> library(lidR)
lidR 4.1.1 using 6 threads. Help on <gis.stackexchange.com>. Bug report on <github.com/r-lidar/lidR>.
> packageVersion('lasR')
[1] '0.6.2'
> packageVersion('lidR')
[1] '4.1.1'
> f <- 'D:/Taswood/Towerhill UAV Trial/LAS/TPPL_Day_1_Area_2_5ms_clip_qha.las'
> lasf <- readLAS(f)
> summary(lasf@data[, .(R,G,B)])
       R               G               B        
 Min.   :    0   Min.   :    0   Min.   :    0
 1st Qu.: 9984   1st Qu.:12288   1st Qu.: 8960
 Median :16640   Median :19200   Median :12800
 Mean   :22492   Mean   :24279   Mean   :14973
 3rd Qu.:32256   3rd Qu.:34304   3rd Qu.:19200
 Max.   :65280   Max.   :65280   Max.   :62720
> plot(lasf, color = 'RGB')
> read <- reader_las()
> chmRGB <- rasterize(0.05, operators = c('R_max', 'G_max', 'B_max'))
> pipeline <- read + chmRGB
> anslasf <- exec(pipeline, on = f)
> plot(anslasf)
> summary(anslasf[[1]][])
     R_max       
 Min.   :    0
 1st Qu.:12544
 Median :19712
 Mean   :25671
 3rd Qu.:37376
 Max.   :65280
 NA's   :221692
> summary(anslasf[[2]][])
     G_max       
 Min.   :    0
 1st Qu.:12544
 Median :19712
 Mean   :25671
 3rd Qu.:37376
 Max.   :65280
 NA's   :221692
> summary(anslasf[[3]][])
     B_max       
 Min.   :    0
 1st Qu.:12544
 Median :19712
 Mean   :25671
 3rd Qu.:37376
 Max.   :65280
 NA's   :221692
```
Thanks in advance,

Jean-Romain, 2024-06-20:
Fixed! Thank you for reporting. Metrics on RGB were indeed computed on RRR.

Jean-Romain, 2024-06-20:
Btw if you install from github it is ok. If you install from r-universe you must wait a solid hour in order to get the latest fix.

closed 2024-06-20 by Jean-Romain

robmusktppl, 2024-06-20:
rightio. thanks for the heads up!

commit fb53023e (2024-06-20): Fix #64