r-lidar/lasR #66: ERROR: WARNING: no payload for LASF_Projection VLR with record_id 2112.

wiesehahn, 2024-07-01:
When trying to create a vpc I get the following error

```r
ans = lasR::exec(lasR::write_vpc(tempfile(fileext = ".vpc")), on = file)

ERROR: WARNING: no payload for LASF_Projection VLR with record_id 2112.
Error: in 'write_vpc' while processing the catalog: Invalid CRS. Cannot write a VPC file
```

I presume this is due to some erroneous CRS encoding. 
Defining the CRS manually works, however I still receive an error.

```r
> ans = lasR::exec(lasR::set_crs(25832) + lasR::write_vpc(tempfile(fileext = ".vpc")), on = file)
ERROR: WARNING: no payload for LASF_Projection VLR with record_id 2112.
ERROR: WARNING: no payload for LASF_Projection VLR with record_id 2112.
```

Is this expected?

The file header looks as following:

```r
> lidR::readLASheader(file)
WARNING: no payload for LASF_Projection VLR with record_id 2112.
File signature:           LASF 
File source ID:           0 
Global encoding:
 - GPS Time Type: Standard GPS Time 
 - Synthetic Return Numbers: no 
 - Well Know Text: CRS is GeoTIFF 
 - Aggregate Model: false 
Project ID - GUID:        00000000-0000-0000-0000-000000000000 
Version:                  1.3
System identifier:        LAStools (c) by rapidlasso GmbH 
Generating software:      las2las64 (version 220828) 
File creation d/y:        340/2022
header size:              235 
Offset to point data:     289 
Num. var. length record:  1 
Point data format:        1 
Point data record length: 28 
Num. of point records:    57559238 
Num. of points by return: 27005945 14927868 9906298 5266573 427329 
Scale factor X Y Z:       0.001 0.001 0.001 
Offset X Y Z:             415115 5543054 0 
min X Y Z:                411000 5545000 71.713 
max X Y Z:                412000 5546000 384.508 
Variable Length Records (VLR):
   Variable Length Record 1 of 1 
       Description: intentionally empty OGC WKT 
       WKT OGC COORDINATE SYSTEM:  [...] (truncated)
Extended Variable Length Records (EVLR):  void

> lidR::readLASheader(file)@VLR
$LASF_Projection
$LASF_Projection$reserved
[1] 0

$LASF_Projection$`user ID`
[1] "LASF_Projection"

$LASF_Projection$`record ID`
[1] 2112

$LASF_Projection$`length after header`
[1] 0

$LASF_Projection$description
[1] "intentionally empty OGC WKT"

```

Jean-Romain, 2024-07-02:
`write_vpc` is not a *real* stage in the pipeline. In 

```r
ans = lasR::exec(lasR::set_crs(25832) + lasR::write_vpc(tempfile(fileext = ".vpc")), on = file)
```

`write_vpc` is actually executed before `set_crs` which, in your case, lead to unexpected behavior. But once you understand why it makes sense.

A `lasR` pipeline applies on a point cloud. You have a point cloud that passes through stages and you transform the point cloud or you produced derived products. `write_vpc` DOES NOT apply to point clouds but rather to the existing files read from the disk. What would be the behavior of:

```r
pipeline = reader_las() + delete_points(filter = drop_z_below(1.3)) + write_vpc()
```

`write_vpc()` cannot applies to the point cloud with points above 1.3. This dataset does not even exist on disk and will never exist. It necessarily apply on the collection of existing files i.e on `files` in 

```r
ans = exec(pipeline, on = files)
```

Thus you understand that `write_vpc` is not a stage, strictly speaking, but rather a preprocessing step that can be ran before the actual processing. And this is exactly what is happening. This is why `set_crs()` does not applies on it.

From the user side I do agree that you are expecting your code to work. Can I add a workaround? I will think about it.

wiesehahn, 2024-07-02:
> write_vpc is actually executed before set_crs

I think this is not true, as both pipelines (with and without `set_crs()`) lead to different behaviour and results. I think `set_crs()` modifies the header in memory and then `write_vpc()` uses this in-memory information to create the vpc. 

> write_vpc is not a stage, strictly speaking, but rather a preprocessing step that can be ran before the actual processing

This seems even more true for `set_crs()` to me.

The thing I am wondering about is the error message, but I guess it is invoked when reading the file. So there is a warning (missing crs) which we fix on-the-fly with `set_crs()` and then use this version to create the VPC.

wiesehahn, 2024-07-02:
BTW: 
Lets say we want to set the CRS not only in memory but actually write this to the file to make them valid. We could do `lasR::exec(lasR::set_crs(25832) + lasR::write_las(tempfile(fileext = ".laz")), on = file)`, however this does re-write the entire file, would it be possible to update the header information without re-writing the entire point cloud data?

Jean-Romain, 2024-07-02:
> would it be possible to update the header information without re-writing the entire point cloud data?

No, you can't add a CRS without re-writing the files. You can happen data to a file in place but you can't insert data in place.

> I think this is not true, as both pipelines (with and without set_crs()) lead to different behaviour and results. I think set_crs() modifies the header in memory and then write_vpc() uses this in-memory information to create the vpc.

Well you may be correct. In your first example you have an actual error that stop R. In the second example you don't have any error. You have a print from LASlib without any failure. In the second case the VPC files should have been produced right?

Jean-Romain, 2024-07-02:
So everything is correct right? Except some annoying message that I can't remove because they are important.

wiesehahn, 2024-07-02:
> So everything is correct right? Except some annoying messages that I can't remove because they are important.

alright, no problem on my side. Was just reporting because I thought the error might be relevant for you.

Jean-Romain, 2024-07-02:
On my side it works as expected (it was a little late yesterday when I first replied and missed the actual behavior). Closing.

closed 2024-07-02 by Jean-Romain