r-lidar/lasR #67: Problem with VPC - Error: In the parser while reading the file collection

wiesehahn, 2024-07-04:
write_vpc created the following

```json
{
  "type": "FeatureCollection",
  "features": [
  {
    "type": "Feature",
    "stac_version": "1.0.0",
    "stac_extensions": [
      "https://stac-extensions.github.io/pointcloud/v1.0.0/schema.json",
      "https://stac-extensions.github.io/projection/v1.1.0/schema.json"
     ],
    "id": "MixedConifer",
    "geometry": {
      "coordinates": [
        [ [-111.204028964,34.457659149,0.000], [-111.203049216,34.457660780,0.000], [-111.203051179,34.458471515,32.070], [-111.204030936,34.458469884,32.070], [-111.204028964,34.457659149,0.000] ]
      ],
      "type": "Polygon"
    },
    "bbox": [-111.204030936, 34.457659149, 0.000, 34.458471515, 34.458471515, 32.070],
    "properties": {
      "datetime": "2017-01-01T00:00:00Z",
      "pc:count": 37657,
      "pc:type": "lidar",
      "proj:bbox": [481260.000, 3812921.090, 481349.990, 3813010.990],
      "proj:wkt2": "PROJCRS[\"NAD83 / UTM zone 12N\",BASEGEOGCRS[\"NAD83\",DATUM[\"North American Datum 1983\",ELLIPSOID[\"GRS 1980\",6378137,298.257222101,LENGTHUNIT[\"metre\",1]]],PRIMEM[\"Greenwich\",0,ANGLEUNIT[\"degree\",0.0174532925199433]],ID[\"EPSG\",4269]],CONVERSION[\"UTM zone 12N\",METHOD[\"Transverse Mercator\",ID[\"EPSG\",9807]],PARAMETER[\"Latitude of natural origin\",0,ANGLEUNIT[\"degree\",0.0174532925199433],ID[\"EPSG\",8801]],PARAMETER[\"Longitude of natural origin\",-111,ANGLEUNIT[\"degree\",0.0174532925199433],ID[\"EPSG\",8802]],PARAMETER[\"Scale factor at natural origin\",0.9996,SCALEUNIT[\"unity\",1],ID[\"EPSG\",8805]],PARAMETER[\"False easting\",500000,LENGTHUNIT[\"metre\",1],ID[\"EPSG\",8806]],PARAMETER[\"False northing\",0,LENGTHUNIT[\"metre\",1],ID[\"EPSG\",8807]]],CS[Cartesian,2],AXIS[\"(E)\",east,ORDER[1],LENGTHUNIT[\"metre\",1]],AXIS[\"(N)\",north,ORDER[2],LENGTHUNIT[\"metre\",1]],USAGE[SCOPE[\"Engineering survey, topographic mapping.\"],AREA[\"North America - between 114°W and 108°W - onshore and offshore. Canada - Alberta; Northwest Territories; Nunavut; Saskatchewan. United States (USA) - Arizona; Colorado; Idaho; Montana; New Mexico; Utah; Wyoming.\"],BBOX[31.33,-114,84,-108]],ID[\"EPSG\",26912]]",
      "proj:epsg": 26912,
    },
    "links": [],
    "assets": {
      "data": {
      "href": "./../../renv/library/windows/R-4.4/x86_64-w64-mingw32/lasR/extdata/MixedConifer.las",
      "roles": ["data"]
      }
    }
  }
  ]
}
```

which produces an error when using it in a pipeline and also cannot be read by QGIS 

``` r
library(lasR)

f <- system.file("extdata", "MixedConifer.las", package="lasR")
ans <- exec(write_vpc(tempfile(fileext = ".vpc")), on = f)
exec(summarise(), on = ans)
#> Error: In the parser while reading the file collection: JSON parsing error: [json.exception.parse_error.101] parse error at line 26, column 5: syntax error while parsing object key - unexpected '}'; expected string literal
```

<sup>Created on 2024-07-04 with [reprex v2.1.0](https://reprex.tidyverse.org)</sup>

See also https://github.com/r-lidar/lasR/issues/34
I am pretty confident it worked since then.

Jean-Romain, 2024-07-04:
Lucky you, I just fixed it few second ago. I also added `use_gpstime` in `write_vpc()` (see documentation).

Jean-Romain, 2024-07-04:
It fails on Windows. Let me fix it first.

Jean-Romain, 2024-07-04:
Fixed

closed 2024-07-04 by Jean-Romain