r-lidar/lasR #77: smaller file size with `sort_points(spatial = TRUE)`

wiesehahn, 2024-07-18:
I just tried to compress LAS files with `sort_points(spatial = TRUE)` and without.

In the documentation it is stated that
> This increases data locality, speeds up spatial queries, but slightly increases the final size of the files when compressed in LAZ format.

In my case the opposite is true, spatially sorted files are more compact than without applying the stage, Maybe this is not true for all cases, but generally saying this increases file size seems also not true. 

```r
> file.size(sorted)
[1] 168612089 370205154 292094084 194992644

> file.size(unsorted)
[1] 201293174 441217478 352011215 243785893
```

Jean-Romain, 2024-07-18:
`sorted` is after `sort_point()`. But what is `unsorted`? The original file shared by the data provider?

wiesehahn, 2024-07-18:
> `sorted` is after `sort_point()`.

Not sure what you mean here. 

> But what is `unsorted`? The original files shared by the data provider? Or with `spatial = FALSE`? 

I converted uncompressed LAS files with 
1.  `pipeline = set_crs(25832) + write_las(ofile = paste0(folder, "/*_unsorted.laz")),`
2. `pipeline = set_crs(25832) + sort_points(spatial = TRUE) + write_las(ofile = paste0(folder, "/*_sorted.laz"))`

I you want I could sent you the uncompressed file and you try yourself

wiesehahn, 2024-07-18:
https://e.pcloud.link/publink/show?code=XZGhmgZlqjfjm586PBf60hnac5onkWJgF4y

Jean-Romain, 2024-07-18:
I think you misinterpreted the documentation. You are comparing the original files given by the data provider with the optimized one where points were sorted. Unless the data provider already provided well sorted files, yes, you have a significant gain. What the documentation says is that `sort_point(spatial = FALSE)` will allows more compression than `sort_points(spatial = TRUE)`. But `sort_points(spatial = TRUE)` should allow faster spatial queries.

Jean-Romain, 2024-07-18:
By the way, it not related to your question but you could do:

```r
pipeline = set_crs(25832) + write_las(ofile = paste0(folder, "/*_unsorted.laz")) + sort_points(spatial = TRUE) + write_las(ofile = paste0(folder, "/*_sorted.laz"))
```

In order to get optimally fast processes in `lasR` it is important to think in term of complex pipelines where everything is done in a single execution when possible.

Jean-Romain, 2024-07-18:
Your original point cloud looks already well sorted for what I can see visually. I'm surprised you have a significant gain of 30 MB (15%) but cool.

- Original file after compression: 201 MB
- Sorted `spatial = FALSE` : 168.0 MB (-15%)
- Sorted `spatial = TRUE` : 168.6 MB

wiesehahn, 2024-07-18:
Did you use the file I provided in the link above? I get different sizes...

``` r
exec(
  pipeline = set_crs(25832) + write_las(ofile = paste0(folder, "/*_unsorted.laz")) + sort_points(spatial = TRUE) + write_las(ofile = paste0(folder, "/*_sorted.laz")),
  on = folder
)

round(file.size(list.files(folder, pattern = "*.las$", full.names = T))/1024.0 ^ 2, 2) 
#> [1] 1118.58

round(file.size(list.files(folder, pattern = "*_unsorted.laz$", full.names = T))/1024.0 ^ 2, 2) 
#> [1] 191.97

round(file.size(list.files(folder, pattern = "*_sorted.laz$", full.names = T))/1024.0 ^ 2, 2) 
#> [1] 160.8
```

Jean-Romain, 2024-07-18:
I get the same. You reported the size in Mebibytes while I reported in Megabytes

201292568 bytes = 191.97 Mebibytes = 201.29 Megabytes

wiesehahn, 2024-07-18:
ah ok, sorry you are right, I was confused that you reported three sizes for the compressed files. 
Now I also see that `spatial = TRUE` is larger than `spatial = FALSE`. 

I thought that the stage is just for the case that we want to sort spatially so that

`write_las("*.laz")` == `sort_points(spatial = FALSE) + write_las("*.laz")`

but now I see that points are not automatically sorted without the stage.

Jean-Romain, 2024-07-18:
`sorting` is a relatively expensive operation. So by default there is no sort. This is a decision of the user to sort or not.

I modified a bit the wording of the documentation to avoid the confusion. I guess I can close.

closed 2024-07-18 by Jean-Romain

Jean-Romain, 2024-07-18:
You are writing new files. They are not supposed to inherit the index of the original file. New files are not indexed. But this is an excellent question. I'll verify. I don't know how embedded index are embedded.

Jean-Romain, 2024-07-18:
I confirm. No problem.