r-lidar/lasR #79: writing raster on subset does not reduce extent

wiesehahn, 2024-07-23:
When we write a raster after e.g. ` reader_las_circles` the resulting raster has the same extent as the underlaying tile (with lots of NA). I expected it to be cropped to the extent of our subsample. Also in this case I thought it would be clipped to a circular area. 

``` r
library(lasR)
f <- system.file("extdata", "Topography.las", package = "lasR")
dtm <- temptif()

pipeline = reader_las_circles(273500, 5274500, 20) + dtm(ofile = dtm)
ans <- exec(pipeline, on = f)

r <- terra::rast(dtm)
terra::plot(r)
```

[image]

``` r
terra::summary(r)
#>  file1167063ae6afc
#>  Min.   :801.3    
#>  1st Qu.:803.2    
#>  Median :807.3    
#>  Mean   :806.5    
#>  3rd Qu.:809.3    
#>  Max.   :810.7    
#>  NA's   :80115

terra::ext(r)
#> SpatExtent : 273357, 273643, 5274357, 5274643 (xmin, xmax, ymin, ymax)

pipeline = dtm(ofile = dtm)
ans <- exec(pipeline, on = f)

r <- terra::rast(dtm)
terra::plot(r)
```

[image]

``` r
terra::summary(r)
#>  file1167063ae6afc
#>  Min.   :789.0    
#>  1st Qu.:802.3    
#>  Median :805.8    
#>  Mean   :805.1    
#>  3rd Qu.:807.5    
#>  Max.   :814.8    
#>  NA's   :621

terra::ext(r)
#> SpatExtent : 273357, 273643, 5274357, 5274643 (xmin, xmax, ymin, ymax)
```

<sup>Created on 2024-07-23 with [reprex v2.1.0](https://reprex.tidyverse.org)</sup>

Jean-Romain, 2024-07-23:
What if you use a wildcard ? I don't have the time to check today and probably not this week. Your use case and expectations are valid yet. Tagging as enhancement.

```r
 dtm(ofile = "/path/to/folder/*_dtm.tif)
```

Jean-Romain, 2024-07-23:
I'm more concerned by the fact that the DTM is squared despite you used `reader_las_circles()`

Jean-Romain, 2024-07-24:
The original behavior was to write a raster with the full extent of the point cloud when using a single output file. This implied mostly empty rasters with large point cloud and small queries

```r
f = paste0(system.file(package="lasR"), "/extdata/bcts/")
f = list.files(f, pattern = "(?i)\\.la(s|z)$", full.names = TRUE)
pipeline = reader_las_circles(c(885150, 885150), c(629300, 629600) , 10, filter = keep_ground()) + rasterize(2, "min")
ans = exec(pipeline, on = f)
```
[image]

The new behavior in `v 0.10.0` (`devel` branch) is to create a raster with the minimal bounding box.

[image]

Which lead to the desired behavior for a single query

```r
pipeline = reader_las_circles(885150, 629300 , 10, filter = keep_ground()) + rasterize(2, "min")
ans = exec(pipeline, on = f)
```

[image]

This behavior makes more sense I guess.

When the file has a wildcard the behavior is to write each query in its own file. This does not change.

closed 2024-07-24 by Jean-Romain

wiesehahn, 2024-07-24:
I wasnt aware that the wildcard solves this, but now its more concise I guess.

Jean-Romain, 2024-07-24:
Well having a wildcard in the path or a regular path was more like

- No wildcard: I want one output file that covers the entire point cloud
- Wildcard: I want multiple output files corresponding to each processed chunk.

But now with this change it corresponds to a more expected and previsible output. No wildcard adjusts the files to the extent of the processed data.

wiesehahn, 2024-07-24:
I just checked again (with 0.9.1) and the sqaure seems to be related to `triangulate` rather than the wildcard I guess

``` r
library(lasR)
f <- system.file("extdata", "Topography.las", package = "lasR")

read = reader_las_circles(273500, 5274500, 20) 
del = triangulate(filter = keep_ground())

pipeline1 = read + dtm(1)
pipeline2 = read + rasterize(1)
pipeline3 = read + del + rasterize(1, del)

ans <- exec(pipeline1, on = f)
terra::plot(ans)
```

[image]

``` r

ans <- exec(pipeline2, on = f)
terra::plot(ans)
```

[image]

``` r

ans <- exec(pipeline3, on = f)
terra::plot(ans)
```

[image]

<sup>Created on 2024-07-24 with [reprex v2.1.0](https://reprex.tidyverse.org)</sup>