r-lidar/lasR #82: automatic selection of optimal parallel processing strategy

wiesehahn, 2024-07-26:
Just an unreflected thought, I am almost certain this is not easy/possible

would it be possible with limited effords to optimize processing time by automatically selecting the optimal parallel processing strategy?

I guess when we run the pipeline on a single file `concurrent_points()` might be a better choice than `concurrent_files()`, which probably would be preferable in many cases if we run it on a folder with several files. Also some stages can make use of `concurrent_points()` or `nested()` while others can't. 
And I think for most users (me included) its not easy to determine which strategy to choose. So if there is some way to automatically determine from the input pipeline which strategy to choose this would be really helpful to realize fast processing I think. 

This goes in the same direction as https://github.com/r-lidar/lasR/discussions/46, setting the correct filter when reading files can also speed up processing but is not a trivial task for most users. 

I think in the current state lasR is fast but in most cases it is not as fast as it could be, because its hard for users to provide the optimal processing pipeline / settings.

Jean-Romain, 2024-07-26:
On large collections this is easy. Use concurrent files. Always! That is why `exec(..., ncores = 4)` is a shortcut to concurrent files.

What is extremely hard to guess is how many cores you can use without ram overflow. It depends on the complexity of the task, the density of the points cloud and so on. I recently process a very big single file with `ncores = 8, chunk = 250`. Hard to guess from a coding point of view but with more test a decent `ncores = "auto"` mode might be possible

wiesehahn, 2024-07-29:
> On large collections this is easy. Use concurrent files. Always! That is why `exec(..., ncores = 4)` is a shortcut to concurrent files.

Yet the default is `concurrent_points(half_cores())`
Wouldn't it make sense to set `concurrent_files(half_cores())` as default if the pipeline is applied on more than X files?

> What is extremely hard to guess is how many cores you can use without ram overflow. It depends on the complexity of the task, the density of the points cloud and so on. I recently processed a very big single file with `ncores = 8, chunk = 250`. Hard to guess from a coding point of view but with more test a decent `ncores = "auto"` mode might be possible

Yeah I thought so, however if it is hard to guess for an algorithm which has access to underlaying hardware parameters, I am afraid it is close to impossible for a user. :/

Jean-Romain, 2024-07-29:
> Wouldn't it make sense to set concurrent_files(half_cores()) as default if the pipeline is applied on more than X files?

No, because it could use more than your available memory. This is the user responsibility to choose a strategy that matches their machine capacity.

> Yeah I thought so, however if it is hard to guess for an algorithm which has access to underlying hardware parameters, I am afraid it is close to impossible for a user. :/

It is easier for a user actually. A user can make a try on 10 files, check if it works well and upscale. On the programmer point of view having a cross platform access to the available memory is not that easy and guessing in advance how much memory the pipeline is going to take is nearly impossible.

Jean-Romain, 2024-07-30:
I'm closing. If someday I find a solution for that request I will do it for sure. But it is definitively not easy.

closed 2024-07-30 by Jean-Romain