r-lidar/lasR #83: Problem with reading tif file using load_raster

lefsky, 2024-07-26:
I am trying to load a raster to put into the tree segmentation routine but it is having problems with the tif file, although gdalinfo reports no problems with the file

===========================
Code
================================
# File is subt_1918_004_003_54b.tif
# Format and file contents have been validated by gdalinfo
# Unused las files are collected and passed to exec as with load_raster example

f <- paste0(system.file(package = "lasR"), "/extdata/bcts/")
f <- list.files(f, pattern = "(?i)\\.la(s|z)$", full.names = TRUE)
r <- list.files('/content', pattern = 'subt_1918_004_003_54b.tif', full.names = TRUE)

chm=load_raster(r,band=1L)
seed = local_maximum_raster(chm, 3)
tree = region_growing(chm, seed, ofile="/content/check4.tif")
pipeline = chm
ans = exec(pipeline, on=f)
================================
The response is: 
================================

Error: in load_raster while initalizing chunk: /content/subt_1918_004_003_54b.tif, band 1: The buffer into which the data should be read is null
Traceback:

1. exec(pipeline, on = "/content")

================================
gdalinfo for the file:
================================
gdalinfo subt_1918_004_003_54b.tif
Driver: GTiff/GeoTIFF
Files: subt_1918_004_003_54b.tif
Size is 120, 120
Coordinate System is:
PROJCRS["NAD27 / BLM 13N (ftUS)",
    BASEGEOGCRS["NAD27",
        DATUM["North American Datum 1927",
            ELLIPSOID["Clarke 1866",6378206.4,294.978698213898,
                LENGTHUNIT["metre",1]]],
        PRIMEM["Greenwich",0,
            ANGLEUNIT["degree",0.0174532925199433]],
        ID["EPSG",4267]],
    CONVERSION["BLM zone 13N (US survey feet)",
        METHOD["Transverse Mercator",
            ID["EPSG",9807]],
        PARAMETER["Latitude of natural origin",0,
            ANGLEUNIT["degree",0.0174532925199433],
            ID["EPSG",8801]],
        PARAMETER["Longitude of natural origin",-105,
            ANGLEUNIT["degree",0.0174532925199433],
            ID["EPSG",8802]],
        PARAMETER["Scale factor at natural origin",0.9996,
            SCALEUNIT["unity",1],
            ID["EPSG",8805]],
        PARAMETER["False easting",1640416.67,
            LENGTHUNIT["US survey foot",0.304800609601219],
            ID["EPSG",8806]],
        PARAMETER["False northing",0,
            LENGTHUNIT["US survey foot",0.304800609601219],
            ID["EPSG",8807]]],
    CS[Cartesian,2],
        AXIS["easting (X)",east,
            ORDER[1],
            LENGTHUNIT["US survey foot",0.304800609601219]],
        AXIS["northing (Y)",north,
            ORDER[2],
            LENGTHUNIT["US survey foot",0.304800609601219]],
    USAGE[
        SCOPE["No official usage."],
        AREA["United States (USA) - between 108°W and 102°W - Colorado; Montana; Nebraska; New Mexico; North Dakota; Oklahoma; South Dakota; Texas; Wyoming."],
        BBOX[28.98,-108,49.01,-102]],
    ID["EPSG",4413]]
Data axis to CRS axis mapping: 1,2
Origin = (1562256.000000000000000,12976694.000000000000000)
Pixel Size = (1.300000479012189,-1.300000479012189)
Metadata:
  AREA_OR_POINT=Area
Image Structure Metadata:
  INTERLEAVE=BAND
Corner Coordinates:
Upper Left  ( 1562256.000,12976694.000) (105d15'48.50"W, 35d44'35.71"N)
Lower Left  ( 1562256.000,12976538.000) (105d15'48.49"W, 35d44'34.16"N)
Upper Right ( 1562412.000,12976694.000) (105d15'46.61"W, 35d44'35.71"N)
Lower Right ( 1562412.000,12976538.000) (105d15'46.60"W, 35d44'34.17"N)
Center      ( 1562334.000,12976616.000) (105d15'47.55"W, 35d44'34.94"N)
Band 1 Block=120x17 Type=Float32, ColorInterp=Gray
  NoData Value=-3.4028235e+38

Jean-Romain, 2024-07-28:
Please provide a minimal reproducible example including data if required. I tested the following (which is basically your pipeline) successfully 

```r
r <- system.file("extdata/bcts", "bcts_dsm_5m.tif", package = "lasR")
f <- paste0(system.file(package = "lasR"), "/extdata/bcts/")
f <- list.files(f, pattern = "(?i)\\.la(s|z)$", full.names = TRUE)
chm = load_raster(r,band=1L)
pipeline = chm
ans = exec(pipeline, on=f)
```

Jean-Romain, 2024-07-30:
I was able to reproduce with a raster that does not match with the point cloud. There is now an error when trying to `load_raster()` and the raster has no overlap with the point cloud. I don't know if it fixes 100% of the issue but in absence of more information I will consider the issue as fixed.

closed 2024-07-30 by Jean-Romain