r-lidar/lasR #86: limited number of geometric features are added

wiesehahn, 2024-07-31:
When we calculate geometric features not all of the defined attributes are added. Is this expected? I guess this is related to the las specification about a limited number of (extrabyte) attributes?

While the documentation says that it is dicouraged to include them all it does not state that there is a hard limit. 

In the example `angle` is added if we define it alone, but in conjuction with principal components it is not added (as the limit is reached?).

``` r
library(lasR)
f <- system.file("extdata", "Example.las", package = "lasR")

pipeline <- geometry_features(k = 8, r = 1.5, features = "i") + write_las()
ans <- exec(pipeline, on = f)
las <- lidR::readLAS(ans)
length(las@data)
#> [1] 17
las@data$angle
#>  [1] 89.18209 80.62259 78.67996 66.39858 78.67996 82.92635 84.34153 78.67996
#>  [9] 76.74328  0.00000 54.46864 81.40933 82.03606 74.55217 68.91040 83.16940
#> [17] 89.99516 82.47673 77.57462 83.24518 83.24518 23.42594 83.24518 23.42594
#> [25] 77.57462 77.57462 89.17646 83.33920 83.33920 83.33920

pipeline <- geometry_features(k = 8, r = 1.5, features = "C") + write_las()
ans <- exec(pipeline, on = f)
las <- lidR::readLAS(ans)
length(las@data)
#> [1] 25

pipeline <- geometry_features(k = 8, r = 1.5, features = "Ci") + write_las()
ans <- exec(pipeline, on = f)
las <- lidR::readLAS(ans)
length(las@data)
#> [1] 25
las@data$angle
#> NULL
```

Jean-Romain, 2024-07-31:
All your features are added properly by `lasR` but then `rlas` (used by `lidR` to read LAS files) has a limit of 10 extra bytes. Here there are exactly 10 extra bytes (`C` + `i` = 9 + 1). Your file is valid but the limit comes from `rlas` and it seems that there is a little bug since I'm expecting `rlas` to read extrabytes  0 to 9 but actually read 0 to 8.

wiesehahn, 2024-07-31:
ok, but does it make sense than, that with the las 1.4 testfile some more attributes are recorded (27) and with an own file in pdrf 8 it is 30? I guess yes because it is just about the extra attributes?

``` r
library(lasR)
f <- system.file("extdata", "las14_pdrf6.laz", package="lasR")
pipeline <- geometry_features(k = 8, r = 1.5, features = "Ci") + write_las()
ans <- exec(pipeline, on = f)
las <- lidR::readLAS(ans)
#> Warning in CPL_crs_from_input(x): GDAL Error 1: compound CRS should have at
#> least 2 components
length(las@data)
#> [1] 27
```

Jean-Romain, 2024-07-31:
Absolutely. All the cores attributes are read. pdrf 8 has RGB compared to pdrf 6 (+3 cores attributes). The limits applies only to extrabytes and only to `rlas/lidR`when loading a `data.frame`. This is a limit that is hard imposed by the `select` argument that allows to select extrabytes with number from 0 to 9

```r
readLAS(file, select = "xyzi024689")
```

Impossible to write `10` it would be interpreted as `0+1`. This limit is there since almost 8 years.

Jean-Romain, 2024-07-31:
I'm closing because it is related to `rlas`/`lidR` and not `lasR`