r-lidar/lasR #88: `sort_points()` hangs with certain files

wiesehahn, 2024-08-08:
Executing a pipeline with the stage `sort_points()` (with or without spatial=TRUE) keeps processing forever with certain files. 

The files have errors such as 
```
1: Invalid data: 905 points with a return number equal to 0 found. 
2: Invalid data: 905 points with a number of returns equal to 0 found. 
```
however, other files with these characteristics are processd normally.

Here is a samplefile and the process which results in this behavior. This is a 1m voxel sample from the original file which also caused the issue, sampling to 10m voxels however did not cause it. 

[lassample.laz](https://e.pcloud.link/publink/show?code=XZs7EgZTLcMAG9b8hYnUk43aI7HtVIjQ9dk)
```r
ans <- lasR::exec(
    lasR::sort_points(spatial = TRUE) +
    lasR::write_las(ofile = lasR::templaz()),
  on = lassample
)
```

Jean-Romain, 2024-08-08:
While your bug requires a fix this is also a limit case. 

All your points are first returns and all your gpstime are equal to 6.367313e-314. The value 6.367313e-314 is a denormalized (subnormal) value in the IEEE 754 double-precision floating-point representation. Denormalized values are used to represent numbers that are very close to zero but are not zero, providing a gradual underflow.

So basically your point cloud IS NOT sortable AND you have weird value for GPStime. Either these values are the real values recorded by the data provider and we have a problem or the file has been corrupted by `lasR` and you should figure out where it happened so I can fix it.

I'm pretty sure that the sorting works even if all the gpstime and return numbers are equal (no sort). Here you gpstime is very special (I did non even know that this value was possible in the standard). It is smaller than the smallest representable value which is 2.225074e-308. It is a very special value out of my computer science knowledge and this is probabaly the reason why it failed. If I'm right I won't fix this bug.

Jean-Romain, 2024-08-08:
Btw on my computer it sorts spatially without infinite loop. Then, inside each spatial cell, the points are sorted by return number only, because of the absence of valid gpstime which is a very bad sort. But it works.

wiesehahn, 2024-08-09:
that sounds pretty technical. 
I guess it really is a very edge case, it was the case in ~0.01% of files. The problem was just that I included this pipeline in a function and because of infinite loop the following steps were not even started for all of the files. 

I am wondering does it make sense that it is infinite loop due to equal gpstimes if I ran into this with all points but not with a small sample (10m voxel), I mean if all pulses are the same shouldnt this also happen in the sample?

If it works on your side its probably related to windows?!

Jean-Romain, 2024-08-09:
Really I don't know. What I see is that:

1. All gpstime equal should not prevent the task completion. You eventually end up with a bad ordering because it should be ordered by gpstime and it is not ordered by gpstime. But it should works.
2. Your gpstime are all 6.367313e-314 which is very special a value that I did know was even possible and which is below the minimum possible storable value. Today I learned about subnormal numbers. These are very special numbers.
3. Even with this value the task completes on my computer. Is it related to Windows? Yes, I was able to reproduce you problem on windows
4. Is is related to subnormal number? Maybe not? I was able to reproduce your issue on windows with all gpstime set to 0. So maybe 6.367313e-314  is not the issue.

wiesehahn, 2024-08-09:
Ok probably lets forget about it and hope it will not affect others 😄

Jean-Romain, 2024-08-09:
Ok I got it. It was my fault and the code did not handle duplicated value properly but for some reason it worked on linux. The weird 6.367313e-314 put me in the wrong direction. I did not try on Windows but I'm confident I fixed it.

closed 2024-08-09 by Jean-Romain

wiesehahn, 2024-08-09:
> Anyway I think you should wonder why you have this weird value.

I gave up wondering about data, its mostly older federal laserscanning campaigns which has the weirdest data formatting of all kinds, and I cannot do much about it anyway. I just come across these while I at least try to give them some kind of workable standard shape (compression, sorting, indexing, naming,...) 

So if you one day developed a way to fix all those weird data (`fix(las_check)`) let me know, ;)

Jean-Romain, 2024-08-09:
I cannot create a magical fix function. Look at the problems you are exposing to me! Today I learn about new kind of numbers in computer science. I can definitively create some validation function. `lidR::las_check()` does a pretty good job by the way and should have noticed you about duplicated gpstime. The difficulty comes when trying to automatically fix the infinite number of possible issues robustly.

wiesehahn, 2024-08-09:
I know, was just kidding, your software is already doing a good job in handling most of these cases.

commit 07acc0b7 (2024-08-09): Fix #88