r-lidar/lasR #89: Unexpected values in CHM resulting from triangulation of dense point cloud

FloFranz, 2024-08-16:
I have a dense point cloud (about 50 p/m²), normalized it, applied some filtering and wanted to compute a triangulation-based CHM. To show my problem, I also included computation of a point-to-raster-based CHM.

```
f <- file/path/to/input/laz
output_dir <- file/path/to/output

norm <- lasR::normalize()

first_filter <- lasR::delete_points(filter = lasR::drop_z_below(-1) + lasR::drop_z_above(55))

noise_classification <- lasR::classify_with_ivf(res = 5, n = 6, class = 18)

second_filter <- lasR::delete_points(filter = lasR::drop_noise())

write_laz <- lasR::write_las(ofile = file.path(output_dir, '/*.laz'))

chm_p2r <- lasR::chm(res = 0.5, tin = F)

chm_tin <- lasR::chm(res = 0.5, tin = T)

pipeline <- norm + first_filter + noise_classification + second_filter + write_laz + chm_p2r + chm_tin

ans <- lasR::exec(pipeline, on = f)
```

The p2r-based CHM looks quite well, but the tin-based CHM shows some unexpected values, e.g. pixels with higher values directly next to pixels with lower values, like noise.

```
par(mfrow = c(1,2))
terra::plot(ans[[2]], main = 'CHM p2r-based on dense point cloud')
terra::plot(ans[[3]], main = 'CHM tin-based on dense point cloud')
```

[image]

I then tried to do the tin-based CHM computation with `lidR` using the already normalized and filtered point cloud written to disk before. I used the `dsmtin()` algorithm one time keeping only the highest point per pixel before triangulation (`highest = T`) and one time using all first returns (`highest = F`).

```
dense_laz <- lidR::readLAS(file/path/to/normalized_filtered_laz}')

chm_tin_highest_T <- lidR::rasterize_canopy(dense_laz, res = 0.5, algorithm = lidR::dsmtin(highest = T))
chm_tin_highest_F <- lidR::rasterize_canopy(dense_laz, res = 0.5, algorithm = lidR::dsmtin(highest = F))
```

The resulting CHM with `highest = T` looks good, but the other one with `highest = F` looks similar to the one computed with `lasR` (even if it's not quite as bad).

```
par(mfrow = c(1,2))
terra::plot(chm_tin_highest_T, main = 'CHM tin highest = True')
terra::plot(chm_tin_highest_F, main = 'CHM tin highest = False')
```

[image]

It is possible to compute a tin-based CHM with `highest = F` in `lidR` without having such noise by thinning the point cloud.

```
decimated_laz <- lidR::decimate_points(dense_laz, algorithm = lidR::highest(0.5))
chm_tin_decimated <- lidR::rasterize_canopy(decimated_laz, res = 0.5, algorithm = lidR::dsmtin(highest = F))

par(mfrow = c(1,1))
terra::plot(chm_tin_decimated, main = 'CHM tin based on decimated point cloud')
```

[image]

To cut a long story short: I only want to understand if the triangulation implemented in `lasR` uses all first returns (as with `highest = F`in lidR`)? Maybe my problem is somehow related to this [issue](https://github.com/r-lidar/lasR/issues/18)? If you want to reproduce it with my laz file I provided it [here](https://drive.google.com/file/d/1jObN6Z7JGEDcO4SnPlAtH-J4EIp6Tk6N/view?usp=drive_link).

Jean-Romain, 2024-08-16:
In `lasR` the triangulation based chm is simply implemented as:

```r
tri = triangulate(filter = keep_first())
ras = rasterize(x, tri)
tri + rast
```

You can verify it easily:

``` r
lasR::chm(tin = TRUE)
#>  -----------
#> triangulate (uid:21db)
#>   max_edge : 0 
#>   filter : -keep_first 
#>   output :  
#>   use_attribute : Z 
#> -----------
#> rasterize (uid:561b)
#>   res : 1 
#>   connect : uid:21db04e6
#>   filter :  
#>   output : /tmp/RtmpuKEeO0/file1a999922b41173.tif 
#> -----------
```

A triangulation-based CHM is not necessarily better than a simple point-to-raster CHM, especially in high-density point clouds. The goal of a TIN-based CHM is to interpolate and fill missing pixels for fine-grained CHMs when dealing with not very dense point clouds. With 50 points/m² and a resolution of 0.5 m, you have on average 12 points per pixel, which is enough to build a point-to-raster CHM.

A strict TIN-based CHM will have more noise because, for a given pixel, it will query the corresponding Z value in the triangulation without integrating the surrounding environment. If you are unlucky, the pixels fall in a low triangle, and you get a pit. Statistically, you will have a lot of pits.

lasR performs a strict triangulation of the first returns. lidR does a bit more by also pre-filtering the highest points, thus returning a slightly better CHM.

Long story short, TIN-based CHMs are more suitable for low-density point clouds. Use a point-to-raster approach; it is much faster and more accurate in high-density point clouds.

FloFranz, 2024-08-16:
Thank you very much for the detailed explanation. It's now clear to me.