r-lidar/lasR #91: Executing pipeline does not finish

FloFranz, 2024-08-23:
Hi Jean-Romain! It's me again. I know about your limited time for answering questions or addressing issues here, but I also remembered this rule of you: “Rule of thumb: If it crashes, it's NEVER your fault. A crash should absolutely not happen.”  (#31) 

Executing the pipeline below, my R crashes or does not end the process, respectively. I've applied it on hundreds of laz files (which were all classified before using LAStools) and I think it only crashes because of one laz file, which is the largest of them all. I'm not sure if the size has anything to do with it or if it's just a coincidence. This laz file is not written to disk and because of that no CHM is calculated. However, all other laz files are processed. If I apply the pipeline to this laz file alone or with a few others, it also works and finishes.

```
f <- file/path/to/input/laz_files
output_dir <- file/path/to/output

normalization <- lasR::normalize()
first_filter <- lasR::delete_points(filter = lasR::drop_z_below(-1) + lasR::drop_z_above(55))
noise_classification <- lasR::classify_with_ivf(res = 5, n = 6L, class = 18L)
second_filter <- lasR::delete_points(filter = lasR::drop_noise())
write_laz <- lasR::write_las(ofile = file.path(output_dir, '/*.laz'))
chm <- lasR::chm(res = 0.5)
pitfill <- lasR::pit_fill(raster = chm, ofile = file.path(output_dir, '/*.tif'))

pipeline <- normalization + first_filter + noise_classification + second_filter + write_laz + chm + pitfill
  
lasR::set_parallel_strategy(lasR::concurrent_files(ncores = lasR::half_cores()))
lasR::exec(pipeline, on = f, progress = T)
```

Below is the summary of the laz file which makes problems and you can find it also [here](https://drive.google.com/file/d/1yfWPbsgaPO8JlhhErvgXvMnO1ztvqVC7/view?usp=sharing). I first thought it could be of memory limitations, but also if I used `lasR::set_parallel_strategy(lasR::concurrent_files(ncores = lasR::half_cores()))`, the problem remains. Do you have any idea why R is not completing the process?

```
> las <- lidR::readLAS(file.path(f, 'name_of_the_laz_file.laz')
> las
class        : LAS (v1.4 format 8)
memory       : 6.9 Gb 
extent       : 539000, 540000, 5737000, 5738000 (xmin, xmax, ymin, ymax)
coord. ref.  : ETRS89 / ETRS89 / UTM 32N + DHHN2016 
area         : 1 km²
points       : 71.2 million points
density      : 71.2 points/m²
density      : 36.66 pulses/m²

> summary(las)
class        : LAS (v1.4 format 8)
memory       : 6.9 Gb 
extent       : 539000, 540000, 5737000, 5738000 (xmin, xmax, ymin, ymax)
coord. ref.  : ETRS89 / ETRS89 / UTM 32N + DHHN2016 
area         : 1 km²
points       : 71.2 million points
density      : 71.2 points/m²
density      : 36.66 pulses/m²
File signature:           LASF 
File source ID:           0 
Global encoding:
 - GPS Time Type: Standard GPS Time 
 - Synthetic Return Numbers: no 
 - Well Know Text: CRS is WKT 
 - Aggregate Model: false 
Project ID - GUID:        00000000-0000-0000-0000-000000000000 
Version:                  1.4
System identifier:        LAStools (c) by rapidlasso GmbH 
Generating software:      lasground_new64 (210720) academ 
File creation d/y:        289/2023
header size:              375 
Offset to point data:     3115 
Num. var. length record:  5 
Point data format:        8 
Point data record length: 59 
Num. of point records:    71202430 
Num. of points by return: 36661097 20456269 9889569 3310705 759966 121687 2946 187 4 0 0 0 0 0 0 
Scale factor X Y Z:       0.001 0.001 0.001 
Offset X Y Z:             0 5e+06 0 
min X Y Z:                539000 5737000 358.918 
max X Y Z:                540000 5738000 536.639 
Variable Length Records (VLR):
   Variable Length Record 1 of 5 
       Description: by LAStools of rapidlasso GmbH 
       Tags:
          Key 1024 value 1 
          Key 3072 value 25832 
          Key 3076 value 9001 
          Key 4099 value 9001 
          Key 4096 value 7837 
   Variable Length Record 2 of 5 
       Description: RIEGL Extra Bytes 
       Extra Bytes Description:
          Distance: Distance
          Group: Group
          Normal: Normal vector 2+15+15 bits
   Variable Length Record 3 of 5 
       Description:  
       Extra Bytes Description:
          Distance: Distance
          Group: Group
          Normal: Normal vector 2+15+15 bits
   Variable Length Record 4 of 5 
       Description: TerraScan Extra Bytes 
       Extra Bytes Description:
          Distance: Distance
          Group: Group
          Normal: Normal vector 2+15+15 bits
   Variable Length Record 5 of 5 
       Description: by LAStools of rapidlasso GmbH 
       WKT OGC COORDINATE SYSTEM: COMPD_CS["ETRS89 / ETRS89 / UTM 32N + DHHN2016",PROJCS["ETRS89 / ETRS8 [...] (truncated)
Extended Variable Length Records (EVLR):  void
```

Jean-Romain, 2024-08-23:
It's going be hard to debug if we cannot reproduce. 

1. Are you 100% sure this is the problematic file? 
2. Did you try with the file and its 8 neighbors?
3. Can you figure out which stage crashed?

FloFranz, 2024-08-23:
All other files were written correctly to disk, so I think this must be the problematic file.
I haven't tried it with its 8 neighbors yet. I will do this and then report back to you.

Jean-Romain, 2024-08-23:
If you know at which file it crashed you can rerun adding this stage: `lasR:::stop_if_chunk_id_below()` at the begining

FloFranz, 2024-08-26:
I've now tested the file and its 8 neighbors and the problem also occurs. Looking at the processing steps printed on the screen, it seems that everything is going well. But it stops during the process `Write LAS` at almost 100% and the file is not written to disk. The file and its 8 neighboring files can be found [here](https://drive.google.com/drive/folders/1VnTDeKvW5rT-A6ETm91sUoNjQxQIHsX8?usp=sharing) to reproduce. I've also testet it with these files but without the one making problems and it works.

Jean-Romain, 2024-08-26:
I got something. When processing 2023_06_solling_32_539_5737.laz (which matches the one you sent me in the first message) the amount of memory used is colossal. It crashed because I ran out of memory.

The density of ground point look relatively similar to other file so I don't know what is the issue yet.

Can you give me the specs of your hardware.

Jean-Romain, 2024-08-26:
Minimal reproducible error

```r
library(lasR)

f <- "~/Téléchargements/issue 91/"
normalization <- lasR::normalize()
pipeline <- lasR:::stop_if_chunk_id_below(7) + reader_las() + normalization
lasR::exec(pipeline, on = f, verbose = T)
```

FloFranz, 2024-08-26:
> Can you give me the specs of your hardware.

- CPU model: AMD Ryzen Threadripper PRO 5975WX 32-Cores, 3600 MHz, 32 cores, 64 logical processors
- amount of RAM: 256 GB
- storage: AMD-RAID Array 1 SCSI Disk Device 953 GB
- GPU: NVIDIA RTX A5000

I run it under Microsoft Windows 10 Pro.

> The density of ground point look relatively similar to other file

Yes it's a bit denser, but not that much.

Jean-Romain, 2024-08-26:
I have 32 GB and I ran out of memory. But you have 256 GB !! Did it crash for a memory overflow on your computer?

Can you test the MRE on the 9 files you shared and track the memory usage? If it crashed because of memory overflow it means it allocates more than 256 GB :open_mouth: which basically means it is blocked in an infinite loop of memory allocation.

FloFranz, 2024-08-26:
Here's the output of the MRE (including lax writing):

```
9 files do not have a spatial index. Spatial indexing speeds up tile buffering and spatial queries drastically.
Files will be indexed on-the-fly. This will take some extra time now but will speed up everything later.
File processing options:
  Read points: true
  Streamable: false
  Buffer: 20.0
  Concurrent files: 1
  Concurrent points: 32
  Chunks: 9

Processing chunk 1/9 in thread 0: 2023_06_solling_32_538_5736
Stage: write_lax
Stage: stop_if
Processing chunk 2/9 in thread 0: 2023_06_solling_32_538_5737
Stage: write_lax
Stage: stop_if
Processing chunk 3/9 in thread 0: 2023_06_solling_32_538_5738
Stage: write_lax
Stage: stop_if
Processing chunk 4/9 in thread 0: 2023_06_solling_32_539_5736
Stage: write_lax
Stage: stop_if
Processing chunk 5/9 in thread 0: 2023_06_solling_32_539_5737
Stage: write_lax
Stage: stop_if
Processing chunk 6/9 in thread 0: 2023_06_solling_32_539_5738
Stage: write_lax
Stage: stop_if
Processing chunk 7/9 in thread 0: 2023_06_solling_32_540_5736
Stage: write_lax
Stage: stop_if
Processing chunk 8/9 in thread 0: 2023_06_solling_32_540_5737
Stage: write_lax
Stage: stop_if
Stage: reader_las
 Number of point read 59432359
Stage: triangulate
Stage: transform_with
Processing chunk 9/9 in thread 0: 2023_06_solling_32_540_5738
Stage: write_lax
Stage: stop_if
Stage: reader_las
 Number of point read 63993942
Stage: triangulate
Stage: transform_with
NULL
```

Memory usage is not as high, only 14-17 GB are allocated. During `transform_with()`, CPU usage is rising sharply to approx. 60%. But at this second `transform_with()` stage at the end, it suddenly interrupts the processing. It's strange...

FloFranz, 2024-08-26:
Ah okay the interrupting because of the `stop_if_chunk_id_below()`.

Jean-Romain, 2024-08-26:
You are not processing in the same order as I do. You skipped `solling_32_539_5737` which is supposed to be the problematic file. And you did not crash. I see the `NULL`  return

FloFranz, 2024-08-26:
Yes got it. Focussing now on the problematic file with your MRE:

```
library(lasR)

f <- "~/Téléchargements/issue 91/"
normalization <- lasR::normalize()
pipeline <- lasR:::stop_if_chunk_id_below(4) + reader_las() + normalization
lasR::exec(pipeline, on = f, verbose = T)
```

it's the same on my computer. RAM is getting full during `tranform_with()`. It seems as if the RAM remains at a high level, then drops and rises again. In the end, R crashes and the computer hangs.

Jean-Romain, 2024-08-26:
I'm no longer able to reproduce. I completed the MRE on the 9 files with a normal memory usage.

FloFranz, 2024-08-26:
But if you process the file alone you ran out of memory? 

I will try it on another computer tomorrow and see if the same problem occurs there.

Jean-Romain, 2024-08-26:
I eventually found the issue. There a few invalid giant triangles in the triangulation. Why? I don't know.

closed 2024-08-26 by Jean-Romain

Jean-Romain, 2024-08-26:
I’ve fixed the issue, which was caused by a serious memory addressing problem with very large point clouds. This explains why the problem occurred with your largest file. The issue is now resolved, and I greatly appreciate your input—it helped me catch something that my datasets alone didn’t reveal. It took me the entire day to address this, and I’m exhausted.

Now, let’s look at some numbers:

- Each processed chunk of your point cloud dataset, including the buffer, contains about 60 million points per file on average.
- Each point occupies 59 bytes. It is a lot because you have a lot of extrabytes attributes
- The total size of the loaded point cloud is around 3.5 GB per file on average, sometimes reaching up to 5 GB.
- You have millions of ground points per file—easily 2 million per file.
- Triangulating these points requires memory to store the triangulation and adjacency data. This could be estimated, but you’ll need about 1.5 to 2 GB for that alone.

In total, processing a single file requires roughly 5 to 7 GB of RAM depending on the file. If you want to process with 32 cores: 32 * [5-7] GB = 125-225 GB of RAM. Given that you’re on Windows, which already uses a considerable amount of memory just after booting, you’re operating close to your limit. If you’re unlucky and processing several particularly large files simultaneously, you could encounter a memory overflow.

My advice use less core or process smaller chunks. If you chunk into 500x500 meters pieces you could even use 64 cores.

Jean-Romain, 2024-08-26:
By the way I'm recommending you to reprocess your dataset. Your files should be ok but considering the nature of the bug it would be safer to redo the processing. Well actually as long as you did not load more  than 4.3 GB files (LAS) it should be ok but with the loaded buffer you can hardly know how much you loaded. `solling_32_539_5737` was on the limit. The addition of the buffer pushed it beyond 4.3 GB.

FloFranz, 2024-08-27:
Thanks a lot Jean-Romain! I apologize for the time it took to fix the problem, as it is probably not a common crash (only with such large point clouds). Great that `lasR` can now also process such large point clouds. You're right with your words:

> "Rule of thumb: If it crashes, it's NEVER your fault."

:wink:

Thank you also for the hints regarding the processing settings!

Jean-Romain, 2024-08-27:
Please do not apologize. There was a serious bug. You revealed it, you reported it, you provided reproducible data. Rather call it a good contribution :wink:

commit fd9f0b66 (2024-08-26): Fix #91 interger overflow for memory addressing