r-lidar/lasR #94: Search bar on https://r-lidar.github.io/lasR/ does not work.

harryseely, 2024-11-07:
The search bar on the [lasR reference website](https://r-lidar.github.io/lasR/) does not work and returns the error pasted below.

Of course this is not an essential feature, but since many methods are spread across several vignettes, it would be helpful to have a functional search bar.

Thanks!

```404
There isn't a GitHub Pages site here.

If you're trying to publish one, [read the full documentation](https://help.github.com/pages/) to learn how to set up GitHub Pages for your repository, organization, or user account.```

Jean-Romain, 2024-11-07:
The link is not working. You may have been unlucky and found a temporary bug on Github.

closed 2024-11-15 by Jean-Romain

kadyb, 2024-11-17:
@Jean-Romain

https://github.com/r-lidar/lasR/blob/37b07d56462e25ea7b63a802497a16f83aa57776/_pkgdown.yml#L1

Change this line to `url: https://r-lidar.github.io/lasR/` and it should work.

Jean-Romain, 2024-11-17:
Nice. Thank you for the tip