r-lidar/lasR #96: exec pipeline stopped working

summitbri, 2024-11-19:
I've been using lasR for some time now, thank you for the great work! 

Just recently my system simply stops while executing any pipelines.  Not sure if this is related to the 0.12 release or R 4.4.2.

This is a relatively small .las file, one of several I've tried.  The pipeline simply stops after reading the header.  Any suggestions on how to troubleshoot?

> f <- "T:/_TESTPOINTCLOUDS/cloud3.las"
> read <- reader_las()
> pipeline <- read + summarise()
> ans <- exec(pipeline, on = f)
Read files headers: [==========] 100% (1 threads)

summitbri, 2024-11-19:
Quick update, exec pipeline seems to be working fine on Ubuntu, just not either of my 2 Windows 11 machines under both an older and most recent version of R and RStudio.

Ubuntu 24.04.1 LTS: R version 4.3.3 (works)
Windows 11 machine 1: R version 4.4.1, RStudio 2024.04.2 (doesn't work)
Windows 11 machine 2: R version 4.4.2, RStudio 2024.09.1 (doesn't work)

Jean-Romain, 2024-11-19:
Well, is passes all the unit test https://github.com/r-lidar/lasR/actions/runs/11919268809. Please try to reinstall. I made some mess when pushing 0.12.  Finger crossed.

summitbri, 2024-11-19:
Unfortunately, same result.

Jean-Romain, 2024-11-19:
Well that sucks. I will investigate on my virtual machine.

Jean-Romain, 2024-11-20:
Tested on windows with 4.3.3. Everything looks fine. All units tests are passing in a normal time frame.  I suggest you to get back to 10.1 waiting for a fix. Another user reported that it pipeline is taking 4 hours instead of 3 minutes. Weird.

```r
remotes::install_github("r-lidr/lasR", ref = "ab345f0")
```

summitbri, 2024-11-20:
Well I'm glad it wasn't just me missing something obvious.  Working in Ubuntu now so I'll circle back to Windows later.  Thanks!

Jean-Romain, 2024-11-20:
It should be fixed now. Binaries should be available within an hour on r-universe. You can install from GitHub right now.

closed 2024-11-20 by Jean-Romain

summitbri, 2024-11-20:
Thanks for the quick fix!

summitbri, 2024-11-20:
Seems to be working!  Thanks again.

commit 760b3979 (2024-11-20): Fix #96