r-lidar/lasR #104: `write_las()` crashes R when input data has extrabyte attribute with unknown/deprecated data type

georgewoolsey, 2024-12-19:
`write_las()` crashes R (`R Session Aborted`) when the input data has an extrabyte attribute with deprecated data type which `lasR` reads as "Unknown" type.

I could only reproduce with [this data `bad_eb_type.las`](https://colostate-my.sharepoint.com/:u:/g/personal/gwoolsey_colostate_edu/ERlA4j6_1mdBiJWtZV1p7DkBARHnqrA3nSktDhoQw166fQ?e=OTOiEl) which has extrabyte data named "normals" with data type of `22` which `rlas`, `lidR`, and `lasR` all issue the following warning:

`WARNING: data type 22 of attribute 1 ('normals') is deprecated`

Here is the MRE and note that the pipeline only crashes R when the `write_las()` command is added.

I am using `lasR_0.13.0` on `R version 4.3.0` running under Windows 10 x64
```r
f <- file.path(getwd(), "bad_eb_type.las")
## quick look at the header using rlas
h <- rlas::read.lasheader(f)
h[["Variable Length Records"]][["Extra_Bytes"]]
## check what info lasR reads in
p <- lasR::reader() + lasR::info()
lasR::exec(p, on = f) ## notice the "Unknown" data type of the "normals" extrabyte attribute
## lasR pipelines with the data work well until a lasR::write.las() step is added
pipeline_works <- lasR::reader() + lasR::classify_with_csf()
pipeline_fails <- pipeline_works + lasR::write_las()
## check out the working pipeline
ans <- lasR::exec(pipeline_works, on = f)
## check out the failing pipeline
ans <- lasR::exec(pipeline_fails, on = f) # R crashes
```

Jean-Romain, 2024-12-20:
Ok I see. The issue is pretty obvious. The output of `info` shows that `normals` is read as unknown type. This attribute is not actually read and has a size of 0 byte. But the writer does not handle the case.

The solution is to do not read that attribute at all.

```r
Size         : 19.89 MB
Extent       : 608790.00 608816.09 4891969.97 4892010.00 (xmin, xmax, ymin, ymax)
Points       : 685.94 thousands
Coord. ref.  : NAD83 / UTM zone 13N
Schema       :
Name: flags           | uchar  | Desc: Internal 8-bit mask reserved lasR core engine
Name: X               | int    | Desc: X coordinate
Name: Y               | int    | Desc: Y coordinate
Name: Z               | int    | Desc: Z coordinate
Name: Intensity       | ushort | Desc: Pulse return magnitude
Name: ReturnNumber    | uchar  | Desc: Pulse return number for a given output pulse
Name: NumberOfReturns | uchar  | Desc: Total number of returns for a given pulse
Name: Classification  | uchar  | Desc: The 'class' attributes of a point
Name: UserData        | uchar  | Desc: Used at the user’s discretion
Name: PointSourceID   | short  | Desc: Source from which this point originated
Name: ScanAngle       | char   | Desc: Rounded angle at which the laser point was output
Name: R               | ushort | Desc: Red image channel
Name: G               | ushort | Desc: Green image channel
Name: B               | ushort | Desc: Blue image channel
Name: confidence      | uchar  | Desc: confidence values
Name: normals         | Unknown | Desc: unit normal vectors
``

Jean-Romain, 2024-12-20:
Fixed. Data type > 10 (`double`) are no longer read.

closed 2024-12-20 by Jean-Romain

georgewoolsey, 2024-12-20:
Confirmed as fixed in `lasR_0.13.1` update. The extrabyte attribute with data type 22 in the problem data is no longer read.

```
Size         : 19.89 MB
Extent       : 608790.00 608816.09 4891969.97 4892010.00 (xmin, xmax, ymin, ymax)
Points       : 685.94 thousands
Coord. ref.  : NAD83 / UTM zone 13N
Schema       :
Name: flags           | uchar  | Desc: Internal 8-bit mask reserved lasR core engine
Name: X               | int    | Desc: X coordinate
Name: Y               | int    | Desc: Y coordinate
Name: Z               | int    | Desc: Z coordinate
Name: Intensity       | ushort | Desc: Pulse return magnitude
Name: ReturnNumber    | uchar  | Desc: Pulse return number for a given output pulse
Name: NumberOfReturns | uchar  | Desc: Total number of returns for a given pulse
Name: Classification  | uchar  | Desc: The 'class' attributes of a point
Name: UserData        | uchar  | Desc: Used at the user’s discretion
Name: PointSourceID   | short  | Desc: Source from which this point originated
Name: ScanAngle       | char   | Desc: Rounded angle at which the laser point was output
Name: R               | ushort | Desc: Red image channel
Name: G               | ushort | Desc: Green image channel
Name: B               | ushort | Desc: Blue image channel
Name: confidence      | uchar  | Desc: confidence values
```

commit 4e3307c1 (2024-12-20): Fix #104