r-lidar/lasR #105: Invalid read of extrabytes

Jean-Romain, 2024-12-20:
```r
library(lasR)
pipeline = reader() + lasR:::nothing(T)
f = "~/Téléchargements/230526_161628_216_ground/"
exec(pipeline, on = f[1])
```
With [230526_161628_216_ground.zip](https://github.com/user-attachments/files/18203316/230526_161628_216_ground.zip)

```
==============================================================================================================================================================================================================================================
==75958== Invalid read of size 8
==75958==    at 0x4855205: memmove (vg_replace_strmem.c:1410)
==75958==    by 0x6EABC098: LASattributer::init_attributes(unsigned int, LASattribute*) (lasattributer.hpp:389)
==75958==    by 0x6EAD7FBF: LASreaderBuffered::open() (lasreaderbuffered.cpp:191)
==75958==    by 0x6EA8D250: LASreadOpener::open(char const*, bool) (lasreader.cpp:827)
==75958==    by 0x6EA3225B: LASio::open(Chunk const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >) (LASio.cpp:117)
==75958==    by 0x6E9768B1: LASRlasreader::set_chunk(Chunk&) (readlas.cpp:25)
==75958==    by 0x6E903C71: Pipeline::set_chunk(Chunk&) (pipeline.cpp:306)
==75958==    by 0x6EBA8FE5: process(SEXPREC*, SEXPREC*)  (process.cpp:260)
==75958==    by 0x5711A15: GOMP_parallel (in /usr/lib/x86_64-linux-gnu/libgomp.so.1.0.0)
==75958==    by 0x6EBA5039: process(SEXPREC*, SEXPREC*) (process.cpp:213)
==75958==    by 0x6EBA3DB9: C_process (export.cpp:21)
==75958==    by 0x495A2F9: ??? (in /usr/lib/R/lib/libR.so)
==75958==  Address 0x6b557e80 is 0 bytes inside a block of size 384 free'd
==75958==    at 0x484BB0C: free (vg_replace_malloc.c:985)
==75958==    by 0x6EA9DE78: LASattributer::clean_attributes() (lasattributer.hpp:372)
==75958==    by 0x6EABC02A: LASattributer::init_attributes(unsigned int, LASattribute*) (lasattributer.hpp:382)
==75958==    by 0x6EAD7FBF: LASreaderBuffered::open() (lasreaderbuffered.cpp:191)
==75958==    by 0x6EA8D250: LASreadOpener::open(char const*, bool) (lasreader.cpp:827)
==75958==    by 0x6EA3225B: LASio::open(Chunk const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >) (LASio.cpp:117)
==75958==    by 0x6E9768B1: LASRlasreader::set_chunk(Chunk&) (readlas.cpp:25)
==75958==    by 0x6E903C71: Pipeline::set_chunk(Chunk&) (pipeline.cpp:306)
==75958==    by 0x6EBA8FE5: process(SEXPREC*, SEXPREC*)  (process.cpp:260)
==75958==    by 0x5711A15: GOMP_parallel (in /usr/lib/x86_64-linux-gnu/libgomp.so.1.0.0)
==75958==    by 0x6EBA5039: process(SEXPREC*, SEXPREC*) (process.cpp:213)
==75958==    by 0x6EBA3DB9: C_process (export.cpp:21)
==75958==  Block was alloc'd at
==75958==    at 0x48487EF: malloc (vg_replace_malloc.c:442)
==75958==    by 0x6EABC04C: LASattributer::init_attributes(unsigned int, LASattribute*) (lasattributer.hpp:384)
==75958==    by 0x6EAB306A: LASreaderLAS::open(ByteStreamIn*, bool, unsigned int) (lasreader_las.cpp:809)
==75958==    by 0x6EAB10D2: LASreaderLAS::open(char const*, int, bool, unsigned int) (lasreader_las.cpp:95)
==75958==    by 0x6EA8DA9F: LASreadOpener::open(char const*, bool) (lasreader.cpp:910)
==75958==    by 0x6EAD7EF0: LASreaderBuffered::open() (lasreaderbuffered.cpp:174)
==75958==    by 0x6EA8D250: LASreadOpener::open(char const*, bool) (lasreader.cpp:827)
==75958==    by 0x6EA3225B: LASio::open(Chunk const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >) (LASio.cpp:117)
==75958==    by 0x6E9768B1: LASRlasreader::set_chunk(Chunk&) (readlas.cpp:25)
==75958==    by 0x6E903C71: Pipeline::set_chunk(Chunk&) (pipeline.cpp:306)
==75958==    by 0x6EBA8FE5: process(SEXPREC*, SEXPREC*)  (process.cpp:260)
==75958==    by 0x5711A15: GOMP_parallel (in /usr/lib/x86_64-linux-gnu/libgomp.so.1.0.0)
```

samuelgrubinger, 2024-12-23:
I ran the following lidR code to produce clean input files without extra bytes: 

```future::plan(multicore, workers = 24)
lidR.progress = FALSE

path_als = paste0(root_dir, "/input")

# path to the output of this stage
data_path = paste0(root_dir, "/input_clean")

# define a function to drop extra bytes
drop_extra_bytes <- function(chunk) # user defined function
{
  las <- readLAS(chunk, select = "xyzarnw")  
  if (is.empty(las)) return(NULL)  
  return(las)
}

if (file.exists(data_path)) {

    CTG_clean = readLAScatalog(data_path)
    
} else {

    # create the directory if it doesn't exist
    dir.create(data_path, recursive = TRUE)

CTG = readLAScatalog(path_als)
#opt_select(CTG) = "xyzarnw"
opt_chunk_size(CTG) = 600
opt_output_files(CTG) = paste0(root_dir, "/input_clean/{XCENTER}_{YCENTER}")

CTG_clean = suppressWarnings(catalog_apply(CTG, drop_extra_bytes))
    }
```

It still crashes instantaneously when I try to run anything in lasR on the clean inputs.

Jean-Romain, 2024-12-23:
If you provide an example, please provide a minimal reproducible example. Thank you.

samuelgrubinger, 2024-12-23:
I made a silly mistake. After dropping the extra bytes, the code does run. 

I first run this, in lidR, to drop the extra bytes:
```# _1_classified_points

path_als = paste0(root_dir, "/input_clean")

if (length(list.files(dir_list[1])) == 0) {
    
    # only read one point per 2x2 cm voxel,
    # drop points below the lowest elevation expected at the site
    # only read points with scan angles +/- 15 degrees from nadir,
    # drop likely noise points with very high numbers of returns or return numbers
    pipeline = reader(filter = "-thin_with_voxel .02
                                -keep_z_above 20
                                -drop_abs_scan_angle_above 15
                                -keep_return 1 2 3 4 5 6 7
                                -drop_number_of_returns 15
                                -drop_number_of_returns 14
                                -drop_number_of_returns 13
                                -drop_number_of_returns 12
                                -drop_number_of_returns 11
                                -drop_number_of_returns 10
                                -drop_number_of_returns 9
                                -drop_number_of_returns 8
                                -drop_withheld"
                                             ) +
 

            # classify points as noise if they have fewer than 200 points in their surrounding 10x10 m voxels,
            # for big clouds of above- or belowground noise
            classify_with_ivf(res = 10, n = 200L, class = 18L) +

            # # classify points as noise if they have fewer than 30 points in their surrounding 5x5 m voxels
            classify_with_ivf(res = 5, n = 30L, class = 18L) +
            
            # classify ground points with a cloth simulation filter algorithm
            classify_with_csf(slope_smooth = FALSE,
                                class_threshold = 0.5,
                                cloth_resolution = 0.5,
                                rigidness = 3L,
                                iterations = 500L,
                                time_step = 0.6,
                                class = 2L,
                                # only let the algorithm use last returns that aren't noise
                                filter = "Classification != 18") +
            
            # write the point clouds with ground and noise points classified
            write_las(paste0(root_dir, "/output/_1_classified_points/*_ground.las"))
            
           
    exec(pipeline, 
         #on = CTG,
         on = path_als,
         ncores = 16, 
         with = list(chunk = 600),
         progress = FALSE)

    }
```

Jean-Romain, 2024-12-24:
I pushed a patch in the `devel` branch. It fixes your issue but I want to test other things before to push it in production.

samuelgrubinger, 2024-12-24:
After installing the latest update with `install.packages('lasR', repos = c('https://r-lidar.r-universe.dev',   'https://cloud.r-project.org'), dependencies = TRUE)` it still crashes instantly on read, if extra bytes are not removed.

Jean-Romain, 2024-12-24:
You installed from the production branch.

```r
remotes::install_github("r-lidar/lasR", ref = "devel")
```

samuelgrubinger, 2024-12-25:
My mistake. Thank you, it runs now.

commit 7b5f807b (2024-12-24): Fix #105
closed 2024-12-29 by Jean-Romain