r-lidar/lasR #106: Possible merge pipeline issue with duplicate points.

summitbri, 2024-12-20:
I'm merging 6 small LAS files.  3 areas that were previously split into ground and off-ground points, now merging them together. 

When I merge in cloudcompare, there are about 10M total points.  When I merge the files with the lasr merge pipeline, I get about 24M points - much larger than the sum of the individual files.  

One theory is there may be pulses that are duplicated that have resulted in both ground and off ground points that are being replicated?  Still doesn't explain more than double...

I've reproduced the problem in both 12.1 and 13.1.

This is basically my merge pipeline:

```
		files_to_copy <- list.files(ground_las_dir, pattern = "\\.(las|laz)$", ignore.case = TRUE, full.names = TRUE)
		file.copy(files_to_copy, file.path(merged_dir, basename(files_to_copy)), overwrite = TRUE)
		files_to_copy <- list.files(offground_las_dir, pattern = "\\.(las|laz)$", ignore.case = TRUE, full.names = TRUE)
		file.copy(files_to_copy, file.path(merged_dir, basename(files_to_copy)), overwrite = TRUE)
		ofile = paste0(final_dir, "/Merged_Point_Cloud.las")
		merge = reader_las() + write_las(ofile)
		ans = exec(merge, on = merged_dir)
		ans

```
lasR merge:
[image]

Cloudcompare merge:
[image]

Jean-Romain, 2024-12-21:
I need a reproducible example. I tried this succesfully

```r
library(lasR)
f <- system.file("extdata", "bcts/", package = "lasR")
merge = reader_las() + write_las(templas())
ans = exec(merge, on = f, verbose = TRUE)

library(lidR)
las = readLAS(ans)
ctg = readLAScatalog(f)
las
ctg
``

summitbri, 2024-12-21:
Ok, I will try to follow your explicit example.  Otherwise I can post a link to the files.

summitbri, 2024-12-21:
Similar results below.  What's a good way for me to send you the files?

class        : LAS (v1.2 format 3)
memory       : 1.6 Gb 
extent       : 2821849, 2823072, 1369726, 1370810 (xmin, xmax, ymin, ymax)
coord. ref.  : NA 
area         : 497872 units²
points       : 25.09 million points
density      : 50.39 points/units²
density      : 46.03 pulses/units²
> ctg
class       : LAScatalog (v1.2 format 3)
extent      : 2821849, 2823072, 1369726, 1370810 (xmin, xmax, ymin, ymax)
coord. ref. : NA 
area        : 1.87 kunits²
points      : 10.47 million points
density     : 5.6 points/units²
density     : 5.1 pulses/units²
num. files  : 6

Jean-Romain, 2024-12-21:
Try to reproduce with a smaller dataset if possible. Then send file through any file transfer tool (google drive, wetransfert, microsoft cloud,...). You necessarily have access to one.

summitbri, 2024-12-21:
Was able to reproduce when merging two files, one with ground points, and another with off-ground points, for the same extents (ie overlapping tiles).

Test files:

https://drive.google.com/drive/folders/1iuvfaHmLW5JqU5cf8awdMWBbHQ3D6KjX?usp=sharing

Results:

class        : LAS (v1.2 format 3)
memory       : 496 Mb 
extent       : 2821849, 2822485, 1370240, 1370810 (xmin, xmax, ymin, ymax)
coord. ref.  : NA 
area         : 188432 units²
points       : 7.65 million points
density      : 40.59 points/units²
density      : 36.81 pulses/units²
> ctg
class       : LAScatalog (v1.2 format 3)
extent      : 2821849, 2822485, 1370240, 1370810 (xmin, xmax, ymin, ymax)
coord. ref. : NA 
area        : 0.71 kunits²
points      : 3.83 million points
density     : 5.4 points/units²
density     : 4.9 pulses/units²
num. files  : 2 

When merging files that are different XY tiles, there's no problem.

Jean-Romain, 2024-12-21:
> When merging files that are different XY tiles, there's no problem.

Ok I understand the issue. This is something people have difficulty to understand. Neither `lidR` not `lasR` are working "by file". They are working by arbitrary sized spatial chunk in order to cover a large territory.  `lidR` and `lasR` are processing arbitrary sized square chunks of the territory. Usually a chunk is a file because the dataset is tiled. Here your dataset is not tiled properly. `lidR` would have thrown a warning but `lasR` does not check for tiling pattern quality yet.

Thus what happened? `lasR` takes the first chunk (i.e. the first file here) and processes this regions of interest. But this region is covered by two files so it reads and load the two files as one point cloud for this region. Then it processes the second chunk i.e. the second file. Same problem, it is the same region processed a second time. So you end up with roughly twice the points.

When processing a raster this is not visible. You are processing twice the same region but it overwrites the previous computation. For writing files it has sides effects.

`lasR` is not a tools to manipulate and merge files. For that you have `lastools`, `cloudcompare` and other software. Yet I understand that your use case is legit so I will add an option to force a "by file" processing.

summitbri, 2024-12-21:
Got it!  Makes complete sense.  I can give the large chunk size a go, and if that doesn't work I can use lasmerge for now as I already am using a couple of other lastools binaries in my pipeline.

Thanks for taking a look and no rush on the 'by file' processing.  Happy holidays!

summitbri, 2024-12-21:
PS adding the chunk size seems to do the trick!

closed 2024-12-23 by Jean-Romain