r-lidar/lasR #108: quick test of upcoming 3D viewer

wiesehahn, 2025-01-02:
I had a quick glimpse at the 3D viewer in v 0.14 (devel) under windows 11. 

some observations:

- progress bar for LoD generation ends at 99% while viewer opens in the background. Possibly better to print a message that viewer is ready, or always bring window to front when ready.

- colors are shown in discrete steps for small differences (e.g. flat terrain), possibly color palette could be extended with more colors, or min and max could be calculated based on points in viewport instead of full pointcloud.

<details>

[image]

</details>

- when zooming in, one mouse wheel rotation has less and less effect, while this might make sense to a certain degree the effect is too heavy in my opinion. In the current state it becomes unusable at a certain point at which it is unpossible to zoom out again and also to pan around the map.
- the scene in the 3D viewer is heavily flickering most of the time
<details>

https://github.com/user-attachments/assets/23f2a57f-a44f-4bff-8146-2d3554474649

https://github.com/user-attachments/assets/ea9fd0c8-0284-413a-b9bd-202731cf7e55

</details>

Jean-Romain, 2025-01-02:
Thank you for your feedback.

> Progress bar for LoD generation ends at 99%.

Ok, easy to fix.

> Possibly better to print a message that the viewer is ready, or always bring the window to the front when ready.

We have the same problem with the lidR rgl viewer on Windows. It opens in the background, which is not very convenient. I do not have this issue on Linux; the window is always opened in front both in `lidR` and `lasR`.

> Colors are shown in discrete steps for small differences (e.g., flat terrain). Possibly the color palette could be extended with more colors, or min and max could be calculated based on points in the viewport instead of the full point cloud.

Certainly. Easy to fix. More generally, I still have some work to do on colors.

> When zooming in, one mouse wheel rotation has less and less effect.

I know this one. I will fix it.

> The scene in the 3D viewer is heavily flickering most of the time.

This is not normal. It is not flickering on Linux. Surprisingly, you did not report this issue for `lidRviewer`, which uses pretty much the same code. This is (1) Windows-specific and (2) due to something different than in `lidRviewer`. Well, that one won't be easy to fix.

wiesehahn, 2025-01-02:
> We have the same problem with the lidR rgl viewer on Windows. It opens in the background, which is not very convenient. I do not have this issue on Linux; the window is always opened in front both in `lidR` and `lasR`.

in lidR it might be possible to fix this with `rgl.bringtotop()` 

> This is not normal. It is not flickering on Linux. Surprisingly, you did not report this issue for `lidRviewer`, which uses pretty much the same code. This is (1) Windows-specific and (2) due to something different than in `lidRviewer`. Well, that one won't be easy to fix.

exactly, lidRviewer does not have the problem

Jean-Romain, 2025-01-02:
- [ ] Improve color palettes
- [ ] Fix progress bar to 100%
- [ ] Try to put the windows in front on Windows
- [ ] Fix zoom
- [ ] Fix flickering on Windows

Jean-Romain, 2025-01-03:
I fixed 3 points. The zoom is no longer infinite and stops at a certain level. Zoom sensitivity is no longer linear. Pan sensitivity has been updated to be more comfortable. The windows should open in foreground on windows. However I did not fix the flickering yet. On my virtual box the viewer is black and display nothing :disappointed:

wiesehahn, 2025-01-06:
Zoom is better, but I think it could be stronger/faster/more responsive (more zoom per mouse wheel rotation). Currently we have to keep scrolling quite long to get fully in. 

Panning around seems to work ok, but it feels a little laggy. 

The first time we plot a pointcloud in a R session it opens in the background, the following times in the foreground.

Progress is up to 100% but the following message "NULL no progress" might be irritating.

Regarding the flickering, as long as we keep constantly panning or rotating around the scene there is no flickering, but when we zoom or do nothing there is heavy flickering.