r-lidar/lasR #109: Skewness and kurtosis

danielrode, 2025-01-06:


Jean-Romain, 2025-01-06:
`stdmetrics` returns `zskew` and `zkurt` and `iskew` and `ikurt`

danielrode, 2025-01-07:
But `stdmetrics` is from the `lidR` package, right? Which uses the slower R code (and not C++), or does `lasR` also have a `stdmetrics` function? I didn't see one mentioned in the docs or in the `lasR` namespace.

Jean-Romain, 2025-01-07:
Ho yes sorry. `lidR`/`lasR`, pretty much the same name actually. I though it was a report in `lidR`. My bad.

I guess I can do it yes.

Jean-Romain, 2025-01-07:
`skew` and `kurt` metrics added. Be careful I did not check anything yet. You can now ask for metrics like `z_skew`, `i_kurt` and so on.

danielrode, 2025-01-07:
I just tried your new code on a test tile and it works great! Gives the same result as a custom function I wrote for lidR. Thanks!

closed 2025-01-07 by danielrode

Jean-Romain, 2025-01-07:
Ok thank you for testing it.