r-lidar/lasR #110: Error: Invalid data: Overlap_flag is not logical

wiesehahn, 2025-01-08:
I came across this error when trying to read a las dataset with lidR which was written with lasR. I am not sure if there is a problem with the reader in lidR or the writer of lasR or neither. 
We can read the file with lasR as seen below but not with lidR.

```r
> summary(las@data$Overlap_flag)
   Mode   FALSE 
logical    2672 
> ans <- exec(write_las(), on = las)
> las1 <- lidR::readLAS(ans)
Error: Invalid data: Overlap_flag is not logical
> exec(info(), on = ans)
Size         : 162.99 kB
Extent       : 549074.02 549093.93 5724742.14 5724761.66 (xmin, xmax, ymin, ymax)
Points       : 2.67 thousands
Coord. ref.  : ETRS89 / UTM zone 32N
Schema       :
Name: flags           | uchar  | Desc: Internal 8-bit mask reserved lasR core engine
Name: X               | int    | Desc: X coordinate
Name: Y               | int    | Desc: Y coordinate
Name: Z               | int    | Desc: Z coordinate
Name: Intensity       | ushort | Desc: Pulse return magnitude
Name: ReturnNumber    | uchar  | Desc: Pulse return number for a given output pulse
Name: NumberOfReturns | uchar  | Desc: Total number of returns for a given pulse
Name: Classification  | uchar  | Desc: The 'class' attributes of a point
Name: UserData        | uchar  | Desc: Used at the userâ€™s discretion
Name: PointSourceID   | short  | Desc: Source from which this point originated
Name: ScanAngle       | char   | Desc: Rounded angle at which the laser point was output
Name: gpstime         | double | Desc: Time tag value at which the point was observed
Name: R               | ushort | Desc: Red image channel
Name: G               | ushort | Desc: Green image channel
Name: B               | ushort | Desc: Blue image channel
Name: EdgeOfFlightline | uchar  | Desc: 
Name: Synthetic_flag  | uchar  | Desc: 
Name: Keypoint_flag   | uchar  | Desc: 
Name: Withheld_flag   | uchar  | Desc: 
Name: Overlap_flag    | uchar  | Desc: 
Name: Amplitude       | double | Desc: 
Name: Reflectance     | double | Desc: 
Name: Deviation       | int    | Desc: 
NULL
```

Jean-Romain, 2025-01-08:
Ho I see, you used a lidR LAS object into a lasR pipeline and bit flags were not handled properly because lasR no longer handles bit flags. They were rather stored as extrabyte but then lidR can't read it back because a name clash

wiesehahn, 2025-01-08:
do you still nee a file to reproduce?

Jean-Romain, 2025-01-08:
```r
library(lidR)
library(lasR)
f <- system.file("extdata", "las14_prf6.laz", package="rlas")
las = readLAS(f)
summary(las@data$Overlap_flag)
ans <- exec(write_las(), on = las)
las1 <- lidR::readLAS(ans)
exec(info(), on = ans)
``

Jean-Romain, 2025-01-09:
Thank you for reporting. This forced me to work on bit attributes and LAS1.4.

- reader now reads bit attributes from LAS files in single bit attributes
  ```
  Size         : 4.72 kB
  Extent       : 487805.98 487842.96 5313781.18 5313818.66 (xmin, xmax, ymin, ymax)
  Points       : 135.00 
  Coord. ref.  : (null)
  Schema       :
  19 attributes | 35 bytes per points
   Name: flags             | uchar  | Desc: Internal 8-bit mask reserved for lasR core engine
   Name: X                 | int    | Desc: X coordinate
   Name: Y                 | int    | Desc: Y coordinate
   Name: Z                 | int    | Desc: Z coordinate
   Name: Intensity         | ushort | Desc: Pulse return magnitude
   Name: ReturnNumber      | uchar  | Desc: Pulse return number for a given output pulse
   Name: NumberOfReturns   | uchar  | Desc: Total number of returns for a given pulse
   Name: Classification    | uchar  | Desc: The 'class' attributes of a point
   Name: UserData          | uchar  | Desc: Used at the user’s discretion
   Name: PointSourceID     | short  | Desc: Source from which this point originated
   Name: ScanAngle         | float  | Desc: Angle at which the laser point was output
   Name: ScannerChannel    | uchar  | Desc: Channel (scanner head) of a multi-channel system
   Name: gpstime           | double | Desc: Time tag value at which the point was observed
   Name: EdgeOfFlightline  | bit    | Desc: Set when the point is at the end of a scan
   Name: ScanDirectionFlag | bit    | Desc: Direction in which the scanner mirror was traveling 
   Name: Synthetic         | bit    | Desc: Point created by a technique other than direct observation
   Name: Keypoint          | bit    | Desc: Point is considered to be a model key-point
   Name: Withheld          | bit    | Desc: Point is supposed to be deleted)
   Name: Overlap           | bit    | Desc: If set, point is within an overlap region of 2+ swaths
  ```
 - writer now automatically writes LAS 1.4 if needed.
 - reader with LAS object from `lidR` now handles and recognizes attributes `Withheld_flag`, `Keypoint_flag`, `Synthetic_flag`, `Overlap_flag`, ` EdgeOfFlightline`, `ScanDirectionFlag`. They are respectively named `Withheld`, `Keypoint`, `Synthetic`, `Overlap`, ` EdgeOfFlightline`, `ScanDirectionFlag` in `lasR`

wiesehahn, 2025-01-10:
it seems to work. Thanks!

closed 2025-01-26 by Jean-Romain