r-lidar/lasR #118: DTM output of lasR is different from lidR

jmmonnet, 2025-02-04:
Hi, 
I decided to move from lidR to lasR for a more efficient production of DTMs. When comparing outputs I noticed that the maps are similar but slightly shifted. I would expect outputs to be equal as a TIN is used in both cases. Is there a parameter that I did not set correctly for the comparison ?
Thank you !

Here is an example obtained with lasR 0.13.4 and lidR 4.1.2:
```
f <- system.file("extdata", "Topography.laz", package="lidR")
# dtm resolution
res <- 2
# lasR pipeline
library(lasR)
tri = triangulate(filter = keep_class(c(2, 9)))
dtm = rasterize(res, tri)
dtm_lasR = exec(reader() + tri + dtm , on = f)
#
# lidR processing with catalog
library(lidR)
cata <- lidR::catalog(f)
opt_filter(cata) <- "-keep_class 2 9"
dtm_lidR <- lidR::rasterize_terrain(cata, res, lidR::tin())
#
# compare extent
terra::ext(dtm_lidR) == terra::ext(dtm_lasR)
# histogram of difference
terra::hist(dtm_lidR - dtm_lasR)
#
# # lasR dtm has NA cells along left and upper borders
# there seems to be a shift in the dtms
# along NW - SE direction
terra::plot(dtm_lidR)
terra::plot(dtm_lasR)
# map of difference looks like a hillshade
terra::plot(dtm_lidR - dtm_lasR, range = c(-1, 1))
```

Jean-Romain, 2025-02-04:
It is not an issue with the raster engine. CHMs are aligned. So it is specific to the triangulation.

```r
f <- system.file("extdata", "Topography.laz", package="lidR")
res <- 8

library(lasR)
chm = rasterize(res, "max")
chm_lasR = exec(chm , on = f)
#
library(lidR)
cata <- lidR::catalog(f)
chm_lidR <- lidR::rasterize_canopy(cata, res)

terra::plot(chm_lidR)
terra::plot(chm_lasR)

terra::plot(chm_lidR - chm_lasR, range = c(-1, 1))
```

jmmonnet, 2025-02-04:
If the "shift" in the dtm is greater when resolution increases, would it be possible that altitudes are interpolated from the TIN on a point grid that is shifted from the raster cells (e.g. altitudes are interpolated at the upper left corner instead of the center of cells) ?

Jean-Romain, 2025-02-04:
You are absolutely right. It was interpolated at the corner of the cell instead of the center. This is fixed now.

commit 4f3400b6 (2025-02-04): Fix #118
closed 2025-02-04 by Jean-Romain