r-lidar/lasR #119: merge operation continues before write_las is complete

summitbri, 2025-02-05:
ofile = paste0(final_dir, "/LiDAR_Point_Cloud.las")
merge = reader_las() + write_las(ofile)
ans = exec(merge, on = merged_dir, chunk_size = 10000)
ans
		
I've discovered that on some larger projects, the code after 'ans' commences before the write_las task is complete.
My temporary fix adds an arbitrary 30 second pause (5 seconds wasn't enough in some cases), but this is probably longer or shorter than needed depending on the case: Sys.sleep(30)

summitbri, 2025-02-05:
Correct.  The merged LAS also results in missing data as if the write operation hasn't completed.  I'm not sure how to send you a reproducible example without sending a very large dataset.  My pause action is working for now, but perhaps keep an eye on this.

Jean-Romain, 2025-02-05:
Ok so rather `write_las` did not complete? But it you add a delay it completes? Is is correct. This is absolutely weird.

Btw you do not need , `chunk_size = 10000` here. But this is not related.

summitbri, 2025-02-05:
Yes, that's correct.  Weird, I know.  Took me awhile to figure it out.

I think the chunk_size was lingering from another merge related item that you since fixed.

Jean-Romain, 2025-02-05:
How many data do we need to reproduce? The merge pipeline is tested internally and works.

summitbri, 2025-02-06:
Will get back to you soon with the details.

summitbri, 2025-02-12:
Apologies for the delay, I was at Geo Week this week.  Here's a screen shot of the folder of files (on SSD) where I first noticed the problem.

[image]

Also, while I don't *think* it is related, I am using a parallel strategy, however I see a note that parallel is not used with merge.  

set_parallel_strategy(nested(ncores = ncores()/4L, ncores2 = 4L))

This is the only messages I see when executing the merge pipeline:

8 files do not have a spatial index. Spatial indexing speeds up tile buffering and spatial queries drastically.
Files will be indexed on-the-fly. This will take some extra time now but will speed up everything later.
WARNING: This pipeline is not parallizable using 'concurrent-files' strategy.

60 second pause to complete write_las task. #my fix

closed 2025-06-30 by Jean-Romain