r-lidar/lasR #120: callback overwrites existing attributes

samuelgrubinger, 2025-02-09:
When an attribute is exposed to a callback function, that attribute is overwritten with 1s. In this case, NumberOfReturns and ReturnNumber are overwritten. 

```dir_input = paste0(dir_root, "/SantaRita_test/input")

pipeline = reader(filter = "-drop_withheld" ) +

            # function to classify ground points that are not last returns as unclassified
            # equivalent to -keep_last
            add_extrabytes(data_type = "float", 
                           name = "LastFlag", 
                           description = "last returns") +
    
            callback(fun = function(data) {
                data |>
                    dplyr::mutate(ReturnNumber = ReturnNumber,
                                 NumberOfReturns = NumberOfReturns,
                                  LastFlag = if_else(ReturnNumber == NumberOfReturns & Classification != 18,
                                                 1, 0))
            }, 
                     expose = "cnr") + 
   
            # write the point clouds with ground and noise points classified
            write_las(paste0(dir_root, "/SantaRita_test/output/_1_classified_points/*_ground.las"))
            
           
    exec(pipeline, 
         on = dir_input,
         progress = FALSE)```

[SantaRita_test_folder.zip](https://github.com/user-attachments/files/18724853/SantaRita_test_folder.zip)

Jean-Romain, 2025-02-09:
Cannot reproduce, everything looks good. What makes you think there is an issue? Can you produce a minimal reproducible example (without `dplyr`, with example data)

``` r
library(lasR)
library(rlas)

f <- system.file("extdata", "Topography.las", package = "lasR")

eby = add_extrabytes(data_type = "float", name = "LastFlag", description = "last returns")
cbk =  callback(fun = function(data) {
    data |>
      dplyr::mutate(ReturnNumber = ReturnNumber,
                    NumberOfReturns = NumberOfReturns,
                    LastFlag = dplyr::if_else(ReturnNumber == NumberOfReturns & Classification != 18, 1, 0))
  }, 
  expose = "cnr")
wrt = write_las(templas())

ans = exec(eby + cbk + wrt, on = f, progress = FALSE)

rlas::read.las(f, select = "rn0")
#>               X       Y        Z ReturnNumber NumberOfReturns
#>           <num>   <num>    <num>        <int>           <int>
#>     1: 273357.1 5274360 806.5340            1               1
#>     2: 273357.2 5274359 806.5635            2               2
#>     3: 273357.2 5274358 806.0248            2               2
#>     4: 273357.2 5274510 809.6303            2               2
#>     5: 273357.2 5274509 809.3880            1               1
#>    ---                                                       
#> 73399: 273642.8 5274579 806.2622            1               2
#> 73400: 273642.8 5274578 806.9623            1               1
#> 73401: 273642.8 5274577 807.2855            1               1
#> 73402: 273642.8 5274577 807.1540            1               3
#> 73403: 273642.8 5274576 806.0433            1               3
rlas::read.las(ans, select = "rn0")
#>               X       Y        Z ReturnNumber NumberOfReturns LastFlag
#>           <num>   <num>    <num>        <int>           <int>    <num>
#>     1: 273357.1 5274360 806.5340            1               1        1
#>     2: 273357.2 5274359 806.5635            2               2        1
#>     3: 273357.2 5274358 806.0248            2               2        1
#>     4: 273357.2 5274510 809.6303            2               2        1
#>     5: 273357.2 5274509 809.3880            1               1        1
#>    ---                                                                
#> 73399: 273642.8 5274579 806.2622            1               2        0
#> 73400: 273642.8 5274578 806.9623            1               1        1
#> 73401: 273642.8 5274577 807.2855            1               1        1
#> 73402: 273642.8 5274577 807.1540            1               3        0
#> 73403: 273642.8 5274576 806.0433            1               3        0
```

<sup>Created on 2025-02-09 with [reprex v2.1.1](https://reprex.tidyverse.org)</sup>

samuelgrubinger, 2025-02-09:
I get 1 when I run:
```
test = rlas::read.las(paste0(dir_root, "/output/_1_classified_points/SantaRita_634300_1032100_ground.las"))
max(test$NumberOfReturns)
```
It runs when you try it on the attached Santa Rita data?

Jean-Romain, 2025-02-09:
I see. Why with your dataset and not mine?!? I'm investigating.

``` r
library(lasR)
library(rlas)

  f = "/home/jr/Téléchargements/SantaRita_test/input/SantaRita_634300_1032100.las"
  
  pipeline = add_extrabytes(data_type = "float", name = "LastFlag", description = "last returns") +
    callback(fun = function(data) { dplyr::mutate(data, LastFlag = dplyr::if_else(ReturnNumber == NumberOfReturns & Classification != 18, 1, 0)) }, expose = "cnr") + 
    write_las(templas())
  
  
  ans = exec(pipeline, on = f, progress = FALSE)

  rlas::read.las(f, select = "rn3")
#> Warning: There are 2353 points flagged 'withheld'.
#>                X       Y      Z ReturnNumber NumberOfReturns
#>            <num>   <num>  <num>        <int>           <int>
#>      1: 634400.0 1032001 225.10            1               1
#>      2: 634399.9 1032000 224.97            1               3
#>      3: 634400.0 1032000 224.31            2               3
#>      4: 634400.0 1032000 223.38            3               3
#>      5: 634399.9 1032000 225.41            1               3
#>     ---                                                     
#> 861644: 634313.2 1032000 178.19            2               4
#> 861645: 634313.7 1032000 177.34            3               4
#> 861646: 634314.6 1032000 175.75            4               4
#> 861647: 634315.1 1032000 176.72            1               1
#> 861648: 634317.4 1032000 178.01            1               1
  rlas::read.las(ans, select = "rn3")
#> Warning: There are 2353 points flagged 'withheld'.
#>                X       Y      Z ReturnNumber NumberOfReturns LastFlag
#>            <num>   <num>  <num>        <int>           <int>    <num>
#>      1: 634400.0 1032001 225.10            1               1        1
#>      2: 634399.9 1032000 224.97            1               1        0
#>      3: 634400.0 1032000 224.31            1               1        0
#>      4: 634400.0 1032000 223.38            1               1        1
#>      5: 634399.9 1032000 225.41            1               1        0
#>     ---                                                              
#> 861644: 634313.2 1032000 178.19            1               1        0
#> 861645: 634313.7 1032000 177.34            1               1        0
#> 861646: 634314.6 1032000 175.75            1               1        1
#> 861647: 634315.1 1032000 176.72            1               1        1
#> 861648: 634317.4 1032000 178.01            1               1        1
```

<sup>Created on 2025-02-09 with [reprex v2.1.1](https://reprex.tidyverse.org)</sup>

Jean-Romain, 2025-02-09:
95% sure it is an bug with LAS 1.4 format 6

Jean-Romain, 2025-02-09:
Fixed

closed 2025-02-09 by Jean-Romain

samuelgrubinger, 2025-02-10:
Thanks JR!

commit ca4aa71e (2025-02-09): Write LAS 1.4 fix #120