r-lidar/lasR #126: r-universe mislabeled as 0.13.6

wildintellect, 2025-03-19:
The current release on r-universe seems be labeled 0.13.6 but actually contains 0.13.7 code. I noticed this when the hash changed recently but the version on the website did not increment.

[image]

Jean-Romain, 2025-03-19:
Absolutely. I choose a discutable strategy. Because the package is not on CRAN  and won't be on CRAN the package I choose to check on github what is the latest version every time `library(lasR)` is called. This way I can print an update message.

However I pushed fixes in the `master` branch without increasing the version in the `DESCRIPTION` file in order to do not trigger a message and a new version every time a small bug is fixed. This way people who installed recently get the very last version with minor bug fixed and those who installed earlier will have a notification when I will actually trigger a new version. I'm triggering a new version when I have a significant fix or several minor fixes.

wildintellect, 2025-03-19:
Ah, the challenge for me packaging for conda #121 is that the hash is checked, and needs to be updated in match with version number that's in the file name of the tar.gz

I'll need to look into writing a github workflow to take care of updating the conda feedstock yaml to do this automatically.

closed 2025-03-30 by Jean-Romain

wildintellect, 2025-04-24:
I'd like to suggest that perhaps you could change your version check to only look at `major.minor` for your messaging, so that you can still use `patch` (aka bugfix) in `major.minor.patch` every time you push small fixes. It's very confusing and actually a little bit worrying on the security side to have the hash change but the version stay the same, and requires reading the git changelog every time to understand which fixes are actually in the version someone has installed. 

https://semver.org/