r-lidar/lasR #142: lasR::chm has smaller extent when calculated on a lidR catalog in combination with lasR::reader_circles

eikeschott, 2025-06-03:
Sorry to bother you again but I noticed some more irregularities when working with lidR catalogs.

For me the second `terra::plot(small_file)` also outputs a smaller looking chm compared to the first plot which is super weird and might not have anything todo with lasR but I wanted to mention it. Even after clearing the plot window between the first and second plots.

```R
file <- c(system.file("extdata", "Megaplot.las", package="lasR"))
ctg <- lidR::readLAScatalog(file)

read_small <- lasR::reader_circles(xc = 684876.6, yc = 5017902, r = 80)
read_large <- lasR::reader_circles(xc = 684876.6, yc = 5017902, r = 8000)

chm <- lasR::chm(res = 1) # The same is true for lasR::dtm

small_file <- lasR::exec(read_small + chm, on = file) 
terra::plot(small_file) # As expected

small_ctg <- lasR::exec(read_small + chm, on = ctg)
terra::plot(small_ctg) # Smaller than expected
terra::plot(small_file) # Also smaller than expected which is super weird

large_file <- lasR::exec(read_large + chm, on = file)
terra::plot(large_file) # As expected (Megaplot.las is only 227m wide)

large_ctg <- lasR::exec(read_large + chm, on = ctg)
terra::plot(large_ctg) # Way to small
terra::plot(large_file) # Smaller than expected
```

Jean-Romain, 2025-06-03:
Like in the previous bug report it is all related to buffer management.

``` r
file <- c(system.file("extdata", "Megaplot.las", package="lasR"))
ctg <- lidR::readLAScatalog(file)

read_small <- lasR::reader_circles(xc = 684876.6, yc = 5017902, r = 10)

chm <- lasR::chm(res = 1) # The same is true for lasR::dtm

res1 <- lasR::exec(read_small + chm, on = file) 
terra::plot(res1) # As expected
```

[image]

``` r

res2 <- lasR::exec(read_small + chm, on = file, buffer = 30)
terra::plot(res2)
```

[image]

``` r

read_small <- lasR::reader_circles(xc = 684876.6, yc = 5017902, r = 80)

chm <- lasR::chm(res = 1) # The same is true for lasR::dtm

res1 <- lasR::exec(read_small + chm, on = file) 
terra::plot(res1) # As expected
```

[image]

``` r

res2 <- lasR::exec(read_small + chm, on = file, buffer = 30)
terra::plot(res2)
```

[image]

<sup>Created on 2025-06-03 with [reprex v2.1.1](https://reprex.tidyverse.org)</sup>

eikeschott, 2025-06-04:
Great explanation, it makes sense. Thanks for the fix!