r-lidar/lasR #159: Adding extrabytes named "angle" does not seem to work properly

eikeschott, 2025-06-18:
A slightly modified example copied from ?lasR::add_extrabytes:

I don't know what it is but there seems to be an issue with adding an extrabyte named "angle". 

This also applies to filters in potential stages further down the line. e.g. `filter = "angle > 40"` will not work but `filter = "RANDOM > 40"` will.

``` r
f <- system.file("extdata", "Topography.las", package = "lasR")

# This works
fun_RANDOM <- function(data) { data$RANDOM <- runif(nrow(data), 0, 100); return(data) }
pipeline <- lasR::reader() +
  lasR::add_extrabytes("float", "RANDOM", "Random numbers") +
  lasR::callback(fun_RANDOM, expose = "xyz") +
  lasR::summarise(metrics = c("RANDOM_min", "RANDOM_max", "RANDOM_mean", "count"))
lasR::exec(pipeline, on = f)$metrics
#>   RANDOM_max RANDOM_mean   RANDOM_min count
#> 1   99.99761    49.85262 0.0005222391 73403

# This works
fun_an <- function(data) { data$an <- runif(nrow(data), 0, 100); return(data) }
pipeline <- lasR::reader() +
  lasR::add_extrabytes("float", "an", "Random numbers") +
  lasR::callback(fun_an, expose = "xyz") +
  lasR::summarise(metrics = c("an_min", "an_max", "an_mean", "count"))
lasR::exec(pipeline, on = f)$metrics
#>     an_max  an_mean       an_min count
#> 1 99.99313 50.11934 0.0003611436 73403

# This doesn't
fun_angle <- function(data) { data$angle <- runif(nrow(data), 0, 100); return(data) }
pipeline <- lasR::reader() +
  lasR::add_extrabytes("float", "angle", "Random numbers") +
  lasR::callback(fun_angle, expose = "xyz") +
  lasR::summarise(metrics = c("angle_min", "angle_max", "angle_mean", "count"))
lasR::exec(pipeline, on = f)$metrics
#>   angle_max angle_mean angle_min count
#> 1         1  -2.331785        -6 73403

# This works again
fun_randomword <- function(data) { data$randomword <- runif(nrow(data), 0, 100); return(data) }
pipeline <- lasR::reader() +
  lasR::add_extrabytes("float", "randomword", "Random numbers") +
  lasR::callback(fun_randomword, expose = "xyz") +
  lasR::summarise(metrics = c("randomword_min", "randomword_max", "randomword_mean", "count"))
lasR::exec(pipeline, on = f)$metrics
#>   count randomword_max randomword_mean randomword_min
#> 1 73403       99.99976        50.13884    0.001717894
```

<sup>Created on 2025-06-18 with [reprex v2.1.1](https://reprex.tidyverse.org)</sup>

Jean-Romain, 2025-06-18:
`angle` is a reserved word. `angle_min` is actually interpreted as `ScanAngle_min`

You can find the matching table here https://github.com/r-lidar/lasR/blob/7ffc82c7ae84f7bd4448e4b24ac5b53fab02286d/src/LASRcore/PointSchema.h#L22

The goals are of two kinds:
- Make it easy to write metrics 
- Parse difference files format with different standard in a consistent way. For example if a PCD files has an attribute angle it is understood in a consistant way to be converted to LAS.

I do agree it deserve a better documentation. And you should have add at least a warning it would have helped. And maybe we can be smart and detect the clash to find the correct attribute.

Jean-Romain, 2025-06-18:
I added an explicit error at least temporarily

eikeschott, 2025-06-18:
Thank you for adding a warning. That should clear things up in the future.

I think both goals are pretty well met. But as you said the documentation around extrabytes and dealing with them in later stages (filtering and metrics) could be improved

Jean-Romain, 2025-06-18:
Feel free to make suggestion.