r-lidar/lasR #170: classify_with_sor() is quite slow

wiesehahn, 2025-07-07:
As originally mentioned in https://github.com/r-lidar/lasR/discussions/169_ I recognized that `classify_with_sor()` stage is quite slow, on a sample dataset it was multiple times slower than `classify_with_ivf()` which might be expected as it is more computaionally intensive, but it was also approximately 4x slower than `geometry_features()` which also detects neighbouring points.

The difference is also visible but not that clearly with small sample files included in the pakage.

Not sure but as was mentioned it might be somehow related to https://github.com/r-lidar/lasR/issues/164.

Jean-Romain, 2025-07-07:
Can you reproduce with a file of the package?

``` r
library(lasR)

file <- system.file("extdata", "MixedConifer.las", package = "lasR")

system.time(
  exec(classify_with_ivf(),on = file)
)
#> utilisateur     système      écoulé 
#>       1.818       0.033       1.853

system.time(
  exec(classify_with_sor(),  on = file)
)
#> utilisateur     système      écoulé 
#>       2.295       0.001       2.295

system.time(
  exec(geometry_features(8, features = "pi"), on = file)
)
#> utilisateur     système      écoulé 
#>       3.277       0.003       3.281
```

<sup>Created on 2025-07-07 with [reprex v2.1.1](https://reprex.tidyverse.org)</sup>

Jean-Romain, 2025-07-07:
The first and easy issue that by default it no longer uses half cores....

``` r
library(lasR)

file <- system.file("extdata", "MixedConifer.las", package = "lasR")

system.time(
  lasR::exec(lasR::geometry_features(8, features = "pi"), on = file)
)
#> utilisateur     système      écoulé 
#>       4.792       0.027       4.821

system.time(
  lasR::exec(lasR::geometry_features(8, features = "pi"), ncores = 6, on = file)
)
#> utilisateur     système      écoulé 
#>       4.733       0.006       0.943
```

<sup>Created on 2025-07-07 with [reprex v2.1.1](https://reprex.tidyverse.org)</sup>

wiesehahn, 2025-07-08:
I was just wondering that processing time increases for me when increasing the number of cores, but then I figured out that 
>  If NULL the default is concurrent_points(half_cores()). If a simple integer is provided it corresponds to concurrent_files(ncores).

So increasing time makes sense for a single file, but then your benchmark above makes no sense to me. 

``` r
library(lasR)
#> Warning: Paket 'lasR' wurde unter R Version 4.5.1 erstellt
file <- system.file("extdata", "MixedConifer.las", package = "lasR")

system.time(
  exec(geometry_features(8, features = "pi") + write_las(),
       on = file,
       with = list(ncores = 6)
  )
)
#>        User      System verstrichen 
#>        3.14        0.04        3.22

system.time(
  exec(geometry_features(8, features = "pi") + write_las(),
       on = file
  )
)
#>        User      System verstrichen 
#>        0.56        0.02        0.12
```

<sup>Created on 2025-07-08 with [reprex v2.1.1](https://reprex.tidyverse.org)</sup>

Jean-Romain, 2025-07-08:
> So increasing time makes sense for a single file,

No actually if you have a single file the code is smart enough to convert concurrent files to concurrent points

> is this some caching?

Yes, reading files twice, the second time is always faster. This is something to consider when benchmark. I always perform a pre-run. But the amount here is too big. Actually the first run of `exec` is always takes longer than the subsequent ones because it loads `terra` to read potential rasters produced. And `terra`  has a bug, it takes literally 1 seconds maybe more to load and the maintainer does not want to fix it. To avoid that you can pre-run a dummy pipeline or:

```r
system.time(
  exec(geometry_features(8, features = "pi"),
       on = file,
       with = list(ncores = 6, noread  = T)
  )
)
```
On my side I get 0.13 second both. Both are using 6 cores on my computer. So, same timing than you.

Jean-Romain, 2025-07-15:
This took me way too much time. But knn search is now as fast as in lidR and maybe even faster. `geometry_feature`, `classify_with_sor()` are not computed in normal computation time. The modification are available in the branch `nanoflann` and I will merge as soon as I will be sure that everything is correct on all operating system.