r-lidar/lasR #193: Includes in python binfing

Jean-Romain, 2025-08-13:
I see in `python/binding.cpp`

```cpp
#include "LASRapi/api.h"
#include "LASRcore/RAM.h"
#include "LASRcore/openmp.h"
#include "LASRcore/error.h"
```

You are not supposed to include anything else than `api.h`.  Otherwise you are exposing non api stuff and I can't guarantee to provide backward compatibility.

I can see why  you need `RAM.h`  and `openmp.h`. My `Rbinding.cpp` needs them too. I'm exposing non api stuff to R because it is my code. But you should not. If you want to use ram and openMP helpers functions, I could rather rework `api.h` to provide theses tools natively. 

However I do not see any reason to include `error.h`. Errors are handled internally and if something went wrong you should receive an exception. You must use a `try-catch` block but `error.h` is definitively not supposed to be used. I see that you are using `last_error` line 61. Line 126 `api::execute(config_file, async_communication_file)` you should have a `try-catch` and catch the exception message.

Jean-Romain, 2025-08-13:
I modified `api.h` to add RAM and OpenMP utilities in the `devel` branch. I modified the python `binding.cpp`. You now only have to deal with `error.h`.

agrigoriev, 2025-09-22:
Fixed by #211

closed 2025-09-22 by agrigoriev